esp8266/Arduino #1905: Question about the SDK

edautz, 2016-04-12:
I have a question about the SDK.

It seems that the SDK is an part of the ESP8266/arduino package.
Is it true that when the sketch is compiled to code the needed SDK code is used

I found also a memory map:

Address	 Size	 Name	 Description
00000h	 248k	 app.v6.flash.bin	 User application
3E000h	 8k	 master_device_key.bin	 OTA device key
40000h	 240K	 app.v6.irom0text.bin	 SDK libraries
7C000h	 8k	 esp_init_data_default.bin	 Default configuration
7E000h	 8k	 blank.bin	 Filled with FFh. May be WiFi configuration.

In this memory map I see SDK libraries. Is this also the case when using Ide for compiling the code?

If so is it possible to update the SDK files in the ESP flash memory?

me-no-dev, 2016-04-14:
the final binary is linked with the libraries and then uploaded to the flash memory on the ESP module.
Most of those libs are not open sourced and are released by espressif. We do try to be on the latest when possible.

edautz, 2016-04-14:
Thanks for your reply.

For my understanding how things work:

Is the file app.v6.flash.bin the compiled sketch in binary format?
And the file app.v6.irom0text.bin the uploaded linked libraries?
So when I want to use an newer version of the SDK, I replace the current version within the ..\hardware\esp8266\2.0.0\tools\sdk directory?
And with the next upload the new SDK will be installed in the flash rom of the ESP8266?

igrr, 2016-04-14:
We don't use two separate bin files, instead we structure the sections so that loadable sections follow non-loadable ones. So only one bin file is uploaded, which contains the sketch and linked libraries.
If you'd like to upgrade, you can update the SDK there, but take note that we don't use SDK as is, some modifications to header files and static libraries are done. Also you can update the boards manager package to 2.1.0 or 2.2.0-rc1 and get the latest SDK.

edautz, 2016-04-14:
Ok. That clarifies a lot. Thanks for your feedback.

closed 2016-04-14 by edautz