esp8266/Arduino #1912: SoftAP sometimes do not load page timeout ping

mpena2099, 2016-04-15:
### Basic Infos

#### Hardware
Hardware:			NodeMCU ESP-12E
Core Version:      	2.1.0

### Description

I'm trying to show an HTML file on AP mode. Sometimes it works, sometimes it doesn't (when I reload the page it gets stuck and the board stops to respond the ping request).

My code is just "WiFiAccessPoint" + "FSBrowser" examples.

It is easy to replicate this error if you reload the page fast many times (sometimes it get stuck on the first attempt) or if, in my case, the page has many js and css to load like:

```
		<!-- Include jQuery Mobile stylesheets -->
		<link rel="stylesheet" href="jquery.mobile-1.4.5.min.css">

		<!-- Include the jQuery library -->
		<script src="jquery-1.12.2.min.js"></script>

		<!-- Include the jQuery Mobile library -->
		<script src="jquery.mobile-1.4.5.min.js"></script>

		<!-- jQuery-Clock-Plugin -->
  		<link rel="stylesheet" type="text/css" href="jqClock.css" />
  		<script type="text/javascript" src="jqClock.min.js"></script>
```

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB (3M SPIFFS
CPU Frequency:		160Mhz
Upload Using:		SERIAL

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>
#include <FS.h>

/* Set these to your desired credentials. */
const char *ssid = "ESPap";
const char *password = "thereisnospoon";

ESP8266WebServer server(80);

String formatBytes(size_t bytes)
{
  if (bytes < 1024) {
    return String(bytes)+"B";
  } else if (bytes < (1024 * 1024)) {
    return String(bytes/1024.0)+"KB";
  } else if (bytes < (1024 * 1024 * 1024)) {
    return String(bytes/1024.0/1024.0)+"MB";
  } else {
    return String(bytes/1024.0/1024.0/1024.0)+"GB";
  }
} 
String getContentType(String filename)
{
  if (server.hasArg("download")) return "application/octet-stream";
  else if (filename.endsWith(".htm")) return "text/html";
  else if (filename.endsWith(".html")) return "text/html";
  else if (filename.endsWith(".css")) return "text/css";
  else if (filename.endsWith(".js")) return "application/javascript";
  else if (filename.endsWith(".png")) return "image/png";
  else if (filename.endsWith(".gif")) return "image/gif";
  else if (filename.endsWith(".jpg")) return "image/jpeg";
  else if (filename.endsWith(".ico")) return "image/x-icon";
  else if (filename.endsWith(".xml")) return "text/xml";
  else if (filename.endsWith(".pdf")) return "application/x-pdf";
  else if (filename.endsWith(".zip")) return "application/x-zip";
  else if (filename.endsWith(".gz")) return "application/x-gzip";
  return "text/plain";
}

bool handleFileRead(String path) 
{
  Serial.println("handleFileRead: " + path);
  if (path.endsWith("/")) path += "index.html";
  String contentType = getContentType(path);
  String pathWithGz = path + ".gz";
  if (SPIFFS.exists(pathWithGz) || SPIFFS.exists(path))
  {
    if (SPIFFS.exists(pathWithGz))
      path += ".gz";
    File file = SPIFFS.open(path, "r");
    size_t sent = server.streamFile(file, contentType);
    file.close();
    return true;
  }
  return false;
}

 
void handleRoot() {
	server.send(200, "text/html", "<h1>You are connected</h1>");
}

void setup() {
	delay(1000);
	Serial.begin(115200);
	Serial.println();

  SPIFFS.begin();
  {
    Dir dir = SPIFFS.openDir("/");
    while (dir.next()) 
    {    
      String fileName = dir.fileName();
      size_t fileSize = dir.fileSize();
      Serial.printf("FS File: %s, size: %s\n", fileName.c_str(), formatBytes(fileSize).c_str());
    }
    Serial.printf("\n");
  }

 
	Serial.print("Configuring access point...");
	/* You can remove the password parameter if you want the AP to be open. */
	WiFi.softAP(ssid, password);

	IPAddress myIP = WiFi.softAPIP();
	Serial.print("AP IP address: ");
	Serial.println(myIP);

  // SERVER INIT
  server.on("/", HTTP_GET, []() {
    if (!handleFileRead("/index.htm")) server.send(404, "text/plain", "FileNotFound");
  });

  //called when the url is not defined here
  //use it to load content from SPIFFS
  server.onNotFound([](){
    if(!handleFileRead(server.uri()))
      server.send(404, "text/plain", "FileNotFound");
  });

  server.begin();
  Serial.println("HTTP server started");  
}

void loop() {
	server.handleClient();
}
```

### Debug Messages

```
chksum 0x42
csum 0x42
rl l��| �l�|�l�b|����r�b�b��nn�lnn���bp�lrlrlp�n��l��bn�|l����b��nn� l��l`�nnl`nr���n��l`r��n��l`��n�
FS File: /edit.htm.gz, size: 4.02KB
FS File: /favicon.ico, size: 1.12KB
FS File: /graphs.js.gz, size: 1.92KB
FS File: /index.htm, size: 3.63KB

Configuring access point...AP IP address: 192.168.4.1
HTTP server started
handleFileRead: /index.htm
handleFileRead: /graphs.js
handleFileRead: /all
handleFileRead: /index.htm
handleFileRead: /graphs.js
handleFileRead: /all
handleFileRead: /all
handleFileRead: /all
```

### Ping Test
```
64 bytes from 192.168.4.1: icmp_seq=7644 ttl=255 time=2.663 ms
64 bytes from 192.168.4.1: icmp_seq=7645 ttl=255 time=3.952 ms
64 bytes from 192.168.4.1: icmp_seq=7646 ttl=255 time=2.958 ms
64 bytes from 192.168.4.1: icmp_seq=7647 ttl=255 time=3.862 ms
64 bytes from 192.168.4.1: icmp_seq=7648 ttl=255 time=2.954 ms
64 bytes from 192.168.4.1: icmp_seq=7649 ttl=255 time=8.292 ms
Request timeout for icmp_seq 7650
Request timeout for icmp_seq 7651
Request timeout for icmp_seq 7652
Request timeout for icmp_seq 7653
Request timeout for icmp_seq 7654
Request timeout for icmp_seq 7655
Request timeout for icmp_seq 7656
64 bytes from 192.168.4.1: icmp_seq=7657 ttl=255 time=12.436 ms
64 bytes from 192.168.4.1: icmp_seq=7658 ttl=255 time=4.029 ms
64 bytes from 192.168.4.1: icmp_seq=7659 ttl=255 time=1.707 ms
64 bytes from 192.168.4.1: icmp_seq=7660 ttl=255 time=3.380 ms
```

igrr, 2016-04-17:
Make sure you add `WiFi.mode(WIFI_AP)` if you aren't using STA interface. Otherwise it might happen that STA interface is trying to connect to some AP, and fails constantly, which interrupts SoftAP connection.
See comments in Espressif SDK docs: https://github.com/esp8266/Arduino/issues/119#issuecomment-157493799

mpena2099, 2016-04-29:
No effect adding "WiFi.mode(WIFI_AP)", or all the following lines:

```
ESP.eraseConfig();  
WiFi.disconnect();
WiFi.softAPdisconnect();
WiFi.setOutputPower(20.5);
WiFi.persistent(false);
wifi_station_disconnect();
wifi_set_phy_mode(PHY_MODE_11B);
WiFi.mode(WIFI_AP);
```

Sometimes the page loads OK, sometimes it doesn't. :-/

mpena2099, 2016-04-29:
Same problem here in STA mode, so that's not a SoftAP problem... :(

SPIFFS perhaps.

mpena2099, 2016-04-30:
Enabling mDNS help me solve this problem: its is not about SoftAP, SPIFFS and stuff. 

It's all because "the maximum number of TCP sockets is at 5"(https://github.com/esp8266/Arduino/issues/928#issuecomment-150810446)

So... that error happens when I can't put my jQuery and other files on remote servers on SoftAP mode... ;)

Thanks @Links2004.

End.

closed 2016-04-30 by mpena2099