esp8266/Arduino #1913: Disconnecting periodically with ssl->need_bytes debug message

InconceivableDuck, 2016-04-15:
Hardware is an Adafruit Feather, but seeing same behavior on NodeMcu boards as well. Version is 2.1.0 as shown in Arduino board manager.

I have a persistent connection open to an MQTT server. TLS termination is being performed by Haproxy with the tune.ssl.maxrecord size reduced to 1419 from the default 16kb.

The device is sending a short MQTT message every 600 milliseconds. The server immediately replies. This means there's a message roughly every 300 milliseconds.

The sketch typically runs successfully for several minutes and then drops the connection. Below are a few of the messages received while testing.

ssl->need_bytes=19177 > 17403
ssl->need_bytes=41147 > 17403

The frequency of receiving these message appears to be correlated to the amount of delay added between loops. For instance, I receive more if I delay 100ms between loops as compared to 10ms.

Digging through the code I understand where the 17403 is allocated. I'm not sure why ssl is requiring arbitrarily large buffers at random times. I suspect it's due to a large amount of TCP data being read on a single loop that requires decryption. Is there a way to control the total amount of TCP data that will read and decrypted at a single time.

InconceivableDuck, 2016-04-15:
The TLS spec puts the record size in bytes 3 and 4 of the record header. 2 bytes can only be a max of 16kb. There appears to be some bug in axTLS causing some wonky reads.

igrr, 2016-04-15:
This also happens when you connect to a server which doesn't talk TLS to you (e.g. plain HTTP). At least we had one similar report which was due to a wrong port.
Could you try capturing wireshark dump to see the real bytes transferred?

igrr, 2016-04-15:
Edit: sorry, I missed the "runs successful for several minutes" part. The port is obviously correct then.

igrr, 2016-05-10:
Linking to #1935, same reason.

closed 2016-06-23 by igrr