esp8266/Arduino #1914: NewPing not working when using pins D5-D8

vermut, 2016-04-15:
### Basic Infos

#### Hardware
Hardware:			ESP-8266EX (WeMos D1 mini)
Core Version:      	2.1.0

### Description

[NewPing](https://github.com/PaulStoffregen/NewPing/blob/master/NewPing.cpp#L20) not working when using pins D5-D8. 

### Settings in IDE

Module:  			WeMos D1 R2 mini
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Upload Using:		SERIAL

### Sketch

```cpp
#define PIN     D6     // Not working
// #define PIN     D3     // Working

uint32_t _triggerBit = digitalPinToBitMask(PIN); // Get the port register bitmask for the pin.
volatile uint32_t *_triggerOutput = portOutputRegister(digitalPinToPort(PIN)); // Get the output port register for the pin.
volatile uint32_t *_triggerMode = (uint32_t *) portModeRegister(digitalPinToPort(PIN)); // Get the port mode register for the pin.

void setup() {
  *_triggerMode |= _triggerBit; // Set trigger pin to output.
}

void loop() {
  *_triggerOutput &= ~_triggerBit; // Set the trigger pin low
  delay(1000);
  *_triggerOutput |= _triggerBit;  // Set trigger pin high
  delay(1000);              // wait for a second
}
```

tablatronix, 2016-08-04:
` pinMode(PIN,OUTPUT);`
 works

` *_triggerMode |= _triggerBit; `
does not

I am also having trouble doing the same thing with a capsense sketch I am testing.
D2 here

tablatronix, 2016-08-06:
On the esp don't you have to set `gpio mode` also ? I do not see how this is being done,I tried debugging this but once i got to the register definitions I was lost...

devyte, 2017-10-14:
@vermut please use pinMode instead of the low level register handling.
This looks like user error to me, closing.

closed 2017-10-14 by devyte