esp8266/Arduino #1915: Extremely slow AP association

lbedogni, 2016-04-16:
Hi, I have a simple program which reads from a sensor and send the data to a remote server.However, I notice that it takes considerable time to associate to the AP. I switched to static IP, but still the problem is in the association, which takes around 6 seconds.

The relevant code is this
`WiFi.begin(ssid, password);
  if (ip != (0,0,0,0) && gateway != (0,0,0,0) && netmask != (0,0,0,0)) {
    WiFi.config(ip, gateway, netmask);
    Serial.println("Configuring..");
    delay(10);
  }
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.println("Not Connected!");
  }
ip = WiFi.localIP();
  netmask = WiFi.subnetMask();
  gateway = WiFi.gatewayIP();`

What takes most of the time is the while, other steps, including the data send (not included here) take very little time, but the loop accounts for nearly 6 out of 6.little seconds on the average.

devyte, 2018-02-28:
@lbedogni is this issue still valid with 2.4.0? is it still valid with latest git?

devyte, 2018-07-12:
Closing due to lack of feedback and age. Also, I don't observe this behavior in my own app.

closed 2018-07-12 by devyte