esp8266/Arduino #1918: Exception (29):

framb00s, 2016-04-16:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	?2.1.0-rc2? the board manager version is 2.1.0
 Arduino Ide 1.6.8 on Lenovo Yoga3 with dual boot windows 10 and Ubuntu 15.10
+
Macbook with dualboot OSX10.7 and Ubuuntu 14.04
All have dropbox except the Ubuntu 15.10

### Description

Problem description
The program compiles on all OS's but generates an exception (29) error

However after I did a clean install of Ubuntu 15.10 on the Lenovo the program worked without problems. But after I imported a software (on the Ubuntu 15.10)  with a USB stick from the Macbook I got the same error as subscribed. 

I found some sort off workaround. I download the original version from https://github.com/Dustproof/dustcube and run it on the nodeMCU it doesn't work (because I've written it for Aduino 328) BUT then the software (which I copied below) runs normal without exception 29.

### Settings in IDE

Module:  			NodeMCU 1.0(ESP-12E module)
Flash Size: 		4MB(3M SPIFFS)
CPU Frequency:	80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

// Constants
//const int pinP1 = 8;                      // Dust sensor P1 connected to digital pin 8 on Arduino
//const int pinP2 = 9;                      // Dust sensor P2 connected to digital pin 9 on Arduino
const int pinP1 = 5;                   // Dust sensor P1 connected to digital pin D1 on ESP
const int pinP2 = 4;                   // Dust sensor P2 connected to digital pin D2 on ESP
const char tab = 9;                     // Tab character used for human readable output

// Variables
unsigned long sampleTime = 30000;      // Length of the particle sampling period (30000ms = 30s)
// Variable can be changed depending on dataHandler (Lora).
unsigned long seconds;                  // Time the MCU is running (in seconds = millis()/1000 )
unsigned long startTime;                // The particle sampling period start time
unsigned long durationP1 = 0;           // Duration of time in sampling period during which
// P1 particles are detected,
// aka Low Pulse Occupancy (lpo)
unsigned long durationP2 = 0;           // Low Pulse Occupancy for P2 - can be regarded as PM2.5
float ratioP1 = 0;                      // % of time a P1 particle is detected during the sampling period 
float ratioP2 = 0;                      // % of time a P2 particle is detected during the sampling period 
                                        // P2 can be regarded as PM10
float ratioPM25 = 0;                    // % of time PM2.5 is detected
unsigned long startP1;                  // P1 pulse duration start
unsigned long startP2;                  // P2 pulse duration start
unsigned long pulseP1;                  // Duration of one P1 pulse
unsigned long pulseP2;                  // Duration of one P2 pulse
int nP1 = 0;                            // Number of P1 reads
int nP2 = 0;                            // Number of P2 reads
boolean P1 = HIGH;                      // P1 dust detected? LOW = yes - HIGH = no
boolean P2 = HIGH;                      // P2 dust detected? LOW = yes - HIGH = no
boolean prevP1 = HIGH;                  // Previous state of P1
boolean prevP2 = HIGH;                  // Previous state of P2

void setup() {

  // initialize Serial communication at 9600 bits per second:
  Serial.begin(9600);
  Serial.println("Program started, discard first 90 warmup seconds");
  Serial.print("sec");
  Serial.print(tab);
  Serial.print("rPM2.5");
  Serial.print(tab);
  Serial.print("rPM10");
  Serial.print(tab);
  Serial.print("nP1");
  Serial.print(tab);
  Serial.println("nP2");

  // configure the dust sensor P1 and P2 pins as input:
  pinMode(pinP1, INPUT_PULLUP);   // changed to pullup because there is nothing attached
  pinMode(pinP2, INPUT_PULLUP);

  startTime = millis();
}

void loop() {
  // calculate P1 pulses:
  prevP1 = P1;                                      // store old value in prevP1
  P1 = digitalRead(pinP1);

  if ((P1 == LOW) && (prevP1 == HIGH)) {            // pulse from low to high, dust detected
    startP1 = micros();                             // store start time
    nP1++;                                          // increment number of particles read
  }
  else if ((P1 == HIGH) && (prevP1 == LOW)) {       // pulse back from high to low, dust has passed, calculate the time
    pulseP1 = micros() - startP1;
    durationP1 = durationP1 + pulseP1;
  }

  // calculate P2 pulses:
  prevP2 = P2;                                      // store old value in prevP2
  P2 = digitalRead(pinP2);

  if ((P2 == LOW) && (prevP2 == HIGH)) {           // check if dust was detected
    startP2 = micros();                             // store start time
    nP2++;                                          // increment number of particles read
  }
  else if ((P2 == HIGH) && (prevP2 == LOW)) {       // dust has passed, calculate the time
    pulseP2 = micros() - startP2;
    durationP2 = durationP2 + pulseP2;
  }

  // at the end of the sampling period:
  if ((millis() - startTime) >= sampleTime) {       // check if the sampling period has passed

    if ((P1 == LOW) && (prevP1 == LOW)) {           // if a P1 particle was detected during the sampling period
      pulseP1 = micros() - startP1;                 // add the time
      durationP1 = durationP1 + pulseP1;
    }
    if ((P2 == LOW) && (prevP2 == LOW)) {           // if a P2 particle was detected during the sampling period
      pulseP2 = micros() - startP2;                 // add the time
      durationP2 = durationP2 + pulseP2;
    }
    seconds = millis() / 1000;                                    // save the seconds
    ratioP1 = durationP1 / (sampleTime * 10.0);                   // calculate the P1 lpo ratio
    ratioP2 = durationP2 / (sampleTime * 10.0);                   // calculate the P2 lpo ratio
    ratioPM25 = (durationP1 - durationP2) / (sampleTime * 10.0);  // calculate the PM2.5 lpo ratio

    data2Serial();                    // print human readable data to serial

    // reset pulse calculation variables for the next sampling period:
    prevP1 = HIGH;
    prevP2 = HIGH;
    durationP1 = 0;
    durationP2 = 0;
    nP1 = 0;
    nP2 = 0;
    startTime = millis();
  }
}

void data2Serial() {
  // print measurements in human readable format:
  Serial.print(seconds);
  Serial.print(tab);
  Serial.print(ratioPM25);
  Serial.print(tab);
  Serial.print(ratioP2);
  Serial.print(tab);
  Serial.print(nP1);
  Serial.print(tab);
  Serial.print(nP2);
  Serial.println();
}

### Debug Messages
 1264, room 16 
tail 0
chksum �txW�X6��$�p�Program started, discard first 90 warmup seconds
sec	rPM2.5	rPM10	nP1	nP2
30	0.00	0.00	0	0

Exception (29):
epc1=0x4021adff epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000019 depc=0x00000000

ctx: sys 
sp: 3ffffcb0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe50:  ffffffd3 80af1999 40240000 ffffff1a  
3ffffe60:  3ffffe80 4021af0f 3ffed414 ffffff1a  
3ffffe70:  00000007 00000007 42d60000 42d60000  
3ffffe80:  00000018 3ffedb34 3ffe83c0 3ffe83c0  
3ffffe90:  3fff010f 00000000 3ffefd0c 0000000d  
3ffffea0:  4021cb6f 3ffedae8 3ffedbc0 0000010f  
3ffffeb0:  00000001 4020a4c4 000002e1 ffffffd3  
3ffffec0:  4021c824 4021c7f8 3ff20200 00000000  
3ffffed0:  4021e5a9 00000000 4020a59e 00000001  
3ffffee0:  3ffed498 4021e570 00000001 00000012  
3ffffef0:  4021c7ee 00000001 3ffedae8 0000010f  
3fffff00:  4021c795 00000001 00000012 0000010f  
3fffff10:  4021cc88 00000001 3ffedae8 0000010f  
3fffff20:  4021c710 3ffed588 00000064 40107208  
3fffff30:  3ffed498 00000000 4021ebd0 4021e98c  
3fffff40:  0000010f 3ffe8850 025254e5 60000600  
3fffff50:  4021e866 3ffefd10 fffffffc 3ffe8cbc  
3fffff60:  4021eb42 3ffefd10 3ffe8850 3ffe8850  
3fffff70:  3ffefd0c 40205295 00000000 00000000  
3fffff80:  402052da 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe8878 3fffdad0 3ffeeb40 40202987  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<
messages here
```

igrr, 2016-04-17:
Could you please decode the stack dump and paste the result here? Thanks.
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

framb00s, 2016-04-17:
Hi iggr
Here a dump of the stack and the decode
Luc
Exception (29):
epc1=0x4021adff epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000019 depc=0x00000000

ctx: sys 
sp: 3ffffd00 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffea0:  00000001 00001388 3ff20200 ffffffd3  
3ffffeb0:  3ffffed0 4021af0f 3ffed414 0000000e  
3ffffec0:  4020e832 0000000e 3ffedbc0 00000064  
3ffffed0:  00000018 3ffedb34 3ffe83c0 3ffe83c0  
3ffffee0:  3fff010f 00000001 3ffefd0c 0000000d  
3ffffef0:  4021cb6f 3ffedae8 3ffedbc0 0344ce98  
3fffff00:  00000001 4020b830 3ffed498 40106418  
3fffff10:  4021c824 4021c7f8 0000001c 40100774  
3fffff20:  4021e5a9 4021c7f8 00000064 40107208  
3fffff30:  3ffed534 4021e570 3ffedb00 3ffe8850  
3fffff40:  4021c7ee 3ffed414 40106431 00000000  
3fffff50:  4021c795 3ffedb00 3ffe8850 0344ce98  
3fffff60:  4021cc48 40205295 3ffe8cbc 3ffe8850  
3fffff70:  40205295 60000600 00000000 00000000  
3fffff80:  402052da 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe8868 3fffdad0 3ffeeb40 40202987  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<
Paste your stack trace here

Decoding 24 results
0x4021adff: ieee80211_add_xrates at ?? line ?
0x4021af0f: ieee80211_send_probereq at ?? line ?
0x4020e832: phy_after_init_enrx at ?? line ?
0x4021cb6f: scan_remove_probe_ssid at ?? line ?
0x4020b830: chip_v6_set_chanfreq at ?? line ?
0x40106418: chm_get_current_channel at ?? line ?
0x4021c824: scan_pm_channel_op_cb at ?? line ?
0x4021c7f8: scan_pm_channel_op_cb at ?? line ?
0x40100774: malloc at /home/flashbook/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/umm_malloc/umm_malloc.c line 1658
0x4021e5a9: chm_start_op at ?? line ?
0x4021c7f8: scan_pm_channel_op_cb at ?? line ?
0x40107208: pvPortMalloc at /home/flashbook/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/heap.c line 13
0x4021e570: chm_start_op at ?? line ?
0x4021c7ee: scan_pm_channel_op_cb at ?? line ?
0x40106431: chm_get_current_channel at ?? line ?
0x4021c795: scan_pm_channel_op_cb at ?? line ?
0x4021cc48: scan_clear_channles at ?? line ?
0x40205295: ets_timer_handler_isr at ?? line ?
0x40205295: ets_timer_handler_isr at ?? line ?
0x402052da: ets_timer_handler_isr at ?? line ?
0x40202987: loop_task at /home/flashbook/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/core_esp8266_main.cpp line 43

closed 2016-04-17 by framb00s