esp8266/Arduino #1919: build.flash_ld not being expanded

sej7278, 2016-04-16:
When using arduino-builder and/or a Makefile I get the following error:

```ld: cannot open linker script file {build.flash_ld}: No such file or directory```

It seems {build.flash_ld} isn't getting expanded, although it works from the IDE (1.6.6-1.6.8 at least)

I only have this problem with the core 2.2.0, the tagged 2.1.0 release works fine (and earlier).

Its not practical to hardcode this in platform.txt, so how was it done in 2.1.0 (compiler.c.elf.ldscript=eagle.app.v6.ld) ?

Seems to have been added in commit https://github.com/esp8266/Arduino/commit/540fdb0f8c6e73df2aec8bfba4b190d64cda32e0

igrr, 2016-04-16:
Could you please post how you invoke arduino-builder, the complete command line if possible?

Regarding the second point ('when did we switch to using GPIO numbers as Arduino digital pin numbers'), this is how it worked since the very first version of this core. This even mentioned in the first line of the documentation: http://esp8266.github.io/Arduino/versions/2.1.0/doc/reference.html#digital-io.

sej7278, 2016-04-17:
Here's my Makefile, its based on the same arduino-builder command the IDE generates, so yes you could just call it as a command rather than from make:

```make
ARDUINO_PATH = $(HOME)/arduino-1.6.8
SKETCHBOOK   = $(HOME)/arduino16
SKETCH 		 = $(notdir $(CURDIR)).ino
TARGET_DIR   = $(CURDIR)/build-esp8266

all:
	@ mkdir -p $(TARGET_DIR)

	$(ARDUINO_PATH)/arduino-builder -compile -logger=machine \
	-hardware "$(ARDUINO_PATH)/hardware" \
	-hardware "$(SKETCHBOOK)/hardware" \
	-tools "$(ARDUINO_PATH)/tools-builder" \
	-tools "$(ARDUINO_PATH)/hardware/tools/avr" \
	-built-in-libraries "$(ARDUINO_PATH)/libraries" \
	-libraries "$(SKETCHBOOK)/libraries" \
	-fqbn=esp8266com:esp8266:generic:UploadTool=esptool,CpuFrequency=80, \
               FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64 \
	-ide-version=10608 \
	-build-path "$(TARGET_DIR)" \
	-warnings=none \
	-prefs=build.warn_data_percentage=75 \
	-verbose "$(SKETCH)"
```

Regarding the pinout, yes I know its documented, but I assumed the documentation had been updated as I was sure it wasn't always that way, how odd!

sej7278, 2016-06-07:
i still get the same error with ide 1.6.9 and core 2.3.0

jntorresn, 2016-06-08:
I am having the same error, but i solved it by changing the file "boards.txt", I used the file from version 2.1.0 and pasted it on 2.2.0 version. Then I compile using 2.2.0 and I get my .bin. But I doesn't work very well for me I still have a problem using openocd with ESP8266. I get error (135).

igrr, 2016-06-08:
This usually happens when you ask for flash size which isn't defined for
the given board. I.e. if you set the board to nodemcu and flash size to
512K64, you will get such an error.

igrr, 2016-06-08:
@sej7278, you Makefile has whitespace in the middle of FQBN value. AFAICT Arduino-builder expects fqbn to be just one command line token, separated by commas. FlashSize happens to go after the space and is not taken into account. Therefore linker script name doesn't get expanded.

igrr, 2016-06-08:
This Makefile is using arduino-builder to build, upload, and run test cases, and it works fine for me: https://github.com/esp8266/Arduino/blob/master/tests/device/Makefile#L13

sej7278, 2016-06-08:
sorry the example Makefile above i added whitespace for it to wrap properly, i've not got it in my actual makefile

closed 2016-06-08 by sej7278

suculent, 2017-08-16:
My workaround was following... as Arduino is installed together with ESP8266 libs into docker:

`echo "d1_mini.build.flash_ld=eagle.flash.4m1m.ld" >> "/root/Arduino/hardware/esp8266com/esp8266/boards.txt"`

Which is default to my board, however I'm wondering how to do this correctly.