esp8266/Arduino #1921: 64 Character WPA2 passwords do not work

MB3hel, 2016-04-16:
### Basic Info
64 Character WPA2 passwords do not work.
#### Hardware
Hardware:			ESP-12 (NodeMCU 1.0 V3)
Core Version:      	2.1.0

### Description

64 byte(character) WPA2 passwords do not work. The library will return WL_CONNECT_FAILED then WL_IDLE_STATUS. In ESP8266WiFiSTA.cpp in the begin function there is a condition stating that if the password is longer than 63 characters return WL_CONNECT_FAILED. This needs to be 64 characters because strlen is 1 indexed not 0 indexed.  When I made that change it worked correctly.
```cpp
if(passphrase && strlen(passphrase) > 63) {  //This needs to be 64
        // fail passphrase too long!
        return WL_CONNECT_FAILED;
}
```
### Settings in IDE

Module:  			NodeMCU 1.0(ESP-12E Module)
Flash Size: 		4M
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <IPAddress.h>
 
const char* ssid = "SSID";
const char* password = "abcdefghijklmnopqrstuvwxyzabcdefghijklmnopqrstuvwxyzabcdefghijkl";
 
void setup(void)
{
  Serial.begin(115200);
  Serial.println("Connecting...");
 
  WiFi.begin(ssid, password);

  int s = -999;
  
  // Wait for connection
  while ((s = WiFi.status()) != WL_CONNECTED) {
    delay(500);
    Serial.println(s);
  }
 
  Serial.println(WiFi.localIP());
 
}

void loop(){}
```
### Debug Messages(Serial Monitor)
```
Connecting...
4
0
0
0
0
```

igrr, 2016-04-16:
Please check 2.2.0-rc1 or git version, this has been fixed in 1b8f6d2e:

https://github.com/esp8266/Arduino/blob/1b8f6d2e8e2dc7008660732b96197fd1ae1a1439/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L109-L124

closed 2016-06-23 by igrr

zgoda, 2017-06-13:
According to 802.11 spec, WPA passphrases are 8-63 ASCII chars so this is not a bug and I guess it should stay as it was. See https://stackoverflow.com/a/21554182/12138

percz, 2017-12-05:
This bug of sorts seems to be in ESP8266WiFiMulti.cpp as well, around line 174...

`    if(passphrase && strlen(passphrase) > 63) {
        // fail passphrase to long!
        DEBUG_WIFI_MULTI("[WIFI][APlistAdd] passphrase to long\n");
        return false;
    }`

While it's right that the passphrase can't be more than 63 chars it should be possible to specify a 64 hex character PSK, as mentioned in zgoda's link, for a non-ASCII based PSK.
In fact just changing that 63 to a 64 seems to work perfectly, although the ESP8266WiFiSTA.cpp error-handling code added by igrr would be proper.