esp8266/Arduino #1924: Two clients using a esp8266 as AP cant talk to each other

DavidXanatos, 2016-04-17:
Two clients using a esp8266 as AP cant talk to each other,
module, etc... not relevant tried multiple.
Apparently this functionality is not implemented.

It would be great if that could be added :D

igrr, 2016-04-17:
We can do this if we switch from Espressif-provided lwIP binary to a GCC-built one. Actually open source lwIP just became part of the SDK release (1.5.3), which is a good sign.

far5893, 2016-04-17:
I suppose that if you introduce switching (client-client), you can introduce routing(client-esp8266AP-esp8266STA-wifiAP)!

devyte, 2017-10-17:
@d-a-v I suppose this is something we would want to look at only after #3362 ?

d-a-v, 2017-10-17:
We discussed about that in #3243.
Routing is possible with current lwip1.4 but needs to be recompiled and tested with ```IP_FORWARD=1``` in ```tools/sdk/lwip/include/lwipopts.h```. It is disabled by default (same in lwip2). I have not tested myself.
Two clients talking to each other (or broadcast) through their AP is not strange and should be enabled. Library size is likely to increase. But in AP+STA mode this will not enable AP's clients to talk to STA peers without NAT (on esp) or specific configuration on STA's router.

d-a-v, 2017-10-26:
This is in fact a bug in lwip1.4. The issue's title should not be routing neither forwarding, it's the regular working way for two clients of the same AP to talk to each other.
Using WiFiAccessPoint example sketch, clients of the same network of an esp AP can talk to each other *without* ```IP_FORWARD=1``` but with lwip2 #3362. I could ssh from my phone to my computer.
With lwip1.4, I could only ping my phone from my computer (but strangely not the other way).

devyte, 2018-01-07:
@d-a-v I think this should be fairly easy to test and close now with lwip2 merged, right?

danielicy, 2018-08-07:
hi there I tried setting IP_FORWARD=1 both in lwip and lwip2 . I rebuilt the libs but with no succes any suggestions?

d-a-v, 2018-10-12:
@danielicy Can you explain what you are trying to achieve with IP_FORWARD ?

To my knowledge IP_FORWARD is useful in two cases:
- **1.** multiple interfaces enabled (AP, STA),
  at least one external router (the STATION interface network router - generally your router-AP) *informed* that the esp8266 is a router for the other "192.168.4./24" network (the one on the SOFTAP interface)
- **2.** if lwIP is NAT-enabled (this is not the case today).

Are you in case 1 ?
If you are not, it is useless to enable this feature, it won't be used.

danielicy, 2019-01-29:
I want to build a mesh network

d-a-v, 2019-01-29:
@danielicy 

https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFiMesh/examples/HelloMesh

danielicy, 2019-01-29:
@d-a-v  G_d bless you

d-a-v, 2019-08-29:
NAT is now available in lwIP v2 w/ features. So even when not using NAT, routing is enabled.
Please open a new issue with MCVE if routing is not working the way it should.

closed 2019-08-29 by d-a-v