esp8266/Arduino #1925: Add support for function objects in attachInterrupt

igrr, 2016-04-17:
Allow something like 
```c++
attachInterrupt(pin, [](){
    // function body
  }, CHANGE);
```
and
```c++
class MyClass() {
public:
  MyClass(int pin1, int pin2) {
    attachInterrupt(pin1, std::bind(&MyClass::onPinChange, this, pin1), CHANGE);
    attachInterrupt(pin2, std::bind(&MyClass::onPinChange, this, pin2), CHANGE);
  }

protected:
  void onPinChange(int pin) {
    // handler here
  }
};
```
without breaking usage of attachInterrupt from C code.

hreintke, 2016-11-26:
Like to use this functionality within my application and willing to give it a try creating a PR for this.
Do you already have a basic idea on how you want to have it implemented.
If so, I can base my work on that.

neu-rah, 2016-11-27:
i've working on something like this, pin change interrupts with an optional cargo, https://github.com/neu-rah/PCINT

hreintke, 2016-12-08:
See https://github.com/esp8266/Arduino/pull/2745
(#2745)

zenmanenergy, 2017-05-15:
Wow. Did this ever get into the core? This is a very cool idea.

ferchinas, 2017-08-31:
Hi:
I am trying to attach a method of my class to attend a Gpio interrupt. But I get the following error:
GwiModules.cpp: 283:122: error: cannot convert 'std::_Bind_helper<false, void (GwiManual::*)(char), GwiManual* const, int>::type {aka std::_Bind<std::_Mem_fn<void (GwiManual::*)(char)>(GwiManual*, int)>}' to 'void (*)()' for argument '2' to 'void attachInterrupt(uint8_t, void (*)(), int)

NodeMCU V1.0
Arduino IDE V1.8.3
Arduino/Esp8266 V2.3

GwiModules.h
```
  class GwiManual : public Controls
  {
  protected:
	  void event(void);							
	  
  public:
	  GwiManual();
	  void enable(void);	 
  };
```
GwiModules.cpp
```
GwiManual::GwiManual() {
	...
}

void GwiManual::event(void) {
	...
}

void GwiManual::enable(void) {
	...
	attachInterrupt(13, std::bind(&GwiManual::event,this), CHANGE
}

```

devyte, 2017-08-31:
Did you #include "FunctionalInterrupts.h"?

ferchinas, 2017-09-01:
I did not include it, but thanks to your message I discovered that this information is for the development version.
I can not find how to install it in the IDE, only found for stable version.

devyte, 2017-09-01:
Read using git version on readthedocs. You have to delete the stable core
dir to make sure the git is picked up.

ferchinas, 2017-09-01:
Finally get it installed and tested, it works very well, thank you very much for the help.

devyte, 2017-10-17:
@igrr the relevant PR is already merged. Should this be closed, or left open with label staged-for-release, or something else?

tonton81, 2018-04-22:
shouldnt you also show the state change not just the pin itself? some people might want to check rising and falling edges, not just changes, if a state was returned the user wouldnt need to call digitalRead when the interrupt fires to know whether the pin changed high or low as well

devyte, 2018-07-12:
This is already implemented via #2745 .

closed 2018-08-01 by devyte