esp8266/Arduino #1926: Make gcc-built lwIP default one

igrr, 2016-04-17:
In SDK 1.5.3 lwIP source code in included with the SDK release, which is a good indication that future SDK versions will include that as well. This issue tracks switching from Espressif-provided xcc-built lwIP to gcc-built one.

- [ ] Remove workarounds made in ClientContext to work around return value passing bug
- [ ] Update lwIP source code from the latest SDK and build new binary
- [ ] Update boards.txt to use gcc built lwip by default

This will enable us to fix #1924, #1826, #1505, #1050, #1262.

andig, 2016-05-08:
@igrr fyi- I remember you metioning that gcc code size is bugger, however according to my tests it became smaller when switching lwip.

igrr, 2016-05-08:
Yep, I think I have used -O0 in my early tests.
@me-no-dev actually pointed out that gcc-prebuilt LwIP produces the smallest binary when he added an option to switch between LwIP versions from the IDE.

closed 2016-06-23 by igrr