esp8266/Arduino #1928: Raspberry Pi version

hulkco, 2016-04-17:
### Description
Hi iam a Teacher in mexico,  after many tries the Arduino IDE (1.6.9) was compiling sucesful, im Raspberry,  but went a trie to install the ESP8266 Arduino tool, i have a error.

1.- In Stable version: Tool esptool is not available for your operating system
2.- In Staging version: Tool mkspiffs is not vailable for your operating system
Thanks for your atention.
Best Regarts

### Screen Capture

[image]
[image]

igrr, 2016-04-17:
I have added ARM tools last week, they are not yet part of stable/staging versions. You can try using git version of the core (see instructions on the front page).

hulkco, 2016-04-17:
Thanks Igor, i check rigth now.
:)

hulkco, 2016-04-17:
Thanks Ivan Works like a charm!!!!
[image]
[image]

closed 2016-04-17 by hulkco