esp8266/Arduino #1929: Raw stging literal containg double slash (//) causes compile error

Zeron, 2016-04-17:
### Basic Infos

#### Hardware
Hardware:		ESP-12
Core Version:      	GIT 2014-04-17

### Description

When using a raw string literal that contains a // compile fails using Arduino 1.6.8 on windows with latest git.

The same code works on Arduino 1.6.5 with 2.0.0 and 2.1.0

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

ESP8266WebServer WebServer(80);

const char HTML_Root[] PROGMEM = R"=-=-=(
  //
)=-=-=";

void setup() {
  WebServer.on("/", WebPageRoot);
  WebServer.begin();
}

void loop() {
  WebServer.handleClient();
}

void WebPageRoot() {
  WebServer.send(200, "text/html", HTML_Root);
}
```

### Debug Messages

```
Arduino: 1.6.8 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 4M (1M SPIFFS), nodemcu, Disabled, None"

C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\arduino-builder -dump-prefs -logger=machine -hardware "C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware" -tools "C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\tools-builder" -tools "C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\tools\avr" -built-in-libraries "C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\libraries" -libraries "C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\portable\sketchbook\libraries" -fqbn=esp8266com:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=4M1M,ResetMethod=nodemcu,Debug=Disabled,DebugLevel=None____ -ide-version=10608 -build-path "C:\Users\Dan\AppData\Local\Temp\buildf27ee2924724667552df4bbb5f7a5f9c.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\portable\sketchbook\sketch_apr17a\sketch_apr17a.ino"
C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\arduino-builder -compile -logger=machine -hardware "C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware" -tools "C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\tools-builder" -tools "C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\tools\avr" -built-in-libraries "C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\libraries" -libraries "C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\portable\sketchbook\libraries" -fqbn=esp8266com:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=4M1M,ResetMethod=nodemcu,Debug=Disabled,DebugLevel=None____ -ide-version=10608 -build-path "C:\Users\Dan\AppData\Local\Temp\buildf27ee2924724667552df4bbb5f7a5f9c.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\portable\sketchbook\sketch_apr17a\sketch_apr17a.ino"
"C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/sdk/lwip/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\variants\generic" "C:\Users\Dan\AppData\Local\Temp\buildf27ee2924724667552df4bbb5f7a5f9c.tmp\sketch\sketch_apr17a.ino.cpp" -o "nul"
"C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/sdk/lwip/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\variants\generic" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src" "C:\Users\Dan\AppData\Local\Temp\buildf27ee2924724667552df4bbb5f7a5f9c.tmp\sketch\sketch_apr17a.ino.cpp" -o "nul"
"C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/sdk/lwip/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\variants\generic" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src" "C:\Users\Dan\AppData\Local\Temp\buildf27ee2924724667552df4bbb5f7a5f9c.tmp\sketch\sketch_apr17a.ino.cpp" -o "nul"
"C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/sdk/lwip/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\variants\generic" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src" "C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFi.cpp" -o "nul"
"C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266/tools/sdk/lwip/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\variants\generic" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src" "-IC:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src" "C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\libraries\ESP8266W [... cut ...]
sketch_apr17a:11: error: 'WebPageRoot' was not declared in this scope

   WebServer.on("/", WebPageRoot);

                     ^

Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\libraries\ESP8266WiFi 
Using library ESP8266WebServer at version 1.0 in folder: C:\Users\Dan\Desktop\arduino-1.6.8 esp git 2016-04-10\hardware\esp8266com\esp8266\libraries\ESP8266WebServer 
exit status 1
'WebPageRoot' was not declared in this scope
```

igrr, 2016-04-17:
Ok, this seems to be an arduino-ctags issue.

TheAustrian, 2016-04-17:
The sketch compiles for me if the function WebPageRoot() is declared in the sketch before it's called, like so often happens with 1.6.8 it seems.

igrr, 2016-04-17:
Ctags is used to scan sketch for functions which need forward declarations. Some things can cause it to break, in which case forward declarations are not generated. Personally I don't feel this is a major issue, but I'll report that on arduino-ctags repository.

devyte, 2017-10-17:
This issue is still valid in Arduino IDE 1.8.3, and the changelog up to 1.8.5 don't mention anything about it. However, it's not an issue with the core in this repo, but rather with the IDE, specifically with the ctags that is bundled with the IDE (version 5.8), which means there is nothing that can be done here. I suggest opening an issue in the Arduino IDE repo, or the Arduino ctags repo.
Closing.

closed 2017-10-17 by devyte