esp8266/Arduino #1932: Enable stats in lwIP config

igrr, 2016-04-18:
lwIP provides some stats which may be useful to benchmark library/network performance and do some troubleshooting.
[stats.h](https://github.com/esp8266/Arduino/blob/18f66e99693020a758b8f5ac92e042cbdecb7760/tools/sdk/lwip/include/lwip/stats.h)

- [ ] check which subset of useful stats we can enable without sacrificing too much RAM (e.g. ≤ 256 bytes).
- [ ] switch to GCC-built lwIP (#1926)
- [ ] enable relevant stats in lwipopts.h
- [ ] update prebuilt lwIP binary

devyte, 2017-10-17:
@igrr what is the status of this? If it's not done, maybe it should be pushed back?
Also, maybe it would make sense to make it a configurable option?

d-a-v, 2017-11-27:
@devyte, as suggested, I made some tests with lwip2.
@igrr what could be useful ?

in lwipopts.h:
```
#define LWIP_STATS 1
#define LWIP_STATS_LARGE 1 // enabled (32 bits counters instead of 16)
... // every stat type enabled, ```stats_display()``` function enabled

```
usage:
```
#include <lwip/stats.h>
...
stats_display();
```

Flash usage / Ram usage / free heap (a minute after sketch starts)

367812 / 40680 heap: 32168 (enabled + include + stats_display())
367056 / 40338 heap: 32456 (enabled + include)
367056 / 40338 heap: 32456 (enabled)
365852 / 39872 heap: 32976 (disabled)

to sum up:

enabled, no display: +466 ram  +1206 flash  -520 heap
enabled, stats_display() called: +808 ram  +1960 flash  -808 heap

The ```stats_display()```'s output may be debugged because block titles are not shown.
I put in (...) what it should be according to what's inside the stats_output function and to my sketch (2 tcp servers, 2 long-run running tcp echo clients, 1 repeated-short-run tcp echo client)
```
(tcp/udp)
xmit: 0
recv: 0
fw: 0
drop: 0
chkerr: 0
lenerr: 0
memerr: 0
rterr: 0
proterr: 0
opterr: 0
err: 0
cachehit: 0

(tcp/udp)
xmit: 3
recv: 2
fw: 0
drop: 10
chkerr: 0
lenerr: 0
memerr: 0
rterr: 0
proterr: 10
opterr: 0
err: 0
cachehit: 607482

(tcp/udp)
xmit: 607490
recv: 381944
fw: 0
drop: 20
chkerr: 0
lenerr: 0
memerr: 0
rterr: 0
proterr: 0
opterr: 0
err: 0
cachehit: 0

(igmp)
xmit: 5
recv: 0
drop: 0
chkerr: 0
lenerr: 0
memerr: 0
proterr: 0
rx_v1: 0
rx_group: 0
rx_general: 0
rx_report: 0
tx_join: 2
tx_leave: 0
tx_report: 3

(tcp/udp)
xmit: 0
recv: 0
fw: 0
drop: 0
chkerr: 0
lenerr: 0
memerr: 0
rterr: 0
proterr: 0
opterr: 0
err: 0
cachehit: 0

(tcp/udp)
xmit: 4
recv: 717
fw: 0
drop: 0
chkerr: 0
lenerr: 0
memerr: 0
rterr: 0
proterr: 0
opterr: 0
err: 0
cachehit: 3

(tcp/udp)
xmit: 302679
recv: 381207
fw: 0
drop: 0
chkerr: 0
lenerr: 0
memerr: 0
rterr: 0
proterr: 0
opterr: 0
err: 0
cachehit: 152553
```

devyte, 2017-12-18:
@d-a-v 

> enabled, no display: +466 ram +1206 flash -520 heap
> enabled, stats_display() called: +808 ram +1960 flash -808 heap
> 

That is significant heap usage just by enabling it. I suggest making an option in the IDE to enable/disable this, and have it disabled by default. If someone wants it, it's available, but let's avoid the mem usage impact for the rest.

d-a-v, 2018-01-07:
What about documenting this in tools/sdk/lwip2/README.md ?

devyte, 2018-01-08:
In that case, I think it would be better to have a formal documentation, i.e.: readthedocs. Something like an lwip customization section.

d-a-v, 2019-08-28:
Closing as the way to recompile lwIP is documented.
Users who know the existence of lwIP stats will find their way to enable and recompile for them.

closed 2019-08-28 by d-a-v