esp8266/Arduino #1933: https: certificate doesn't match, CI has failed

Alexey888, 2016-04-18:
ESP-12e, Core Version: 2.1.0

The problem with the certificate.Please, help.
Thank!

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char* ssid = "...";
const char* password = "...";

const char* host = "remoteboot.pro";
const int httpsPort = 443;

// Use web browser to view and copy
// SHA1 fingerprint of the certificate
const char* fingerprint = "2a 5b b5 29 ab 32 00 b2 fb 7d e3 fb b7 d9 af 98 44 32 60 58";

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  // Use WiFiClientSecure class to create TLS connection
  WiFiClientSecure client;
  Serial.print("connecting to ");
  Serial.println(host);
  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return;
  }

  if (client.verify(fingerprint, host)) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
  }

  String url = "/";
  Serial.print("requesting URL: ");
  Serial.println(url);

  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "User-Agent: BuildFailureDetectorESP8266\r\n" +
               "Connection: close\r\n\r\n");

  Serial.println("request sent");
  while (client.connected()) {
    String line = client.readStringUntil('\n');
    if (line == "\r") {
      Serial.println("headers received");
      break;
    }
  }
[... 570 lines cut ...]
request sent
:rd 5, 261, 0
:rdi 261, 5
:rd 256, 261, 5
:rdi 256, 256
:c0 256, 261
:wcs ra 211headers received
:rn 53
:rcl
:abort
:rd 5, 53, 0
:rdi 53, 5
:rd 48, 53, 5
:rdi 48, 48
:c0 48, 53
Alert: close notify
esp8266/Arduino CI has failed
reply was:
==========
BuildFailureDetectorESP8266
==========
closing connection

PORT CLOSED
```

riyono, 2016-04-18:
Might be related to #1285 
If there are more than one domain served by the server, as the library don't send SNI, the server might response with the wrong certificate (the default one).

Alexey888, 2016-04-18:
wireshark dump

Alexey888, 2016-04-18:
wireshark dump (delete ".txt")
[esp8266dump.pcapng.txt](https://github.com/esp8266/Arduino/files/224157/esp8266dump.pcapng.txt)

riyono, 2016-04-18:
[image]
@Alexey888, your server return a certificate for *.hosting.reg.ru (instead of the one for remoteboot.pro) when HTTPS doesn't send SNI, so, it surely related to #1285 (must wait until the library support SNI, or make sure server return the right certificate even when no SNI sent)

commit 8c65f2fc (2016-04-19): Update axTLS to fe4518d, SNI support in WiFiClientSecure (#1285)

Fixes #1933
closed 2016-06-23 by igrr