esp8266/Arduino #1934: Arduino IDE

PhilColbert, 2016-04-18:
Is there any reason all the options for debugging in the IDE just on the Generic ESP8266 option and not on the rest of the boards, ie on the Nodemcu boards?

Thanks

t3chguy, 2016-04-18:
All the other variants are just preconfigured versions of the "Generic ESP8266"
None of the boards feature USB/Serial on the Debug UART so that's why do not have a Debug option, if you're at the point where you're attaching USB/Serial to Debug, you'll probably have other settings tweaked too and will be using the "Generic ESP8266" which has no pre-chosen configuration values so to speak.

PhilColbert, 2016-04-18:
Thanks for the reply, sorry for the newbie questions, still learning here ! :)

closed 2016-05-10 by igrr