esp8266/Arduino #1937: Q: ESP8266/axTLS: peer certificates in SSL, and certificate verification?

capablegh, 2016-04-19:
Is there any means for an SSL client to provide it's certificate so that the server can validate the client's identity? I believe this capability is available within OpenSSL, not sure how heavy/large it is for the memory/flash available.

Also, just as an SSL connection is about to be established and the remote certificate is presented, can this certificate be validated against the local CA/chain?

igrr, 2016-04-19:
There is an open issue to document client side certificate support: #1743
Regarding server certificate verification, there is limited support for this: currently you can verify certificate SHA-1 fingerprint and subject name (or SAN). See Examples > ESP8266WiFi > HTTPSRequest for example.
There is an open issue to add support for validation against local CA/chain: #1851.

closed 2016-04-19 by igrr

capablegh, 2016-04-19:
Thanks for the quick response, contributions, and the efforts!