esp8266/Arduino #1939: Add save, undo and redo buttons to SDWebServer example

gmag11, 2016-04-19:
### Basic Infos
I'm using [ACE editor](https://ace.c9.io/) in my project [https://github.com/gmag11/FSBrowser](https://github.com/gmag11/FSBrowser).
This example uses CTRL+S, +Z and +CAPS+Z to trigger save, undo and redo commands, what is great but makes difficult to use in mobile plaforms. I'd like to add this commands as buttons too.
Undo and redo seems to be easy as there are direct API calls in ACE editor code. But for save I don't know how to get file name to be uploaded to ESP.
Notice I'm not a javascript expert at all, just learning.

#### Hardware
Hardware:			ESP-12
Core Version:      	Current GitHub master branch

### Code
I've added a `Save` button using this three lines to `createFileUploader()`
```javascript
var savefile = document.createElement("button");
savefile.innerHTML = 'Save';
document.getElementById(element).appendChild(savefile);
```
Callback funcion for `CTRL+S` is 
```javascript
editor.commands.addCommand({
    name: 'saveCommand',
    bindKey: {win: 'Ctrl-S',  mac: 'Command-S'},
    exec: function(editor) {
        httpPost(file, editor.getValue()+"", type);
    },
    readOnly: false
});
```
And callback function for savefile button is
```javascript
savefile.onclick = function (e) {
          httpPost(file, editor.getValue()+"", type);
        };
```
Here neither file neither type are undefined. I've searched all parameters in Chrome debugget but have not found where to get file and type. May you give me a clue about this?

gmag11, 2016-04-19:
I'll try to use `editor.command.exec('saveCommand')` to onClick button event. This evening...

gmag11, 2016-04-20:
I've used this code to add buttons:
```javascript
var saveFile = document.createElement("button");
saveFile.innerHTML = 'Save';
document.getElementById(element).appendChild(saveFile);
var undoButton = document.createElement("button");
undoButton.innerHTML = 'Undo';
document.getElementById(element).appendChild(undoButton);
var redoButton = document.createElement("button");
redoButton.innerHTML = 'Redo';
document.getElementById(element).appendChild(redoButton);
```
and this to assign `onClick` actions:
```javascript
saveFile.onclick = function (e) {
    editor.execCommand("saveCommand");
};
undoButton.onclick = function (e) {
    editor.getSession().getUndoManager().undo(false);
};
redoButton.onclick = function (e) {
    editor.getSession().getUndoManager().redo(false);
};
```
It works!!!

closed 2016-04-20 by gmag11

ldijkman, 2022-02-17:
hello 

do you maybe also know how to make it a tryit alike editor with html preview div or iframe
and an exit button to index.html or url/ or ip/

https://github.com/ldijkman/randomnerd_esp32_wifi_manager/tree/main/ESP8266-LittleFS-wifi-manager/ESP8266-LittleFS-wifi-manager_with_editor

[image]
e

mathertel, 2022-02-17:
Regarding an Editor:
Have look into the enriched example in 
esp8266com\esp8266\libraries\ESP8266WebServer\examples\WebServer (not yet published). The UI fits into the LittleFS.