esp8266/Arduino #1943: SPI API Missing Interrupts

biscuitWizard, 2016-04-19:
### Basic Infos

#### Hardware
Hardware:		ESP-12
Core Version:      	2.1.0-rc2

### Description

The SPI implementation included in the ESP8266 is missing hardware interrupts. The official API lists this function as:

https://www.arduino.cc/en/Reference/SPIusingInterrupt

Without this, one of my libraries is unable to function.

### Debug Messages

```

```

me-no-dev, 2016-04-21:
the library needs to be made compatible with ESP8266. I have used VS1053 just fine with my ESPs.
From what I can tell, it thinks that it's compiling for AVR Arduino.

devyte, 2017-10-17:
It seems that the library has been updated in the meantime to support ESP8266. Closing.

closed 2017-10-17 by devyte