esp8266/Arduino #1944: Memory leak in ESP8266HttpClient only when using SSL/TLS

flyinggorilla, 2016-04-19:
#### Hardware
Arduino IDE 1.6.8
Hardware:		Adafruit Huzzah
Core Version:      	2.2.0

### Description
ESP8266HTTPClient works fine in "http" mode. no leak. But when enabling SSL, it leaks 65 to 100 bytes per request (seems to depend on website how big the leak is). I tested different websites and on different Wifi connections. Other tests ran memory down from 30k free to 5k free after .GET() until it crashed. I even tried your latest libaxtls.a from master - since there were some recent fixes after 2.2.0, but no success.

Note: i tried HTTP keep-alive by also running it with .setReuse(true) following the "ReuseConnection" example code, with similar leak results.

Thanks for looking into this.
Best
flyinggorilla

### Settings in IDE
Module:  			Adafruit Huzzah
Flash Size: 		4MB (3MB spiffs)
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck 

### Sketch

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

void WiFiEvent(WiFiEvent_t event) {
    switch(event) {
        case WIFI_EVENT_STAMODE_GOT_IP:
            Serial.println("WiFi connected. IP address: " + String(WiFi.localIP().toString()) + " hostname: "+ WiFi.hostname() + "  SSID: " + WiFi.SSID());
            break;
        case WIFI_EVENT_STAMODE_DISCONNECTED:
            Serial.println("WiFi client lost connection");
            break;
        case WIFI_EVENT_STAMODE_CONNECTED:
            Serial.println("WiFi client connected");
            break;
        case WIFI_EVENT_STAMODE_AUTHMODE_CHANGE:
            Serial.println("WiFi client authentication mode changed.");
            break;
        //case WIFI_STAMODE_DHCP_TIMEOUT:                             THIS IS A NEW CONSTANT ENABLE WITH UPDATED SDK
          //  Serial.println("WiFi client DHCP timeout reached.");
            //break;
        case WIFI_EVENT_SOFTAPMODE_STACONNECTED:
            Serial.println("WiFi accesspoint: new client connected. Clients: "  + String(WiFi.softAPgetStationNum()));
             break;
        case WIFI_EVENT_SOFTAPMODE_STADISCONNECTED:
            Serial.println("WiFi accesspoint: client disconnected. Clients: " + String(WiFi.softAPgetStationNum()));
             break;
        case WIFI_EVENT_SOFTAPMODE_PROBEREQRECVED:
            //Serial.println("WiFi accesspoint: probe request received.");
            break; 
      }
}

// HTTPClient http;

// initialization routines
void setup ( void ) {
  int serialtimeout = 5000; //ms
  Serial.begin ( 115200 );
  while(!Serial) {
    if (serialtimeout >0) {
      serialtimeout -= 50; 
    } else { 
      break;
[... 243 lines cut ...]
Decoding 18 results
0x4022f60d: tcp_write at ?? line ?
0x40100000: _stext at ?? line ?
0x402039ab: ClientContext::write(char const*, unsigned int) at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src\include/ClientContext.h line 226
:  (inlined by) ax_port_write at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 540
0x40220c64: send_raw_packet at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 979
:  (inlined by) send_packet at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1112
0x402212e8: basic_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1240
0x40221142: DISPLAY_ALERT at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 2182 (discriminator 3)
:  (inlined by) send_alert at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1597 (discriminator 3)
0x402216d2: ssl_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 262
0x4022119e: ssl_free at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 236
0x40203396: ~SSLContext at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 497
:  (inlined by) SSLContext::unref() at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 92
0x40203ac0: WiFiClientSecure::stop() at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 363
0x40203c1a: HTTPClient::end() at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 192
0x402044ea: HTTPClient::writeToStream(Stream*) at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 192
0x4020519a: String::String(char const*) at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x4020455c: HTTPClient::getString() at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 192
0x4010020c: _umm_free at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1285
0x40202524: loop at C:\Users\Bernd\Arduino\TestHTTPS/TestHTTPS.ino line 96
0x40205a84: loop_wrapper at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/cont.S line 109

```

flyinggorilla, 2016-04-19:
I compiled it now with the "Generic ESP8266" board and enabled debug settings:

```
ESP8266 SDK Version: 1.5.2(7eee54f4)
Connecting to Wifi SSID: ssid as host ESP_0158F0
heap at begin: 39040
Wifi Station not yet connected.
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (4)
reconnect
f 0, WiFi client lost connection
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with ssid, channel 6
dhcp client start...
WiFi client connected
chg_B1:-40
ip:10.10.129.129,mask:255.255.255.0,gw:10.10.129.1
WiFi connected. IP address: 10.10.129.129 hostname: ESP_0158F0  SSID: ssid
[HTTP] begin...
[HTTP-Client][begin] url: https://www.herold.at/
[HTTP-Client][begin] host: www.herold.at port: 443 url: /
[HTTP-Client][begin] httpsFingerprint: BE:70:4E:41:10:D9:2E:C9:37:F4:49:BB:06:CF:E6:7D:82:CD:47:1A
[HTTP] GET...
please start sntp first !
State:	sending Client Hello (1)
State:	receiving Server Hello (2)
State:	receiving Certificate (11)
State:	receiving Server Hello Done (14)
State:	sending Client Key Exchange (16)
State:	sending Finished (16)
State:	receiving Finished (16)
[HTTP-Client] connected to www.herold.at:443
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Tue, 19 Apr 2016 21:53:01 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Set-Cookie: JSESSIONID=35718D375E9A5FFA46051EB92F262968; Path=/SP/; HttpOnly'
[HTTP-Client][handleHeaderResponse] RX: 'X-XSS-Protection: 0'
[HTTP-Client][handleHeaderResponse] RX: 'Set-Cookie: hbd-scn=%3CHBD%3E%3CSESSION_ID%3EPWA05_14611027817400101%3C%2FSESSION_ID%3E%3C%2FHBD%3E; Path=/'
[HTTP-Client][handleHeaderResponse] RX: 'Set-Cookie: gso-cookie=%3CHBD%3E%3CLANGUAGE%3E0%3C%2FLANGUAGE%3E%3CCREATE_DATE%3E2016-04-19+23%3A53%3A01%3C%2FCREATE_DATE%3E%3CLAST_ACCESS%3E2016-04-19+23%3A53%3A01%3C%2FLAST_ACCESS%3E%3CCLIENT_ID%3E14611027817490101%3C%2FCLIENT_ID%3E%3C%2FHBD%3E; Expires=Tue, 24-May-2016 15:13:01 GMT; Path=/'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html;charset=UTF-8'
[HTTP-Client][handleHeaderResponse] RX: 'Vary: Accept-Encoding'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Transfer-Encoding: chunked'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] Transfer-Encoding: chunked
heap after GET: 20216
[HTTP] GET... code: 200
[HTTP-Client] read chunk len: 8184
[HTTP-Client][writeToStreamDataBlock] connection closed or file end (written: 8184).
[HTTP-Client] read chunk len: 0
[HTTP-Client][end] still data in buffer (1360), clean up.
[HTTP-Client][end] tcp stop
Alert: close notify
<!DOCTYPE html><html xmlns="http://www.w3.org/1999/xhtml" xmlns:...
[HTTP-Client][end] tcp is closed
[HTTP] begin...
[HTTP-Client][begin] url: https://www.herold.at/
[HTTP-Client][begin] host: www.herold.at port: 443 url: /
[HTTP-Client][begin] httpsFingerprint: BE:70:4E:41:10:D9:2E:C9:37:F4:49:BB:06:CF:E6:7D:82:CD:47:1A
[HTTP] GET...
please start sntp first !
[... 434 lines cut ...]
0x4022f855: tcp_write at ?? line ?
0x40100000: _stext at ?? line ?
0x402039ab: ClientContext::write(char const*, unsigned int) at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src\include/ClientContext.h line 226
:  (inlined by) ax_port_write at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 540
0x40220eac: send_raw_packet at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 979
:  (inlined by) send_packet at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1112
0x40204f3c: Print::printf(char const*, ...) at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 84
0x4022138a: DISPLAY_ALERT at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 2182 (discriminator 3)
:  (inlined by) send_alert at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1597 (discriminator 3)
0x4022191a: ssl_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 262
0x402213e6: ssl_free at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 236
0x40203396: ~SSLContext at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 497
:  (inlined by) SSLContext::unref() at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 92
0x40203ac0: WiFiClientSecure::stop() at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 363
0x40203c5d: HTTPClient::end() at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 192
0x40204697: HTTPClient::writeToStream(Stream*) at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 192
0x40205389: String::copy(char const*, unsigned int) at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402053da: String::String(char const*) at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
[... cut ...]
0x4010020c: _umm_free at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1285
0x40202524: loop at C:\Users\Bernd\Arduino\TestHTTPS/TestHTTPS.ino line 96
0x40205ccc: loop_wrapper at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100958: cont_norm at C:\Users\Bernd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/cont.S line 109

```

flyinggorilla, 2016-04-19:
And finally results with the "Core + SSL + TLSmem" debug options....

already at the 4th .GET() we have memory issues showing up in the debug output
```
ssl->need_bytes=7344 > 6859
ssl/tls1.c:1425 malloc 17408 failed, left 29680
ssl/tls1.c:1425 malloc 17408, left 29680
failed to grow plain buffer
```

now here is the full output...

```
SP8266 SDK Version: 1.5.2(7eee54f4)
Connecting to Wifi SSID: ssidas host ESP_0158F0
heap at begin: 40416
Wifi Station not yet connected.
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 
chg_B1:-40

connected with ssid, channel 6
dhcp client start...
WiFi client connected
ip:10.10.129.129,mask:255.255.255.0,gw:10.10.129.1
WiFi connected. IP address: 10.10.129.129 hostname: ESP_0158F0  SSID: ssid
[HTTP] begin...
[HTTP] GET...
:ref 1
ssl/tls1.c:541 malloc 6864, left 32280
please start sntp first !
State:	sending Client Hello (1)
:wr
:sent 52
:rn 1460
:ww
:rd 5, 1460, 0
:rdi 1460, 5
:rd 74, 1460, 5
:rdi 1455, 74
State:	receiving Server Hello (2)
:rd 5, 1460, 79
:rdi 1381, 5
:rd 1376, 1460, 84
:rdi 1376, 1376
:c0 1376, 1460
:rn 2259
:rd 2250, 2259, 0
:rdi 1460, 1460
:c 1460, 1460, 2259
:rdi 799, 790
State:	receiving Certificate (11)
crypto/bigint.c:1072 realloc 1032, left 27904
crypto/bigint.c:1072 realloc 1032, left 24432
crypto/bigint.c:1072 realloc 1032, left 20952
:rd 5, 799, 790
:rdi 9, 5
:rd 4, 799, 795
:rdi 4, 4
:c0 4, 799
State:	receiving Server Hello Done (14)
crypto/bigint.c:1072 realloc 1024, left 18000
State:	sending Client Key Exchange (16)
:wr
:sent 267
[... 723 lines cut ...]
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
:ww
:wr
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
:ww
:wr
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
:ww
:wr
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
```

jpmens, 2016-04-23:
I can confirm this using either `GET` or `POST`. I notice a leak of 72 octets (obtained with `ESP.getFreeHeap()`)  for the site I'm accessing. I can also confirm that there's no leak (neither with `GET` nor with `POST`) when using plain HTTP (i.e. non-TLS).

jpmens, 2016-04-23:
The JSON payload shown here demonstrates the free heap size remaining (`_heap`) until the device then crashes.

[image]

djoele, 2016-04-25:
I notice the same results with latest Git checkout. After each request 72 bytes are not freed.

Will try some combinations of reUse and timeouts in the HTTP request. Perhaps that could help.

igrr, 2016-04-25:
The issue is due to a missing virtual destructor in the `TransportTraits` class.

djoele, 2016-04-26:
thank you for the quick fix. Tonight I will check to see if the memory leak has been resolved. Will let you know the results.

djoele, 2016-04-27:
@igrr: I can confirm that the latest Git checkout resolves the memory leak in https requests.

@others: does it work for you too?

As far as I'm concerned the issue can be closed.

flyinggorilla, 2016-05-17:
i can confirm the fix too. tested with 2.3.0-rc1.

closed 2016-05-17 by flyinggorilla

lenguyenvu007, 2017-03-01:
Hi Igrr, 
I got the same error "please start sntp first !" when using Firebase with Arduino 1.8.1 with Stable version 2.3.0, as below description.
Would you please advice me if any information for debug this error?
Thank you in advance.

please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !
please start sntp first !

Soft WDT reset

ctx: cont 
sp: 3fff2180 end: 4000050c offset: 01b0

>>>stack>>>
3fff2330:  3fff0e70 00000000 00000000 3fff9f64  
3fff2340:  00000000 0ba81e4a 402012d8 00001388  
3fff2350:  3fff1970 00000045 00000000 00000000  
3fff2360:  00000045 00000000 3fff3f34 4020b4be  
3fff2370:  00000045 1393025a 00000010 00000000  
3fff2380:  00000000 3fff4344 00000045 4022e468  
3fff2390:  00020316 00000010 00000000 00000000  
3fff23a0:  00000010 00000016 00000010 8a6063f4  
3fff23b0:  00000001 3fff86d4 00000010 00000004  
3fff23c0:  00000004 00000000 3fff4344 4022e615  
3fff23d0:  0c000014 8a6a6b19 167ada80 45ae3c45  
3fff23e0:  00000004 00000000 3fff4344 4022f3f8  
3fff23f0:  98a27cd6 ae839883 94555c13 3fff3f34  
3fff2400:  00000000 00000004 00000004 4020adfb  
3fff2410:  3fff461d 3fff4619 3fff4414 00000bc0  
3fff2420:  00000c1b 3fff5230 00000003 00000004  
3fff2430:  3fff4344 3fff4619 00000004 00000004  
3fff2440:  00000004 3fff4619 3fff4344 4022edd8  
3fff2450:  00000000 3fff4619 3fff4344 4022f180  
3fff2460:  0000001e 3fff4584 0000001f 401004d8  
3fff2470:  3fff24c0 0000000e 00000010 00000000  
3fff2480:  3fff3fbc 3fff4584 3fff4344 01000000  
3fff2490:  3fff24c0 3fff4564 3fff4344 4022ef30  
3fff24a0:  4020ac0c 00000000 3fff1b00 00001387  
3fff24b0:  0009ec1e 3fff4564 3fff3eec 4020b29d  
3fff24c0:  000001bb 3fff85d4 3fff4564 4020a632  
3fff24d0:  0dfac568 00000000 3fff34ec 3fff1934  
3fff24e0:  000001bb 3fff4564 3fff4584 4020b5f1  
3fff24f0:  3ffea7a8 0dfac568 3ffea7a8 0dfac568  
3fff2500:  3fff3ff4 00000000 3fff3ff8 4020d834  
3fff2510:  00000000 00000a2e 00000a2e 40211aa4  
3fff2520:  fffffff4 00000000 3fff3ff8 4020e004  
3fff2530:  3ffea2b8 00000053 3ffe9fc2 4020f65b  
3fff2540:  3fff85bc 3fff25b0 3fff3ff4 4020f6b4  
3fff2550:  3fff85c8 00000000 00000000 3ffe9fbf  
3fff2560:  3fff3ff4 3fff2590 3ffe9fbf 3ffe9fbf  
3fff2570:  3fff3ff4 3fff1934 3fff2650 402100c6  
3fff2580:  3fff1918 3fff1934 3fff2650 4020ec6b  
3fff2590:  3fff85c8 00000000 0ccccccc 00000009  
3fff25a0:  00000019 00000001 00000000 3fff67d1  
3fff25b0:  3fff8778 00000001 00000000 0000001f  
3fff25c0:  00000000 3fff265c 3fff2610 3fff2678  
3fff25d0:  402100b0 3fff2610 3ffe8fda 4020f69c  
3fff25e0:  3fff2614 3fff67cf 3fff2630 3fff1914  
3fff25f0:  3fff1918 3fff2678 3fff2650 4020ee97  
3fff2600:  3fff3ff4 3fff2690 00000000 4010053d  
[... 130 lines cut ...]
3fff4680:  23520b92 71569c5d 25d74aad 07a2c506  
3fff4690:  72da64e4 cf8ef52a 1960b503 7cc68975  
3fff46a0:  eb0a650d 245c27bc bfde1606 b71a8113  
3fff46b0:  8d35c41f 79087f52 854138d5 ac69bd13  
3fff46c0:  8de18f00 0c206a47 1b157c0:  4cd8a1bc b643a8f8 214de903 da880807  
3fff4fd0:  39662f58 7805bd29 e8389d8b 7e6ab705  
3fff4fe0:  c4e6a471 efb0a660 2889e480 d6259e0f  
3fff4ff0:  adf383ed 98c791a6 351842c9 98ad9d14  
3fff5000:  4f2e9246 4387f1ca 579516c1 89ef502d  
3fff5010:  577a802d 5feef2ad 8d00d26b 14f814b9  
3fff5020:  c0d93515 727ba346 c9bf91c8 d0cd2b55  
3fff5030:  269c3e97 cedfcc64 ca711983 d5d4e64e  
3fff5040:  cd19a97b ecc8de55 53385ed2 8c4f5ce5  
3fff5050:  2350fe2d e666fc36 39a48ecb 95b70019  
3fff5060:  0b913902 2e38fe0e 9a051dd1 6f3e4df6  
3fff5070:  af1d070f 608f1e2c 36fae239 d4391353  
3fff5080:  db2b265e bd14a83d 0318eb32 71045228  
3fff5090:  3d33abe5 07bb38e1 9c628436 3016ea79  
3fff50a0:  2bc05ff4 e46b71e8 010302f9 81a30100  
3fff50b0:  ed8130f0 03061f30 04231d55 80163018  
3fff50c0:  68e64814 b2d22bf9 d847d795 4f102023  
3fff50d0:  9f909833 061d30d4 0e1d5503 14041604  
3fff50e0:  68987ac0 abfb898d 110c6405 657daa7d  
3fff50f0:  4ecccab8 03060f30 01131d55 0504ff01  
3fff5100:  01010330 060e30

djoele, 2018-09-24:
On May 17th 2016 it was confirmed fixed by the second person.

The comment on March 1st 2017: I guess this is still open.

was this ever confirmed and fixed?