esp8266/Arduino #1950: Fatal exception 0(IllegalInstructionCause):

AlexM600, 2016-04-20:
### Basic Infos
I'm trying to use ESP-01 module as small webserver.

#### Hardware
Hardware:			ESP-01
Core Version:      	2.2.0

### Description

It works on my home AP, on my Android phone AP.
But has Fatal Exception in 10-20 seconds on my work AP. Tried to set different settings on it, change channels, mode, speed, disable encryption and so on.. ESP crashes 
Other WiFi clients work fine.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Reset Method:		ck

### Sketch
Standard example from libraries/ESP8266WebServer/examples/HelloServer/HelloServer.ino
Just changed ssid
### Debug Messages

wifi evt: 0
..wifi evt: 7
wifi evt: 7
.wifi evt: 7
wifi evt: 7
..wifi evt: 3
.
Connected to GOR30
IP address: 192.168.10.20
MDNS responder started
HTTP server started
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
Fatal exception 0(IllegalInstructionCause):
epc1=0x40106992, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (0):
epc1=0x40106992 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd80 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff20:  40226457 d334fe18 4021024c 3ffed430  
3fffff30:  3fff256c 00000001 40226496 40210265  
3fffff40:  402262a5 3fff2794 3fffff80 3fff1dac  
3fffff50:  3ffe0000 3fff256c 3ffef780 40226c94  
3fffff60:  3fff2794 3fff1dac 3ffeaf26 402259f7  
3fffff70:  3fff1dac 00000014 40225fa2 3fff2794  
3fffff80:  3fff1dac 3fffdc80 3fff1e14 3fff0460  
3fffff90:  4021d93b 3fff2794 00000000 4020767f  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

liquidfalcon, 2016-04-20:
Can you comment out MDNS, and see if it runs? There's a known bug with it causing crashes just like this, see issue #1926

AlexM600, 2016-04-27:
Thanks liquidfalcon!
With commented out MDNS it runs w/o exceptions.

closed 2016-04-27 by AlexM600

blastervla, 2016-11-04:
Hi! I wanted to ask you if there is another reason as to which the ESP could be behaving like this. You see, I have a black ESP8266-01 and it will reboot and print a stack trace just after showing this exact same exception. Nevertheless, I do not use MDNS in my program. Does anybody know what could be happening?

liquidfalcon, 2016-11-04:
This sounds like a separate issue, @blastervla - Could you create a new issue, fill out the information, and post a decoded stack trace? That should help us figure it out

wateras, 2017-04-01:
my problem :
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2408, room 16 
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user2 @ 81000

Fatal exception 0(IllegalInstructionCause):
epc1=0x40102687, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

RAlexeev, 2018-06-29:
I have the same problem with my sketch. If I turn on debug mode, then "Fatal exception 0(IllegalInstructionCause)" occurs. I can comment out one of these lines "WiFi.persistent(false);", "wifiDisconnect(true);" or "MDNS.begin("my_esp_hostname");" and everything executes without exceptions. Also I can turn off debug mode, in this case exceptions will disappear. 
This error occurs only with one specific Wi-Fi router Zyxel Keenetic LTE, if Acces Point is not open, but no with two another, which I've tried. SSID and psk don't matter. What is wrong with my sketch? I've tried to execute it on two different ESP8266.

```
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>

void wifiDisconnect(bool wifioff) {
  switch (WiFi.getMode()) {
    case WIFI_STA:
      delay(100);
      WiFi.disconnect(wifioff);
      delay(100);
      break;
    case WIFI_AP:
      delay(100);
      WiFi.softAPdisconnect(wifioff);
      delay(100);
      break;
    case WIFI_AP_STA:
      delay(100);
      WiFi.disconnect(wifioff);
      delay(100);
      WiFi.softAPdisconnect(wifioff);
      delay(100);
      break;
  }
}

void setup(){
  Serial.begin(115200);
  delay(1000);

  WiFi.persistent(false);

  wifiDisconnect(true);

  WiFi.mode(WIFI_STA);
  WiFi.hostname("my_esp_hostname");
  WiFi.begin("MYWIFISSID", "MYWIFIPSK");
  for (uint8_t tries = 0; tries < 30; tries++) {
    if (WiFi.isConnected()) {
      break;
    }
    else {
      delay(500);
    }
  }

  MDNS.begin("my_esp_hostname");
}

void loop() { }
```

Serial monitor output in debug mode:
```
SDK:2.2.1(cfd48f3)/Core:2.4.1-102-g74819a76/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:94e9704
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
bcn 0
del if1
usl
mode : null
wifi evt: 8
mode : sta(dc:4f:22:18:14:58)
[... 88 lines cut ...]
0x4020c26b: ethernet_output at netif/ethernet.c line 305
0x4020c274: ethernet_output at netif/ethernet.c line 305
0x40211ded: etharp_output_LWIP2 at core/ipv4/etharp.c line 893
0x40212874: ip4_output_if_opt_src at core/ipv4/ip4.c line 962
0x40212ee4: mem_malloc at core/mem.c line 136
0x402128b4: ip4_output_if_opt at core/ipv4/ip4.c line 788
0x40212045: igmp_send at core/ipv4/igmp.c line 793
0x4020c2c8: do_memp_malloc_pool at core/memp.c line 301
0x4020c304: memp_malloc at core/memp.c line 404
0x40212116: igmp_lookup_group at core/ipv4/igmp.c line 258
0x402052f8: operator new(unsigned int) at /opt/arduino-1.8.5/hardware/esp8266com/esp8266/cores/esp8266/abi.cpp line 34
0x402122ef: igmp_joingroup_netif at core/ipv4/igmp.c line 694
0x4021237b: igmp_joingroup at core/ipv4/igmp.c line 464
0x4020359a: MDNSResponder::_listen() at /opt/arduino-1.8.5/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp line 198
0x401009e4: free at /opt/arduino-1.8.5/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1755
0x40205350: operator delete(void*) at /opt/arduino-1.8.5/hardware/esp8266com/esp8266/cores/esp8266/abi.cpp line 54
0x402052f8: operator new(unsigned int) at /opt/arduino-1.8.5/hardware/esp8266com/esp8266/cores/esp8266/abi.cpp line 34
0x40204892: MDNSResponder::begin(char const*) at /opt/arduino-1.8.5/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp line 174
0x40202f28: ESP8266WiFiSTAClass::status() at /opt/arduino-1.8.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 498
0x40203250: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data &, const std::_Any_data &, std::_Manager_operation) at /opt/arduino-1.8.5/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional line 1931
0x402036bc: std::_Function_handler ::_M_invoke(const std::_Any_data &, const WiFiEventStationModeDisconnected &) at /opt/arduino-1.8.5/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp line 171
0x402030e4: ESP8266WiFiSTAClass::isConnected() at /opt/arduino-1.8.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 324
0x40202771: setup() at /home/roman/Arduino/sketch_mdns_fatal_exception/sketch_mdns_fatal_exception.ino line 48
0x402054a0: loop_wrapper() at /opt/arduino-1.8.5/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 122
```

popok75, 2019-03-31:
Maybe it helps somebody, because I was breaking my head for hours on this crash. It seems providing temporary variables for WiFi.begin is a bad idea, it makes MDNS.begin to crash, at least in my case, I think in RAlexeev case too. As if Wifi is storing the pointers provided and MDNS is reusing them on begin, after the strings has been deallocated.  I moved my pointers to point on static variables, it works perfectly.