esp8266/Arduino #1951: Arduino IDE does not load the manager

zeroprime790, 2016-04-20:
Hello, 

I'm using this board manager the last few months and today a board manager of the arduino IDE upgraded and the esp8266 board manager disappeared. I checked the link in preferences and everything is on place. I tried also to uninstall and reinstall the IDE without any luck.

Currently i'm running in Arduino IDE 1.6.8 and the Arduino AVR Boards upgraded to v1.6.10

Dom1n1c, 2016-04-20:
I can confirm that issue, the IDE is trying to contact arduino.esp8266.com but it fails.

mcauser, 2016-04-20:
I am unable to reach `http://arduino.esp8266.com/stable/package_esp8266com_index.json` from several computers and networks.

Networking issue?

`$ host arduino.esp8266.com`
`arduino.esp8266.com has address 104.131.82.128`

mcauser, 2016-04-20:
http://www.downforeveryoneorjustme.com/http://arduino.esp8266.com/stable/package_esp8266com_index.json

> It's not just you! http://arduino.esp8266.com looks down from here.

rasendubi, 2016-04-20:
Confirm network issue. http://arduino.esp2866.com is down for me.

mcauser, 2016-04-20:
The .json file is linked in the footer here: https://github.com/esp8266/Arduino/releases/tag/2.2.0
Unfortunately, urls inside also refer to arduino.esp8266.com

monkeyBone, 2016-04-20:
Still confirmed. Are there any alternates?

cfrodca, 2016-04-20:
http://arduino.esp8266.com is up!

zeroprime790, 2016-04-20:
yes, I can confirm that. But since the manager was installed how it is related with the URL and became unavailable?

closed 2016-04-20 by zeroprime790