esp8266/Arduino #1952: Stream ArduinoJson to with ESP266WebServer

hallard, 2016-04-20:
Hi there

I recently moved to ArduinoJson library to do my json stuff. This lib is just awesome. But I have an optimization question, this lib can stream the json data to Serial with for example
```arduino
root.prettyPrintTo(Serial);
```
but I would like to direct stream this to client browser, today I'm copying json buffer to string like this but it's too much overhead (twice buffer size) plus copy time
```arduino
  String myString;
  root.prettyPrintTo(myString);
  server.send ( 200, "text/json", myString );
```

any way to do something that could be like this 
```arduino
  server.send ( 200, "text/json");
  root.prettyPrintTo(server.client);
```
thanks for any advice

hallard, 2016-04-20:
I think something similar to this can do the trick, but once again we need to declare a buffer.
https://github.com/bblanchon/ArduinoJson/issues/166

me-no-dev, 2016-04-21:
there is this: https://github.com/me-no-dev/ESPAsyncWebServer#arduinojson-basic-response

hallard, 2016-04-26:
@me-no-dev 
thanks for this, I did not switched to async TCP yet, is it considered as "stable" ?

kaeferfreund, 2016-04-26:
works very well for me. Have 5 devices running 24/7, with no prob.
It's so incredibly fast... Loving it 💃

me-no-dev, 2016-04-26:
it is really stable at this point :) maybe it's time to change the readme for the TCP lib :D

hallard, 2016-04-26:
Guys, are you testing me, you want me to spend my whole remaining day to integrate async ?!?
No, no, you got other things to do, no, no don't do that ;-)
ok see that tomorrow or this night, good deal ;-)

kaeferfreund, 2016-04-26:
Its totally worth it, seriously! :D

Have fun ;)

me-no-dev, 2016-04-26:
Should not be that much work :) it's mostly the same. The rest is explained in the README (make sure you read it through)

hallard, 2016-04-26:
Guess what, I'm also using WebSocket, I will try all of this ASAP, need to take some time on the README, but looks nice ;-)

me-no-dev, 2016-04-26:
try the included example first :) it has the spiffs editor and websocket example onboard (with the html and all)

devyte, 2017-10-17:
The buffered response is the way to go. If that is not enough, then a streaming json lib is needed.
Closing per #3655 .

closed 2017-10-17 by devyte