esp8266/Arduino #1954: undefined reference to `vasprintf'

joshglendenning, 2016-04-20:
#### Hardware
PlatformIO, version 2.8.6
Hardware:	ESP-01
Sparkfun ESP8266 "Thing"

### Description
`vasprintf` seems to be undefined- I understand this is a GNU extension and not ubiquitous, but I was wondering if this was some error on my part or if I'll have to implement this myself. The only thing I'm including in my source is `#include "Arduino.h"`. Any help is appreciated.

WereCatf, 2016-04-21:
`vasprintf` seems to be defined in stdio.h, include that in your code.

joshglendenning, 2016-04-21:
I've tried both `#include "stdio.h"` and `#include <stdio.h>` and I still get the same error.

WereCatf, 2016-04-21:
You're not using the Arduino IDE to compile with, are you?

joshglendenning, 2016-04-21:
Naw, I'm using PlatformIO. Here's my platformio.ini:
```#
# Project Configuration File
#
# A detailed documentation with the EXAMPLES is located here:
# http://docs.platformio.org/en/latest/projectconf.html
#

# A sign `#` at the beginning of the line indicates a comment
# Comment lines are ignored.

# Simple and base environment
# [env:mybaseenv]
# platform = %INSTALLED_PLATFORM_NAME_HERE%
# framework =
# board =
#
# Automatic targets - enable auto-uploading
# targets = upload

[env:esp01]
platform = espressif
framework = arduino
board = esp01
targets = upload
```

WereCatf, 2016-04-21:
Well, I have no effing idea about PlatformIO, but your include-paths may be wrong. vasprintf is defined in the Espressif SDK's stdio.h, but if your code complains about it being missing then you're obviously using some other stdio.h.

If you're on Linux or you have installed grep under Windows you could always try and do a `grep "vasprintf" -R` in your codebase's directory and seeing if it comes up in some other header-file instead.

joshglendenning, 2016-04-21:
I'll try digging around in PlatformIO and see what I can find.

joshglendenning, 2016-04-21:
`grep -R` wasn't able to find anything for `vasprintf` in the Espressif SDK or the Arduino framework, but I did find references to `sprintf`/`vsprintf` as `ets_vsprintf`. Do I have to use that? I also wasn't able to find any sort of `stdio.h`.

WereCatf, 2016-04-21:
Well, as I said, I don't know 'nuffin 'bout PlatformIO. Just gotta hope someone else comes along who does.

joshglendenning, 2016-04-21:
Thanks for the help anyway!

joshglendenning, 2016-04-21:
It should be noted that I also can't use `vsprintf`.

joshglendenning, 2016-04-24:
After consulting the esp8266 community forum I was able to find a solution:
http://www.esp8266.com/viewtopic.php?f=8&t=9638

Apparently `vsprintf` is not defined, but `vsnprintf` is, so I used that, with a few caveats. Here is what I ended up implementing to replace `vasprintf`:

```c++
		int vasprintf(char** strp, const char* fmt, va_list ap) {
			va_list ap2;
			va_copy(ap2, ap);
			char tmp[1];
			int size = vsnprintf(tmp, 1, fmt, ap2);
			if (size <= 0) return size;
			va_end(ap2);
			size += 1;
			*strp = (char*)malloc(size * sizeof(char));
			return vsnprintf(*strp, size, fmt, ap);
		}
```

closed 2016-04-24 by joshglendenning