esp8266/Arduino #1955: esp8266 board not work with arduino 1.6.8

dkumariet, 2016-04-21:
today auto removed esp8266 board from my arduino...........
then i try the steps according this link
https://github.com/esp8266/Arduino
but i occur some error
"git clone https://github.com/esp8266/Arduino.git esp8266"
this cmd not work, this msg show on terminal
fatal: unable to access 'https://github.com/esp8266/Arduino.git/': Failed to connect to github.com port 443: Network is unreachable
and,arduino show msg

WARNING: Error loading hardware folder /opt/arduino-1.6.8/hardware/esp8266com
No valid hardware definitions found in folder esp8266com.
please help me

re3lex, 2016-04-21:
had the same yesterday.
not sure why Arduino IDE "forgot" ESP8266 stuff but I faced similar issue with android.esp8266.com inaccessibility  so was not able to setup arduino ide to work with esp.
After some hours android.esp8266.com became accessible so I was able to setup esp8266 v2.2

dkumariet, 2016-04-21:
today solved..my issue........
completely removed the old arduino from our pc,uni stalling
and download new setup and installing  againg

joshglendenning, 2016-04-21:
You should close the issue if you've solved it.

dkumariet, 2016-04-23:
esp8266 board not work with arduino 1.6.8 #1953

closed 2016-04-23 by dkumariet