esp8266/Arduino #1956: DHCP Options in AP mode

Manawyrm, 2016-04-21:
Hi,

with the ESP in AP mode, it acts as a DHCP server. 
when doing that, it hands out IP addresses to any client and also it tells them network information like the DNS servers and gateways. 

And thats exactly my problem. When a DHCP server hands out DNS server and default route-adresses, a mobile device like my android smartphone will try to use the ESP8266 as a router to get to the internet -- which will of course fail. 

This could be prevented by not sending the DHCP options 3 and 6. 
Screenshot: https://screenshot.tbspace.de/vetdnfuhizk.png

Is there anyway to disable or configure the DHCP payload which is sent by the ESP? 

Thanks,
Tobias

WereCatf, 2016-04-21:
In user_interface.h there is wifi_softap_set_dhcps_offer_option(), which, as far as I can tell, controls whether to send routing-information to clients or not. `wifi_softap_set_dhcps_offer_option(OFFER_ROUTER, 0)` should disable sending that.

I have not tried this, but after calling WiFi.begin() calling wifi_softap_set_dhcps_offer_option() might work.

igrr, 2016-04-27:
@Manawyrm did this solution work for you?

Manawyrm, 2016-04-27:
I unfortunatly can't test it at the moment, because I have lent out the ESP with the electronics to another team member who is developing on the hardware side of things.
I will come back to you as soon as possible. 

Sorry for the delay and keeping this open for so long!

datacute, 2016-04-27:
The code to include various options is here:
https://github.com/esp8266/Arduino/blob/18f66e99693020a758b8f5ac92e042cbdecb7760/tools/sdk/lwip/src/app/dhcpserver.c#L129

NarinLab, 2017-09-26:
Is there any clear method to customize the DHCP Configuration in ESP8266 Arduino?
I need to customize my DHCP range from 192.168.244.x to 192.168.43.x.
There is not enough documentation on how to setting NodeMCU/ESP8266 Arduino Language DHCP settings in AP Mode.

Thanks!

WereCatf, 2017-09-26:
@NarinLab Take a look at http://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/soft-access-point-class.html#softapconfig -- if you specify the ESP8266's IP-address as e.g. 192.168.43.1, the function softAPConfig() sets the DHCP-range as 192.168.43.100 - 192.168.43.200.

If you want to customize the range further, I recommend taking a look at https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp and setting the DHCP-server up manually.

NarinLab, 2017-09-26:
@WereCatf 

Thankyou very much, 

Problem solved by:

```
IPAddress ip(192, 168, 43, 1);
IPAddress gateway(192, 168, 43, 1);
IPAddress subnet(255, 255, 255, 0);
WiFi.softAPConfig(ip, gateway, subnet);
```

closed 2017-09-26 by devyte

schweini, 2018-05-02:
@Manawyrm did you get this to work? I'm also stuck on getting my Android phone to simultaneously connect to the network 192.168.4.1/24 via wifi, and the rest of the internet via mobile data.

black3806, 2019-01-29:
extern "C" {
#include "user_interface.h"
}
#include <ESP8266WiFi.h>        // Include the Wi-Fi library

const char *ssid = "NightVisionAP";
const char *password = "";

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.println('\n');

  uint8 mode = 0;
  wifi_softap_set_dhcps_offer_option(OFFER_ROUTER, &mode);
  WiFi.softAP(ssid, password);             // Start the access point
  Serial.print("Access Point \"");
  Serial.print(ssid);
  Serial.println("\" started");

  Serial.print("IP address:\t");
  Serial.println(WiFi.softAPIP());         // Send the IP address of the ESP8266 to the computer
}

void loop() { }

abdolahibabak, 2024-08-29:
NarinLab is right
This code worked for me too

Previously, in the standard examples, when I used the following configuration, there was no problem.

IPAddress ip(192, 168, 1, 1);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);
WiFi.softAPConfig(ip, gateway, subnet);

But if I changed the IP address, the connection in some mobile phones was constantly disconnected and reconnected, but in some phones it worked only a little slowly.

I came to the conclusion that **in the ESP8266 12F in access point mode, the IP address and the gateway must be the same**.
I have no reason for this except that whenever I set these two addresses equal, the problem is solved and the program works perfectly on all phones.

I used similar configurations below and they all worked

IPAddress ip(192, 168, 4,10);
IPAddress gateway(192, 168, 4,10);
IPAddress subnet(255, 255, 255, 0);
WiFi.softAPConfig(ip, gateway, subnet);

Maybe in the access point mode, the ISP has the task of the gateway. So both addresses must be the same. Of course, I don't have enough knowledge of the network!

Thanks to NarinLab for the interesting idea