esp8266/Arduino #1957: nodemcu Board crashes at HTTP SPIFFS update

Manawyrm, 2016-04-21:
### Basic Infos

#### Hardware
Hardware:		ESP-12/nodemcu 0.9
Core Version:      	2.2

Hi! 

I'm using an ESP-12/nodemcu 0.9 board and I'm trying to OTA update the SPI filesystem and firmware. 
I have C&Psted the code exactly from the example-code in the library.

`		WiFi.begin(ssidC, pskC);
		while (WiFi.waitForConnectResult() != WL_CONNECTED)
		{
			delay(100);
		}

		Serial.println("Update SPIFFS...");
		t_httpUpdate_return ret = ESPhttpUpdate.updateSpiffs("http://dnarobot.tbspace.de/v1/fetchUpdateSPIFFS.php");
		if(ret == HTTP_UPDATE_OK) {
			Serial.println("Update sketch...");
			ret = ESPhttpUpdate.update("http://dnarobot.tbspace.de/v1/fetchUpdate.php");

			switch(ret) {
				case HTTP_UPDATE_FAILED:
					Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
					break;

				case HTTP_UPDATE_NO_UPDATES:
					Serial.println("HTTP_UPDATE_NO_UPDATES");
					break;

				case HTTP_UPDATE_OK:
					Serial.println("HTTP_UPDATE_OK");
					break;
			}
		}
`

After starting the update, an HTTP request is sent successfully to my server: 
` - - [21/Apr/2016:05:55:44 +0200] "GET /v1/fetchUpdateSPIFFS.php HTTP/1.0" 200 1028384 "-" "ESP8266-http-Update"
`
and it got answered with a 1 MiB file. 

`Request: /updateFirmware
 Arguments:

Update SPIFFS...

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld
�`

The serial-log unfortunatly contains this message, so right after updating the SPIFFS, it reboots. 
Interestingly, the SPIFFS is already correctly written at this point, the new data is present. 
But it should go further and then also flash the firmware, but it unforunatly doesn't do that. 

Any ideas? 
another thing: I'm not to sure about the flash-size of my ESP-module. I think it is the "large" 16Mbit/4 MiB version, because I have successfully uploaded stuff into there, but I'm not really sure. Any way to find out for sure? 

Thanks,
Tobias

WereCatf, 2016-04-21:
A 1 MiB - file? That's quite big, how have you partitioned your flash? Are you using 1M sketch/3M SPIFFS, or 3M sketch/1M SPIFFS? Enabling DEBUG_HTTP_UPDATE ESP8266httpUpdate.cpp might also provide some useful information on this.

Manawyrm, 2016-04-21:
I'm using the eagle.flash.4m1m.ld linker-file, which looks like this: https://screenshot.tbspace.de/pihoeuybqjk.png

The IDE calls that 3M sketch and 1M SPIFFS, but that doesn't seem to be true, judging by the 2048KB marked as empty in the comment up there. 

I will try and compile it again with debugging features.

Manawyrm, 2016-04-21:
Ha! Got it :-) 

```
Update SPIFFS...
Update sketch...
HTTP_UPDATE_OK

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
@cp
```

While enabling the debugging in ESP8266httpUpdate.h, I found that there's the function
`    void rebootOnUpdate(bool reboot)`
That thing should absolutly be documented somewhere :-) 
Setting the value to false fixed the problem and allowed me to successfully update my board! NICE :-)

Thanks for your help!

closed 2016-04-22 by Manawyrm

Vamshi253, 2016-09-26:
Hi Manawyrm,

Thanks for sharing your code.

At this moment I am engaged in developing the ESP8266 module which measures temperature and humidity.

Concerning Over The Air (OTA) update , I need to do it using HTTP Server and is based on continuous automated deployment of firmware on ESP8266.

But I cannot be able to find much resources in detailed on the internet.

Could you help me regarding this?

Moreover, is it necessary to update SPIFFS before updating the firmware?

WereCatf, 2016-09-26:
@Vamshi253 

> But I cannot be able to find much resources in detailed on the internet.

You didn't even look at any of the example-sketches that come with this core, did you? There is a complete example on how to do OTA from an HTTP-server at https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/examples/httpUpdate/httpUpdate.ino , for example.

Also, trying to hijack old, closed threads like this is not OK.

Vamshi253, 2016-09-26:
Hi WereCatf,

Thank you for the quick response and sharing the example.

With this, I can be able to program the client (ESP8266) side.

Moreover, I need to develop the things at server side also. At the moment I have just installed Apache server with php support.

But the thing is, I cannot be able to configure and develop the app on the server side.

I need a step by step guide to implement the server side app.

Could you help me?