esp8266/Arduino #1958: How to connect via WPS?

re3lex, 2016-04-21:
Hi,

My router is setup to accept clients using WPS button.
I'm trying to connect my ESP (please find listing below) to that router using WiFi.beginWPSConfig() method but program hangs about 3 minutes on that line an then reports about failed connection.

I googled a lot but found couple of blog posts but code used there doesn't work for some reasons.

How to connect to router using WPS?
Is that possible?

```
#include <ESP8266WiFi.h>
 
 
void setup() {
  Serial.begin(115200);
 
  // Long delay required especially soon after power on.
  delay(4000);
  // Check if WiFi is already connected and if not, begin the WPS process.
  if (WiFi.status() != WL_CONNECTED) {
      Serial.println("\nAttempting connection ...");
      WiFi.beginWPSConfig();
      // Another long delay required.
      delay(3000);
      if (WiFi.status() == WL_CONNECTED) {
        Serial.println("Connected!");
        Serial.println(WiFi.localIP());
        Serial.println(WiFi.SSID());
        Serial.println(WiFi.macAddress());
      }
      else {
        Serial.println("Connection failed!");
      }
  }
  else {
    Serial.println("\nConnection already established.");
  }
}
 
void loop() {
  // put your main code here, to run repeatedly:
 
}
```

copa2, 2016-04-24:
Worked for me in previous version.
You have to set STA mode (WiFi.mode(WIFI_STA);). In AP mode i could not get it working.

re3lex, 2016-04-24:
hi!
 thanks for reply.
May I ask you share piece of WPS connection code here that's working for you for sure?

copa2, 2016-04-24:
`
#include <ESP8266WiFi.h>

void setup() {
  Serial.begin(115200);

  Serial.println("\nPress WPS button on your router ...");
  delay(5000);
  startWPSPBC();
}

bool startWPSPBC() {
  Serial.println("WPS config start");
  // WPS works in STA (Station mode) only -> not working in WIFI_AP_STA !!! 
  WiFi.mode(WIFI_STA);
  delay(1000);
  WiFi.begin("foobar",""); // make a failed connection
  while (WiFi.status() == WL_DISCONNECTED) {
    delay(500);
    Serial.print(".");
  }
  bool wpsSuccess = WiFi.beginWPSConfig();
  if(wpsSuccess) {
      // Well this means not always success :-/ in case of a timeout we have an empty ssid
      String newSSID = WiFi.SSID();
      if(newSSID.length() > 0) {
        // WPSConfig has already connected in STA mode successfully to the new station. 
        Serial.printf("WPS finished. Connected successfull to SSID '%s'", newSSID.c_str());
        // save to config - we could also save "","" so next WiFi.begin("","") will use stored values
        //qConfig.wifiSSID = newSSID;
        //qConfig.wifiPWD = WiFi.psk();
        //saveConfig();
      } else {
        wpsSuccess = false;
      }
  }
  return wpsSuccess; 
}

void loop() {
  // put your main code here, to run repeatedly:
}
`

This worked for me. 
Important is that the router is in WPS mode before you call WiFi.beginWPSConfig(). 
If you want better handling for errors and timeouts you have to implement your own beginWPSConfig.

re3lex, 2016-04-25:
Hi @copa2 
Thanks a lot for sharing you code!

One question for code above: isn't this piece of code will run infinitely due to  missed password during first run? 
```
  WiFi.begin("foobar",""); // make a failed connection
  while (WiFi.status() == WL_DISCONNECTED) {
    delay(500);
    Serial.print(".");
  }
```

> Important is that the router is in WPS mode before you call WiFi.beginWPSConfig().
Do you mean I should press WPS button on router first before ESP executes `WiFi.beginWPSConfig()` ?

copa2, 2016-04-25:
Well this code was meant just exemplary for the WPS connect.

I explicitly create a failed connection there, as you might have already some state for ssid and password stored in EPROM.
The while is waiting for another state than WL_DISCONNECTED(in this case it would normaly go into WL_NO_SSID_AVAIL, except you have a SSID foobar :-)).

> Do you mean I should press WPS button on router first before ESP executes WiFi.beginWPSConfig() ? 
Yes

Probably here a more complete example for your case: 
https://gist.github.com/copa2/fcc718c6549721c210d614a325271389

re3lex, 2016-04-25:
I never knew that WPS buttons pressing order is important :)
Thank you for detailed explanation and example!
I hope this thread can help other guys who is looking for working WPS connect example :)

copa2, 2016-04-25:
Button press order shouldn't be important. But my experiments told me other...
If router is already in WPS state, then ESP connects quite fast. 
The other way around the wait and successrate is variable.

Evert8266, 2016-06-19:
Im fairly new at the arduino ide and esp8266. I have used the  example code from copa. It works well but when using the serial monitor I only see the main text comments after the first time running after flashing. I would have thought that I should see all the serial print text after power down. Had me confused if the 8266 was connected or not. An ip scan showed that it was connected. Is there something that I do not understand in the running of the sketch. Does it do some other check so as not to show all the text. If I reflash then I see all the serial print text but only once. Thanks for helping with the wps code.

rudi48, 2016-10-25:
With the help of copa2 and others, I have written a working example code with application **Transparent Serial Bridge**. It uses a **Push Button Switch** and a **LED** for information.
Please see at [WiFiTelnetToSerialWPS](https://github.com/rudi48/WiFiTelnetToSerialWPS).
Regards, Rudolf

dalbert2, 2017-07-20:
The WPS function seems to work reliably with older 802.11n access points (Trendnet TEW-652, Belkin N300), but fails reliably with 802.11ac access points (AT&T/Pace 5268AC and Asus TM-AC1900).  
 I am using Arduino/EPS8266 (SDK version: 1.5.3(aec24ac9)).

The failure is always the same; it finds the access point, showing it's SSID correctly but fails to successfully process the public key provided by the access point and acquire the PSK (see trace below).  Has anyone been able to connect to an 802.11ac access point using WPS and if so, can you share details (model #, anything special you had to do in the code)?  Thanks!

> wifi_wps_disable
> wifi_wps_enable
> wps scan
> build public key start
> build public key finish
> f r0, wps discover [ATT4PcDe8QR]
> scandone
> WPS: neg start
> f r0, scandone
> state: 0 -> 2 (b0)
> state: 2 -> 3 (0)
> state: 3 -> 5 (10)
> add 0
> aid 4
> cnt
> process pubkey start
> wps: ignore overlap identifier
> wps: ignore overlap identifier
> wps: ignore overlap identifier
> wps: ignore overlap identifier
> wps: ignore overlap identifier
> wps: ignore overlap identifier
> state: 5 -> 2 (17c0)
> rm

devyte, 2017-07-20:
Two things:
Core 1.5.3 is old, try latest git.
WPS is vulnerable, you shouldn't use it. Anyone listening to the
transaction can steal your keys and gain access to your network.

rudi48, 2017-07-20:
@devyte 
Yes, WPS is vulnerable, but what is the  alternative?
The time window for listening is very small, about 2 minutes.
You must have a special box in the reach of your WLAN router, listening 24h a day.
You must have a program to catch the WPS telegram and send it to someone.
That looks like a job for a secret service.
If you are in a position to fear a secret service, do not use WPS.

dalbert2, 2017-07-21:
@devyte 
Security isn't nearly as much of a concern in my application as ease of use so WPS is very much preferred; and as Rudi asked, what's the alternative?

1.5.3 is the SDK version in the latest released Arduino core for ESP8266, so this is a current problem for the Arduino ESP8266 environment and others are likely to encounter it.

I'm an experienced embedded developer using the gcc toolchain for ARM/Cortex, but pretty new to the whole Arduino/ESP8266 environment and Tensilica generally.  I can try a newer SDK to see if it will help, but it will take me time unless there is documentation on how to upgrade the kit within Arduino Core.

Thanks in advance for any help you can provide.

devyte, 2017-07-21:
There is documentation on how to set up the latest git in the formal docs
on readthedocs. Just follow the links.

About the security, I understand about ease of use. But just for
completeness, I've cracked my neighbor's wep, and then when he changed to
WPA I cracked his WPS. Believe me, it's not hard at all, and there are
tutorials and tools everywhere on how to do it, even several methods. He's
a friend, so I just messed with him, but I know several people who do it
for other reasons. I also know that some burglars here have started using
such methods in high-end neighborhoods near where I live, which is why I'm
a bit paranoid about this.
This is an IoT device, and IoT devices, like home automation devices, rely
heavily on the security of the wifi rather than security provided onboard.
If that security has holes, you open up your house to enterprising
neighbors, or worse.
Alternative? Captive portal on the softap, that's what most everyone else
is using. Not a one-button solution, but if you implement it right it
should be just as easy.

dalbert2, 2017-07-21:
@devyte
I know you are trying to be helpful, but I have extensive (25 yrs) experience implementing secure embedded systems and fully understand the vulnerabilities of WEP and WPS.  I've made a business decision to provide WPS on this device and am interested in making it work successfully.  I've already provided a hotspot with captive portal that allows more sophisticated users to configure the network, but for many users of this product, that will add unacceptable complexity and so I want to provide WPS as an option.

If you are knowledgeable about the WPS issues _on the ESP8266_ (the topic of this post) and can help, your assistance will surely be appreciated.  I even understand warning folks (once) about the vulnerabilities of WPS.  However, the purpose of these forums is for developers to help each other and this post specifically is for help getting WPS working on the ESP8266; insisting that users simply not use the feature they are asking for help with or suggesting that they try the development branch (without knowing that the issue is resolved there) or to just RTFM is not actually helping and degrades the utility of these forums.

devyte, 2017-07-21:
@dalbert2 said: 

> I've made a business decision

9/10 people who ask about WPS don't know about its flaws. If you're making a conscious decision despite them, that's up to you, as long as you're aware.

> the topic of this post

This post is about getting WPS to work at all, for which an example has been left as a link. Personally, I think the issue can be closed. This post is not about WPS on n vs. ac routers. You already opened issue #3445 for that, so your comment/question here is just duplicating.

> If you are knowledgeable about the WPS issues on the ESP8266

I'm not, but I _am_ knowledgable about what is expected of this open source issue tracker. And it IS an issue tracker, not a business environment, or a university environment, or even a discussion forum. The purpose of an issue tracker is to track issues and collect solutions, unlike a discussion forum, where you can discuss between many. This is why you're expected to read the docs and try the latest git BEFORE you open an issue, to check if your problem is already addressed. After that, you're also expected to search if there is an issue already opened, or a pending PR that might address your issue. After that, you're expected to ask on forums (not here) whether anyone else has had your issue. And only then are you expected to open an issue. 
Please keep in mind that this is a repository of free contributions from collaborators that work on their free time, and not a service. The people here are very few and overcollapsed as it is (at 939 open issues, and counting), so don't expect anyone to use their time to do things you could do yourself. 
For example:
-a quick search in the SDK changelogs shows that WPS is mentioned in the release notes of 1.5.4.
-there is mention of WPS not working on dual-radio routers, including an Asus one, in the espressif forums. Did you read through that discussion?
-you could try wireshark et al to analyze the exchanges between the ESP and n router and compare vs. the ESP and ac router, to look for differences.

So please go do your homework, and report back on the issue you already opened.

> extensive (25 yrs) experience

Sounds like we're about the same. If I ever go to Maryland, let's have a beer and compare embedded stories.

devyte, 2017-07-21:
@igrr Considering that there is a code example reported as working, I think the original issue here is addressed. Do you prefer to close this as-is, or request a PR with @rudi48 's working example to be added to the code base?

dalbert2, 2017-07-21:
@devyte When you're next in Maryland, I'll buy the beer.  

I agree that many folks will not appreciate the risks inherent in WPS and it is certainly fair to warn them.  Similarly, anyone trying to get ESP8266 WPS working would want to know that it appears not to work on most new access points...that seems relevant to this post because the suggested solution won't work for many.  Per your suggestion, I've opened a separate issue for the 802.11ac AP incompatibility.

If you are the moderator for this forum/tracker, you are of course entitled to enforce whatever posting rules you wish.  I would hazard a guess that 99% of the issues and comments posted here do not meet your quality standard.  I searched this forum for everything related to WPS before reporting the apparent .ac incompatibility; this flags a serious defect in the current released software that was not already in the issue tracker; that seems like exactly the sort of thing you want in an issue tracker...I don't think you want to limit the reporting of defects to folks who are actively developing the codebase or familiar with its internals.  

I would be interested in the notes you found on the dual-radio routers not working; a link would be very nice since your google-fu is apparently better than mine as I've spent quite a lot of time searching for more information on this issue and did not find it (I only found "Optimized WPS function" in the 1.5.4 SDK release notes which doesn't really suggest a fix for complete failure with .ac routers).

If I can make time to learn the SDK codebase and the WPS protocol specifics I'll try to fix the defect, but as I mentioned, I'm new to this platform, up to my eyeballs in alligators that are far removed from WPS, and just thought I'd post the defect to warn others, alert the devs, and see if anyone else had already found a solution.  Either way, I'm very grateful to those who have been working so hard developing this platform and have made it available to the rest of us as open source.

Cheers!

devyte, 2017-07-22:
@dalbert2 ok David, I'm happy we're in agreement. Let's move the discussion to your new issue one.

devyte, 2017-10-15:
Closing as duplicate, example(s) will be added in #2628 .

closed 2017-10-15 by devyte