esp8266/Arduino #1961: analogRead and ESP.getVcc doesn't work when WiFi radio disabled

torntrousers, 2016-04-22:
The ESP.getVcc() always just returns 4094 and analogRead(A0) returns 1024 when the WiFi radio has been disabled (eg following ESP.deepSleep(1, RF_DISABLED).

This is with Arduine IDE 1.6.8 and ESP core 2.2.0.

This code demonstrates the problem (you can just touch the A0 pin with a finger to test the values change):
```
extern "C" {
  #include "user_interface.h"
}
#include <ESP8266WiFi.h>

//ADC_MODE(ADC_VCC); // enables reading the ESP8266 supply voltage

byte rtcStore[1] __attribute__((aligned(4))); 

void setup() {
  Serial.begin(115200); Serial.println();
  system_rtc_mem_read(64, rtcStore, 1);
  if (rtcStore[0] != 125) {
    rtcStore[0] = 125;
    system_rtc_mem_write(64, rtcStore, 1);
    ESP.deepSleep(1, RF_DISABLED);
//    ESP.deepSleep(1, RF_DEFAULT);
  }
}

void loop() {
//   Serial.println(ESP.getVcc());
   Serial.println(analogRead(A0));
   delay(1000);
}
```

duncan-a, 2016-04-22:
I thought the ESP rebooted when coming out of deep sleep - in which case,
it would never get to the loop would it?

pieman64, 2016-04-22:
@duncan-a you are right and wrong. ESP does reboot but if rtc_mem == 125 then it will reach loop().

duncan-a, 2016-04-22:
Yep, I wasn't thinking straight...

I have a sketch that reads the Vcc prior to sleeping (in loop) and that's
been 100% for a couple of weeks. It happily sends DHT22 and Vcc readings to
dweet.io every ten minutes.

petraq, 2016-04-27:
I have same issue ... ESP.getVcc() always return 4094 with same version of Arduino IDE (1.6.8) and ESP core (2.2.0) for this sketch: https://github.com/tzapu/DeepSleepDHT22

torntrousers, 2016-05-03:
FYI - asked about this on the Espressif SDK forum and they say it might be fixed in 1.5.3 http://bbs.espressif.com/viewtopic.php?f=7&t=2093&sid=8410c002f432033e2d464fe66934bb41

igrr, 2016-05-06:
Git version has been updated to 1.5.3. Please give it a go.

torntrousers, 2016-05-06:
Smashing, that does indeed appear to fix it - both analogRead and ESP.getVcc().

closed 2016-05-06 by torntrousers