esp8266/Arduino #1962: addServiceTxt String  return void not bool

far5893, 2016-04-22:
void addServiceTxt(String name, String proto, String key, String value){
    addServiceTxt(name.c_str(), proto.c_str(), key.c_str(), value.c_str());
  }

should be 

bool  addServiceTxt(String name, String proto, String key, String value){
    return addServiceTxt(name.c_str(), proto.c_str(), key.c_str(), value.c_str());
}

I'm trying to access txt of queried services can you give me an example ?

i found only 
  String hostname(int idx);
  IPAddress IP(int idx);
  uint16_t port(int idx);
to access queried services.

igrr, 2016-05-10:
I don't think mDNS library currently supports TXT queries. Looking at the code, only PTR queries are implemented.

devyte, 2017-10-17:
The referenced functions still have the wrong return type in latest git.

devyte, 2018-07-12:
This is already fixed.

closed 2018-08-01 by devyte