esp8266/Arduino #1963: BasicHTTPClient not working ?

bistory, 2016-04-22:
### Basic Infos

#### Hardware
Hardware:			NodeMCU v1.0 (ESP12E)
Core Version:      	2.2.0

### Description

Hi ! I'm new to the esp8266 world and I'm trying to test wifi using Arduino IDE and the NodeMCU dev board.
I have tested the Hello Server example and it works fine but now when I'm trying the BasicHTTPClient, the board crash when connecting to my AP.
I also tried to set an AP with my phone but the results remain the same...

Maybe it's a dumb problem but I'm stuck with it now :-(

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

/**
 * BasicHTTPClient.ino
 *
 *  Created on: 24.05.2015
 *
 */

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

void setup() {

    USE_SERIAL.begin(115200);
   // USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    WiFiMulti.addAP("AndroidAP", "password");

}

void loop() {
    // wait for WiFi connection
    USE_SERIAL.print("[WIFI] Waiting...\n");
    if((WiFiMulti.run() == WL_CONNECTED)) {

        HTTPClient http;

        USE_SERIAL.print("[HTTP] begin...\n");
        // configure traged server and url
        //http.begin("https://192.168.1.12/test.html", "7a 9c f4 db 40 d3 62 5a 6e 21 bc 5c cc 66 c8 3e a1 45 59 38"); //HTTPS
        http.begin("http://192.168.1.5:1880/test.html"); //HTTP

        USE_SERIAL.print("[HTTP] GET...\n");
        // start connection and send HTTP header
        int httpCode = http.GET();

        // httpCode will be negative on error
        if(httpCode > 0) {
            // HTTP header has been send and Server response header has been handled
            USE_SERIAL.printf("[HTTP] GET... code: %d\n", httpCode);

            // file found at server
            if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                USE_SERIAL.println(payload);
            }
        } else {
            USE_SERIAL.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }

        http.end();
    }

    delay(10000);
}

```

### Debug Messages

```
0x401004f4: calloc at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/umm_malloc/umm_malloc.c line 1682
0x40106dd4: pvPortZalloc at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/heap.c line 33
0x4021c891: wifi_param_save_protect_with_check at ?? line ?
0x401056a2: spi_flash_read at ?? line ?
0x4021c979: system_param_save_with_protect at ?? line ?
0x4021cd68: wifi_station_ap_number_set at ?? line ?
0x4021cec3: wifi_station_ap_number_set at ?? line ?
0x4021cf50: wifi_station_ap_number_set at ?? line ?
0x4021cf8c: wifi_station_set_config at ?? line ?
0x402026c4: ESP8266WiFiSTAClass::disconnect(bool) at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 432
0x4021c5de: wifi_get_opmode at ?? line ?
0x4021d407: wifi_station_get_connect_status at ?? line ?
0x40202704: ESP8266WiFiSTAClass::status() at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 440
0x4021ad96: ets_vsnprintf at ?? line ?
0x402024c3: ESP8266WiFiMulti::run() at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp line 156
0x402051a0: Print::write(unsigned char const*, unsigned int) at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.cpp line 76
0x402041fd: Print::write(char const*) at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.cpp line 76
0x402042b4: Print::printf(char const*, ...) at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.cpp line 76
0x40202098: loop at /Users/thomaslete/Projects/Arduino/BasicHttpClient/BasicHttpClient.ino line 41
0x40204cc4: operator new(unsigned int) at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/abi.cpp line 30
0x402025a7: __gnu_cxx::new_allocator ::allocate(unsigned int, void const*) at /Users/thomaslete/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/ext/new_allocator.h line 104
:  (inlined by) std::_Vector_base   >::_M_allocate(unsigned int) at /Users/thomaslete/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_vector.h line 168
:  (inlined by) void std::vector   >::_M_emplace_back_aux (WifiAPlist_t const&) at /Users/thomaslete/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/vector.tcc line 404
0x401004d8: malloc at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/umm_malloc/umm_malloc.c line 1662
0x401053fc: ets_timer_arm_new at ?? line ?
0x40202675: ESP8266WiFiMulti::APlistAdd(char const*, char const*) at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp line 198
0x40204e7c: esp_yield at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/core_esp8266_main.cpp line 43
0x40202690: ESP8266WiFiMulti::addAP(char const*, char const*) at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp line 39
0x40202052: setup at /Users/thomaslete/Projects/Arduino/BasicHttpClient/BasicHttpClient.ino line 36
0x40204ec8: loop_wrapper at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at /Users/thomaslete/Library/Arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/cont.S line 109
```

kaeferfreund, 2016-04-23:
I also have problems regarding the HTTPClient...

bistory, 2016-04-24:
I tested the same request in the same conditions using nodemcu lua 0.9.5 and it is working perfectly... I guess it is not a hardware problem :)

innesm, 2016-04-24:
I couldn't get a simple HTTP GET working on either a NodeMCU, or a Wemos D1R2. The error: "connection refused". After pulling the label "2.2.0" it works - previously I was on the latest code.

kaeferfreund, 2016-04-25:
should be fixed with edaae2c19417e038c5ed12fc692b013d9c7c7a2e

bistory, 2016-04-26:
@kaeferfreund thanks, I will test that ASAP !

igrr, 2016-05-10:
Looking at the stack trace, i think this was actually the same issue as #1997. Anyway, should be fixed now in git version.

closed 2016-06-23 by igrr

Thijxx, 2017-11-16:
This seems te be an issue still. Copy/paste of example code and I get the 'connection refused'. Quadrupple checked the URL, read probably 100 tickets about this problem but not a solution anywhere. I use platformio, maybe they use an old version?

riker65, 2018-01-06:
Have the same issue
running lolin and arduino ide

manishsasmal, 2018-03-15:
i am facing the same issue. I think it might a problem with server. I am using a shared hosting from vultr.com i am getting error 400 Bad request. Server could not understand your request. but when i am using a shared hosting from others it is working perfectly. May be i am missing something in server part. I am using apache, php. When i am hitting the url from browser it is working perfectly but not from esp. If you find any solution please post here. my code is as below

#include <ESP8266WiFi.h>
#include <SPI.h>
#include "MFRC522.h"

int frequency=1000; //Specified in Hz
int buzzPin=0; //D3
int ledPin=2; //D4

#define RST_PIN 5
#define SS_PIN 4
MFRC522 rfid(SS_PIN, RST_PIN);
unsigned int hex_num;

const char* ssid = "ssid";
const char* password = "password";
const char* host = "hostname";

void setup() {
Serial.begin(9600); // Initialize serial communications
SPI.begin(); // Init SPI bus
rfid.PCD_Init(); // Init MFRC522
pinMode(ledPin,OUTPUT);
pinMode(buzzPin,OUTPUT);
WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED)
{
delay(500);
}

}

void loop() {
if(WiFi.status() == WL_CONNECTED)
{
digitalWrite(ledPin, HIGH);
}
else
{
digitalWrite(ledPin, LOW);
}
// Look for new cards
String rfid_esp="rfid=";
if ((rfid.PICC_IsNewCardPresent()) & (rfid.PICC_ReadCardSerial()))
{
Serial.print(F("Card UID:"));
hex_num = rfid.uid.uidByte[0] << 24;
hex_num += rfid.uid.uidByte[1] << 16;
hex_num += rfid.uid.uidByte[2] << 8;
hex_num += rfid.uid.uidByte[3];
long int NFC_id= (long int)hex_num;
if(NFC_id < 0)
{
NFC_id = NFC_id * -1;
}
Serial.println(NFC_id);
/*for (byte i = 0; i < rfid.uid.size; i++)
{
Serial.println(rfid.uid.uidByte[i]);
rfid_esp += rfid.uid.uidByte[i];
if(i<3)
rfid_esp += ".";
}
Serial.println(rfid_esp);
*/
//hit cloud api
WiFiClient client;
const int httpPort = 80;
if (!client.connect(host, httpPort))
{
Serial.println('not connected');
// return;

}
String url = "project/register/";
url += String(NFC_id);
Serial.println(url);

client.print(String("GET ") + url + " HTTP/1.1\r\n" +
           "Host: " + host + "\r\n" + 
           "Connection: close\r\n\r\n");
tone(buzzPin, frequency);

delay(1000);
noTone(buzzPin);
while(client.available())
{
String line = client.readStringUntil('\r');
Serial.print(line);
}
}

}