esp8266/Arduino #1964: WiFiClient connect() fails randomly

HorstWalter, 2016-04-23:
Hardware:			ESP-07
Core Version:      	2.1.0 (i think, where do I see it?)

Problem description
Every 5 minutes, I connect fromthe ESP to a Linux remote server (via WAN) which has an open port (and implements my own binary protocol). The ESP8266 connects to this remote port and exchanges data.

It works, but randomly the connect() call fails. The Linux server is running rock solid. I can connect from another Linux box to the same port with the same protocol without any failures.

Is it possible that the yield() call within connect() is sometimes not sufficient, so that connect() returns a failure because the ESP has not enough time to establish a connection?

igrr, 2016-04-27:
Can you do a wireshark/tcpdump trace of what is happening on the network?

HorstWalter, 2016-04-27:
How do I put a PC with wireshark between the ESP and the remote server, which is not in the LAN (has a public IP and stands hundreds kilometres away at an ISP)?

igrr, 2016-05-10:
With wireshark you can actually capture packets from the wireless network your ESP is connected to.

devyte, 2017-10-17:
No feedback in over a year, closing per #3655 . If the issue is still valid, please open a new one, and add the relevant details, as well as a MCVE sketch.

closed 2017-10-17 by devyte