esp8266/Arduino #1967: EEPROM can only read one string at once?

dandandandandandandan, 2016-04-24:
IDE 1.6.8
ESP8266 Arduino 2.2.0
nodeMCU 0.9

I don't get any compilation errors or warnings.

If i read more than 1 string from the EEPROM the ESP only displays the latest one. It's like it somehow empties the first string.

```
while (addr1 < 133)
    {
      //read saved wifi SSID
      savedSsid[addr3] = EEPROM.read(addr1);
      addr3 = addr3 + 1;
      addr1 = addr1 + 1;
    }
    //read saved wifi PASS
    while (addr2 < 183)
    {
      savedPass[addr4] = EEPROM.read(addr2);
      addr4 = addr4 + 1;
      addr2 = addr2 + 1;
     }
  Serial.println(savedSsid);
  Serial.println(savedPass);
```

In this case only the "savedPass" would show up in the serial window.
If i put a Serial.print after each while then both of the strings show up in the serial window (so that kinda confirms that i can read them and saved them properly in the EEPROM, i confirmed the latter with an EEPROM dump) but when it comes to thelast two lines the first string "savedSsid" fails to display again.
If i put an EEPROM.end() and EEPROM.begin(512) between the two while()s then the strings show up with all kinds of errors.

dandandandandandandan, 2016-04-24:
This is more interesting:

I put a Serial.print after each while into the code:

```
Serial.println("Printing out SSID and password");
     //read saved wifi SSID
     while (addr1 < 133)
    {
      savedSsid[addr3] = EEPROM.read(addr1);
      addr3 = addr3 + 1;
      addr1 = addr1 + 1;
    }
    Serial.println(savedSsid);
    //read saved wifi PASS
    while (addr2 < 183)
    {
      savedPass[addr4] = EEPROM.read(addr2);
      addr4 = addr4 + 1;
      addr2 = addr2 + 1;
     }
  Serial.println(savedPass);
  Serial.println(savedSsid);
  Serial.println(savedPass);
```

Then this is what it would display in the serial monitor:

```
Printing out SSID and password
myssid
weakwifipass1234

weakwifipass1234
```

WereCatf, 2016-04-24:
It looks like you may have a buffer-overrun. How do you define savedPass and savedSsid?

dandandandandandandan, 2016-04-24:
As
```
char savedSsid[32];
char savedPass[32];
```
after the includes.

WereCatf, 2016-04-24:
Well, then your addr4 is incorrect.

WereCatf, 2016-04-24:
`savedPass[addr4] = EEPROM.read(addr2);`
Is the only line that actually changes anything after the first while-loop, so I would like you to double-check that your addr4 is actually correct.

dandandandandandandan, 2016-04-24:
Inspired by your comment i started to mess with the sizes of the char arrays and setting both of them to [33] seemingly solved the problem.

WereCatf, 2016-04-24:
Aye, very classic buffer-overrun -- you were reading one byte too many in your while-loop. Well, good thing that it's solved then!

closed 2016-04-24 by dandandandandandandan