esp8266/Arduino #1968: ESP8266WebServer conflict with pubsubclient/WiFiClientSecure

AdamMiltonBarker, 2016-04-24:
I have developed a program that boots into an AP to allow you to enter credentials needed for MQTT broker and wifi etc. 

Once the credentials have been saved using SPIFFS, the prgram will then connect to MQTT and carry o, or it should do, I was getting exception 29 and narrowed it down to ESP8266WebServer was conflicting. 

I have tried putting it all in one individual function that is only called if wifi was not successful and get:

In lambda function:

IntelliLanEntrySenseESPDevConfs:375: error: 'server' is not captured

I have tried recoding a program that took we weeks to get sorted and nothing works. I have also tried to find a way to destroy the ESP8266WebServer that is defined if Wifi is connected and I have tried moving the define inside of functions that only run when wifi is connected and none of this allows me to destroy the ESP8266WebServer instance and free up the memory. 

The exception 29 happens on connection attempt to the broker, the exact same program without ESP8266WebServer works as expected, can anyone put me out of my misery on how to either not define the server or destroy it if not needed please. Would be really appreciative this is the last thing in   a very long project.

WereCatf, 2016-04-24:
>  or destroy it if not needed please. 
You could allocate the server dynamically, like with the following pseudocode:
```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

ESP8266WebServer *server;

void setup(){
  //This allocates a new instance of ESP8266WebServer
  server=new(ESP8266WebServer);
  if(server==NULL) //Insert code to handle allocation-error
  //Setup your server here
  server->begin();
}

servernolongerneeded(){
  //This destroys the allocated server and frees up all the resources it used
  delete(server);
}
```

That said, I doubt it fixes your error.

AdamMiltonBarker, 2016-04-24:
Thank you for the quick reply do you have any docs on setting up the server dynamically please, been a long night would be great to find the docs.

AdamMiltonBarker, 2016-04-24:
Oh just seen this "That said, I doubt it fixes your error." what can fix the error then. The error is not there if the server is not created when not needed ie if wifi is connected there is no need to start a server or it will conflict.

WereCatf, 2016-04-24:
I just provided you with a pseudocode-example. You set it up just like you set up a statically defined one, but use `server->` instead of `server.` when calling its functions.

AdamMiltonBarker, 2016-04-24:
OK thanks, The issue is as far as I can see that when pubsub tries to connect it tries to write to the same blocks, so the solution is to not have the server start when it is not needed, will the above fix it or am I better giving up on it ?

WereCatf, 2016-04-24:
I do not know. Try it, that's the easiest way of knowing whether it works or not.

AdamMiltonBarker, 2016-04-24:
Am doing now hopefully it will thanks again.

AdamMiltonBarker, 2016-04-24:
Exception 28 now

Exception (28):
epc1=0x40209613 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000014 depc=0x00000000

ctx: cont 
sp: 3fff25d0 end: 3fff27e0 offset: 01a0

>>>stack>>>
3fff2770:  3ffe9744 ffffffcd 3fff178c 4020c9bc  
3fff2780:  0000000a 00ffffff 3fff178c 4020ca08  
3fff2790:  3fff14d4 ffffffcd 3fff178c 4020781c  
3fff27a0:  3ffe9f18 0c01a8c0 00000000 4020e608  
3fff27b0:  feefeffe feefeffe feefeffe 3fff17b8  
3fff27c0:  3fffdad0 00000000 3fff17b0 4020d648  
3fff27d0:  feefeffe feefeffe 3fff17c0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

WereCatf, 2016-04-24:
Lemme guess: your code tries to use server without you having allocated anything in it?

AdamMiltonBarker, 2016-04-24:
The server doesn't get allocated unless it is needed I will post the code:

`/*
 Copyright (c) 2016 TechBubble Technologies and other Contributors.
 
 The MIT License (MIT)
 Permission is hereby granted, free of charge, to any person obtaining a copy
 of this software and associated documentation files (the "Software"), to deal
 in the Software without restriction, including without limitation the rights
 to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 copies of the Software, and to permit persons to whom the Software is
 furnished to do so, subject to the following conditions:
 The above copyright notice and this permission notice shall be included in all
 copies or substantial portions of the Software.
 THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
 SOFTWARE.
 
 Contributors:
    Adam Milton-Barker  - Initial Contribution
    Adam Mosely  - Tester
    John Ducrest  - Tester
*/

#include <FS.h> 
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <PubSubClient.h>
#include <WiFiClientSecure.h> 
#include <ArduinoJson.h>
#include <ESP8266mDNS.h>

ESP8266WebServer *server;

WiFiClientSecure espClient;
PubSubClient client(espClient);

const char* ssid = "TechBubbleAP";
const char* passphrase = "thereisnospoon";
String st;
String content;
int statusCode;
String serverType ="";

String esid = "";
String epass = "";
  
String locationID = "";
String zoneID = "";
String deviceID = "";

char charBuf[50];
char deviceName[100]; 
char mqttUsername[100]; 
char mqttPassword[100]; 

const char* mqtt_server = "iot.techbubbletechnologies.com";
int  mqttPort = 8883;
 
const int actuator1Pin = 2;
String lastStatus = "";

long lastMsg = 0;
char msg[50];
int value = 0;

[... 400 lines cut ...]
    }
    client.loop();
    long now = millis();
    if (now - lastMsg > 2000) {
      lastMsg = now;
      if(checkDoorSense()){
        Serial.println("OPEN");
        if(lastStatus != "OPEN"){
          lastStatus = "OPEN";
          char json[] = "{\"Sensor\":\"DoorSense1000\",\"SensorID\":1,\"SensorValue\": \"OPEN\"}";
          publishToDeviceSensors(json);
          char jsonW[] = "{\"WarningType\":\"Entry\",\"WarningOrigin\":\"DoorSense1000\",\"WarningValue\": \"OPEN\",\"WarningMessage\": \"Door is open\"}";
          publishToDeviceWarnings(jsonW);
        }
      } else {
        Serial.println("CLOSED");
        if(lastStatus != "CLOSED"){
          lastStatus = "CLOSED";
          char json[] = "{\"Sensor\":\"DoorSense1000\",\"SensorID\":1,\"SensorValue\": \"CLOSED\"}";
          publishToDeviceSensors(json);
        }
      }
    }
  }
}`

AdamMiltonBarker, 2016-04-24:
The server doesn't get allocated unless it is needed I will post the code:

`/*
 Copyright (c) 2016 TechBubble Technologies and other Contributors.
 
 The MIT License (MIT)
 Permission is hereby granted, free of charge, to any person obtaining a copy
 of this software and associated documentation files (the "Software"), to deal
 in the Software without restriction, including without limitation the rights
 to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 copies of the Software, and to permit persons to whom the Software is
 furnished to do so, subject to the following conditions:
 The above copyright notice and this permission notice shall be included in all
 copies or substantial portions of the Software.
 THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
 SOFTWARE.
 
 Contributors:
    Adam Milton-Barker  - Initial Contribution
    Adam Mosely  - Tester
    John Ducrest  - Tester
*/

#include <FS.h> 
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <PubSubClient.h>
#include <WiFiClientSecure.h> 
#include <ArduinoJson.h>
#include <ESP8266mDNS.h>

ESP8266WebServer *server;

WiFiClientSecure espClient;
PubSubClient client(espClient);

const char* ssid = "TechBubbleAP";
const char* passphrase = "thereisnospoon";
String st;
String content;
int statusCode;
String serverType ="";

String esid = "";
String epass = "";
  
String locationID = "";
String zoneID = "";
String deviceID = "";

char charBuf[50];
char deviceName[100]; 
char mqttUsername[100]; 
char mqttPassword[100]; 

const char* mqtt_server = "iot.techbubbletechnologies.com";
int  mqttPort = 8883;
 
const int actuator1Pin = 2;
String lastStatus = "";

long lastMsg = 0;
char msg[50];
int value = 0;

[... 400 lines cut ...]
    }
    client.loop();
    long now = millis();
    if (now - lastMsg > 2000) {
      lastMsg = now;
      if(checkDoorSense()){
        Serial.println("OPEN");
        if(lastStatus != "OPEN"){
          lastStatus = "OPEN";
          char json[] = "{\"Sensor\":\"DoorSense1000\",\"SensorID\":1,\"SensorValue\": \"OPEN\"}";
          publishToDeviceSensors(json);
          char jsonW[] = "{\"WarningType\":\"Entry\",\"WarningOrigin\":\"DoorSense1000\",\"WarningValue\": \"OPEN\",\"WarningMessage\": \"Door is open\"}";
          publishToDeviceWarnings(jsonW);
        }
      } else {
        Serial.println("CLOSED");
        if(lastStatus != "CLOSED"){
          lastStatus = "CLOSED";
          char json[] = "{\"Sensor\":\"DoorSense1000\",\"SensorID\":1,\"SensorValue\": \"CLOSED\"}";
          publishToDeviceSensors(json);
        }
      }
    }
  }
}`

WereCatf, 2016-04-24:
You know... you're calling `server->close();` **before** it is created.

AdamMiltonBarker, 2016-04-24:
Ah that was was in from before I spoke to you I tried it everywhere one second, thanks for pointing it out will check again, sorry been awake all night and day trying to fix, eyes are going a bit funny lol.

AdamMiltonBarker, 2016-04-24:
That solved the issue with the errors when wifi was connected but after clearing the configs and booting again the AP starts up but it is  not possible to connect to it to view the config page.

AdamMiltonBarker, 2016-04-24:
Never mind I think I know why, 2 mins.

AdamMiltonBarker, 2016-04-24:
I cannot thank you enough :D All complete and project now complete. Thanks again buddy.

closed 2016-04-24 by AdamMiltonBarker

WereCatf, 2016-04-24:
You're welcome!

PS.
`const char* ssid = "TechBubbleAP";
const char* passphrase = "thereisnospoon";`

If you use those credentials on your network now is the time to change them since you pasted those publicly and there's no way of scrubbing them off of Internets anymore.

AdamMiltonBarker, 2016-04-24:
Hi thanks, they are just temporary for this one project for the AP, each device will have their own custom ssid and passphrase. Thanks again though mate. Lifesaver :D