esp8266/Arduino #1969: Debug tracing menu items not shown for NodeMCU board

innesm, 2016-04-24:
If NodeMCU board is selected in the IDE, no debug menus (for configuring debug serial output) are shown.

#### Hardware
Hardware:			NodeMCP v1.0
Core Version:      	latest esp8266 from github?

### Description

The required menu defs are just missing from the NodeMCU section of boards.txt. It was possible to get these added with some copy and pasting.

### Settings in IDE
Module:  			NodeMCU v1.0

tkerby, 2017-01-30:
Likewise for the Wemos boards

davisonja, 2017-01-31:
This just needs the entries copied in boards.txt;
I've wondered, given the repetition if it might be worth having a script to
generate the file, which would avoid having to edit all the (other than key
name) identical bits for any and every change.
I've duplicated it for the NodeMCU (which I've been using) so it is just as
simple as copying and pasting the lines (updating the key names).

igrr, 2017-01-31:
Generating boards.txt during build phase (i.e. building boards manager package or bootstrapping a version cloned from git) is the way to go, I think. This also dovetails with your suggestion to keep all flash geometry info in boards.txt. Maybe we can keep it in the linker scripts, and generate boards.txt automatically? Or have another file with all the flash sizes data and generate boards.txt and linker scripts from it.

davisonja, 2017-01-31:
I had wondered about that too, instead of complicating the command lines a
package-build step which would automate the lot including the various menu
options. (Though ultimately a more hierarchical config system for the IDE
would be nice)
If that were the way forward, what do we write such a script in? Are there
any existing similar scripts? I'd rather not add too many arbitrary
dependencies :)

davisonja, 2017-01-31:
> We already employ some amount of Python (tools/get.py, and the unit test
> runner), so writing this in Python wouldn't add more dependencies.
>
> Another possibile option is to define all the input data in XML and use a
> couple of XSLT scripts to generate boards.txt and linker script files. I
> can't immediately tell if this will work well as I don't remember if it is
> possible to do the equivalent of nested loops in XSLT.

I'd vote for Python (would have been my first choice :) so shall I organise
such a script in that in preference to adjusting linker command lines?

igrr, 2017-01-31:
This script would allow us to handle both the linker scripts issue and this one, so I would vote for writing the script. In the past, we had some bugs due to incomplete/inconsistent updates to boards.txt, and such a script could help as well. It will also make adding new options and boards easier.
Input data for this script can also be given directly in Python source code, to avoid defining another format, parsing it, and so on.

devyte, 2017-10-17:
The Menus are undergoing a full rework, which is tracked in #3582 . Closing.

closed 2017-10-17 by devyte