esp8266/Arduino #1971: WiFi.setOutputPower seems not to work

edautz, 2016-04-24:
When I am using the WiFi.setOutputPower function it seems not to work, even with a setting of 0DBm my nodemcu 1.0 board works normally.

I am not a programmer, but could it be that the f in the calculation is wrong? uint8_t val = (dBm*4.0f)

Source:

Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp

void ESP8266WiFiGenericClass::setOutputPower(float dBm) {

    if(dBm > 20.5) {
        dBm = 20.5;
    } else if(dBm < 0) {
        dBm = 0;
    }

    uint8_t val = (dBm*4.0f);
    system_phy_set_max_tpw(val);
}

me-no-dev, 2016-04-24:
0dBm does not mean that the wifi is off. 0dB is actually quite OK if the access point is not too far.
If you can measure the current consumption, you should see it change.

edautz, 2016-04-25:
Thanks for you reply.

The reason why I think the outputpower reduction doesn't is that I
managed to get a NodeMCU 1.0 setup with an HC - SR501 Pir devices stable without false positives using the 2.0.0 board sources (with SDK 1.3.0) and using the SDK commands:

wifi_set_phy_mode(PHY_MODE_11G);
system_phy_set_max_tpw(10);  
WiFi.mode(WIFI_STA);
to reduce the WiFI power.

I tried to upgrade to board version 2.1.0 and 2.2.0, with  

WiFi.setPhyMode(PHY_MODE_11G); 
WiFi.setOutputPower(2.5);
WiFi.mode(WIFI_STA);

commands, but I got tons of false positives.

When I revert my board to 2.0.0 all get stable again.

I also tested the SDK command
system_phy_set_max_tpw(10);
with 2.2.0. 

No result. Lots of false positives.

me-no-dev, 2016-04-25:
could be a bug in espressif's libs then. You are not the first to report issues with wifi since 2.2.0 is out, and 2.2.0 is not on the same SDK version as 2.1.0.

edautz, 2016-04-25:
Agree. I tested SDK 1.5.2 as part of 2.2.0 and SDK 1.5.1 as part of 2.1.0. It seems the problem is present in SDK above 1.3.0  as part of 2.0.0.

I saw SDK 1.5.3 is available. Nothing mentioned in the release notes hoewever.

If I knew how to put it the 2.2.0 package. I could test it.

me-no-dev, 2016-04-25:
It's not that a big deal :) libmain needs to be wakened so umm_malloc can work (there are some docs about it in the repo). Other than that, swapping the libs should do it. Maybe check for changes in the headers in cases where new functions are added.

edautz, 2016-04-26:
@me-no-dev 

In the ../tools/sdk/lib/  directory. The readme.md files states:

Updating SDK libraries

After updating SDK libraries to a new version, do the following changes.

Remove mem_manager.o from libmain.a to use custom heap implementation

xtensa-lx106-elf-ar -d libmain.a mem_manager.o

I suppose that is what you mean with : 
_libmain needs to be wakened so umm_malloc can work (there are some docs about it in the repo)._ 

I just don't know what headers in what files you mean with:
_Maybe check for changes in the headers in cases where new functions are added._

me-no-dev, 2016-04-26:
if the SDK headers have changes between versions 1.5.2 and 1.5.3 (the h files in the sdk/include) then those will need to be updated (but I doubt there are any)

edautz, 2016-04-27:
I wait for changing the libraries until I can actually measure the dBm levels. I found some software that supports displaying the dBm levels of other wifi stations on my laptop. But my build-in wifi card of the laptop is not compatible. I orderred a wifi usb reciever who should support the display of the recieved dBm levels.

edautz, 2016-05-08:
I managed to measure the wifi output levels and noticed that the Wifi output levels decreased when using the SDK command system_phy_set_max_tpw(val). I didn't test the WiFi.setOutputPower(val) command to avoid the chance that there are bugs in this code. But even with a outputlevel of 0dBm and using a Nodemcu 1.0 board I still get false positives. Approx 1/hour.

So there must be another source of interference. After studing the SDK I put SDK commands in my sketch to turn of all the unused wifi functions, but this doesn't resolve the problem. I supect that the interference maybe is caused by a raised wifi powerlevel at a retransmit of the ip packets, but cannot measure that.

I reverted my setup to 2.0.0.0 and using a two unit Esp8266 pir setup with a powerlevel of 2dBM, without any false positives for about two weeks.

closed 2016-05-08 by edautz

Joypacharya, 2016-12-31:
void setup()
{
................................
WiFi.setOutputPower(80);
........
}
with the help of this command you would be increased or extend rf signal.

ArpanHalder, 2017-07-23:
I did get similar effects like you @edautz 
No significant change in wifi strength (used a wifi meter app) and neither any significant change in current drawn (used an ammeter at 3.3v DC power source.)

`#include <ESP8266WiFi.h>
void setup() `{`
  // put your setup code here, to run once:
  WiFi.setOutputPower(0);
  //system_phy_set_max_tpw(10);
  WiFi.softAP("MAVR", "password",3);
}
void loop() {
  // put your main code here, to run repeatedly:
}`
However, I'm not able to use SDK command system_phy_set_max_tpw(val) from my IDE (Arduino 1.6.9). It gives an error: system_phy_set_max_tpw' was not declared in this scope.
Could you please tell what headers to add or how did you use this SDK command in IDE?

edautz, 2017-07-23:
Verstuurd vanaf mijn iPhone

edautz, 2017-08-10:
Alan,

The code to use the SDK commands are:

extern "C" {
#include "user_interface.h"

}

Put them in your variable section of the sketch, before void setup() after the normal #include libraries.

Kind regards,

Eugene Dautzenberg

Verstuurd vanaf mijn iPad