esp8266/Arduino #1972: ESP12-E stops working after multiple requests

J-Arnold, 2016-04-24:
### Basic Infos

#### Hardware
Hardware:               ESP-12e
Core Version:      	2.1.0-rc2

### Description

I developed a Sketch to readout 5 Sensors and other data from the ESP.
When multiple requests are coming from my FHEM application at the same time the ESP stops to response and needs to be resetted.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP12-E)
Flash Size: 		4MB
CPU Frequency:		80Mhz
Upload Using:		115200

### Sketch (witout sensors)

#include <ESP8266WiFi.h>
#include <Ticker.h>

const char* ssid = "xxxx";
const char* password = "xxxxx";
IPAddress ip(192,168,100,201);
IPAddress dns(192,168,100,254);  
IPAddress gateway(192,168,100,254); 
IPAddress subnet(255,255,255,0);

#define GPIO_02_LED 2

Ticker Reset;
int NoOutput = 0;

void  Reset_System () {

  Serial.println("Reset");

  ESP.restart();
}

String milli2time(unsigned long mSekunden)
{  
String TIME = "";
String help = ""; 
      
int Tage          =  mSekunden/86400000;
int Stunden       = (mSekunden-((Tage*86400000)))/3600000; 
int Minuten       = (mSekunden-((Tage*86400000)+(Stunden*3600000)))/60000;
int Sekunden      = (mSekunden-((Tage*86400000)+(Stunden*3600000)+(Minuten*60000)))/1000;
int MilliSekunden =  mSekunden-((Tage*86400000)+(Stunden*3600000)+(Minuten*60000)+(Sekunden*1000));
help = String(Tage);
TIME = help + "Tage ";
help = String(Stunden);
help = "000" + help;
help = help.substring(help.length()-2);
TIME = TIME + help + ":";
help = String(Minuten);
help = "000" + help;
help = help.substring(help.length()-2);
TIME = TIME + help + ":";
help = String(Sekunden);
help = "000" + help;
help = help.substring(help.length()-2);
TIME = TIME + help + " ";
help = String(MilliSekunden);
help = "0000" + help;
help = help.substring(help.length()-3);
TIME = TIME + help;
return TIME;
} 

WiFiServer server(80);

extern "C" {     
  #include "user_interface.h"
  uint16 readvdd33(void);
}// - See more at: http://www.esp8266.com/viewtopic.php?f=13&t=3839#sthash.D22VMrVZ.dpuf

void setup() {
  pinMode(GPIO_02_LED, OUTPUT);
  digitalWrite(GPIO_02_LED, LOW);
  Serial.begin(115200);
  delay(10);

  Serial.println();
  Serial.println("Starting");
  Serial.print("Flash Chip Size = ");
  Serial.println(ESP.getFlashChipSize());

  // Connect to WiFi network

  Serial.println();
  Serial.print("Connecting to ");

  WiFi.mode(WIFI_STA);
  WiFi.config(ip, dns, gateway, subnet); 
  WiFi.begin(ssid, password);

    
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);

    Serial.print(".");

  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println(ssid); 
  Serial.print("WiFi RSSI=");
  Serial.println(WiFi.RSSI());

  
  // Start the server
  server.begin();

  Serial.println("Server started");
 
  // Print the IP address
  Serial.print("Use this URL to connect: ");
  Serial.print("http://");
  Serial.print(WiFi.localIP());
  Serial.println("/");

  digitalWrite(GPIO_02_LED, HIGH);

  Reset.once(300, Reset_System); //reset ESP after 5 Minutes no action from Web
}
 
void loop() {
  String Text = "";
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  
  Reset.detach();
  Reset.once(300, Reset_System); //reset ESP after 5 Minutes no action from Web
  
  // Wait until the client sends some data

  Serial.println("new client");

  while(!client.available()){
    delay(1);
  }
   
  // Read the first line of the request
  String request = client.readStringUntil('\r');

  Serial.println(request);

  client.flush();
  
  // Return the response
  client.println("HTTP/1.1 200 OK");
  client.println("Content-Type: text/html");
  client.println(""); //  do not forget this one
  client.println("<!DOCTYPE HTML>");
  client.println("<html>");
  // Match the request
 
  NoOutput = 0;
  
  if (request.indexOf("/?read=RSSI") != -1) {
    client.print("RSSI=");
    client.println(WiFi.RSSI());

    Serial.print("RSSI=");
    Serial.println(WiFi.RSSI());

    NoOutput = 1;
  } 

  if (request.indexOf("/?read=UPTIME") != -1) {

    Serial.print("UPTIME=");
    Serial.println(milli2time(millis()));

    client.print("UPTIME=");
    client.print(milli2time(millis()));
    client.println("<BR>");
    NoOutput = 1;
  } 
  
  if (request.indexOf("/?info") != -1) {
    //ESP.getResetReason() returns String containing the last reset resaon in human readable format.

    Serial.print("getResetReason = ");
    Serial.println(ESP.getResetReason());

    client.print("getResetReason = ");
    client.print(ESP.getResetReason());
    client.println("<BR>");
    //ESP.getFreeHeap() returns the free heap size.

    Serial.print("getFreeHeap = ");
    Serial.println(ESP.getFreeHeap());

    client.print("getFreeHeap = ");
    client.print(ESP.getFreeHeap());
    client.println("bytes<BR>");
    //ESP.getChipId() returns the ESP8266 chip ID as a 32-bit integer.

    Serial.print("getChipId = ");
    Serial.println(ESP.getChipId());

    client.print("getChipId = ");
    client.print(ESP.getChipId());
    client.println("<BR>");
    //Several APIs may be used to get flash chip info:
    //ESP.getFlashChipId() returns the flash chip ID as a 32-bit integer.

    Serial.print("getFlashChipId = ");
    Serial.println(ESP.getFlashChipId());

    client.print("getFlashChipId = ");
    client.print(ESP.getFlashChipId());
    client.println("<BR>");
    // ESP.getFlashChipSize() returns the flash chip size, in bytes, as seen by the SDK (may be less than actual size).

    Serial.print("Flash Chip Size = ");
    Serial.println(ESP.getFlashChipSize());

    client.print("Flash Chip Size = ");
    client.print(ESP.getFlashChipSize());
    client.println("bytes<BR>");
    //ESP.getFlashChipSpeed() returns the flash chip frequency, in Hz.

    Serial.print("getFlashChipSpeed = ");
    Serial.println(ESP.getFlashChipSpeed());

    client.print("getFlashChipSpeed = ");
    client.print(ESP.getFlashChipSpeed());
    client.println("Hz<BR>");
    //ESP.getCycleCount() returns the cpu instruction cycle count since start as an unsigned 32-bit. This is useful for accurate timing of very short actions like bit banging.

    Serial.print("getCycleCount() = ");
    Serial.println(ESP.getCycleCount());

    client.print("getCycleCount() = ");
    client.print(ESP.getCycleCount());
    client.println("<BR>");
    //ESP.getVcc() may be used to measure supply voltage. ESP needs to reconfigure the ADC at startup in order for this feature to be available. Add the following line to the top of your sketch to use getVcc:

    Serial.print("readvdd33 = ");
    Serial.println(readvdd33());

    client.print("readvdd33 = ");
    client.print(readvdd33());
    client.println("mV<BR>");

    Serial.print("UPTIME=");
    Serial.println(milli2time(millis()));

    client.print("UPTIME=");
    client.print(milli2time(millis()));
    client.println("<BR>");
  } 

  if (request.indexOf("/?read=vdd33") != -1) {
    client.print("vdd33=");
    client.println(readvdd33());

    Serial.print("vdd33=");
    Serial.println(readvdd33());

    NoOutput = 1;
  } 

  if (NoOutput != 1) 
  {
    client.print("Wifi has = ");
    client.print(WiFi.RSSI());
    client.println("DB<br><br>");

    client.print("vdd33=");
    client.print(readvdd33());
    client.println("mV");

    client.print("UPTIME=");
    client.print(milli2time(millis()));
    client.println("<BR>");
  }
  client.println("</html>");
  delay(1);

  Serial.println("Client disonnected");
  Serial.println("");

}

liquidfalcon, 2016-04-26:
I don't see anywhere in your sketch where you're explicitly closing the client with `client.stop();`

Most likely you're hitting the ESP max connection limit (Which is 5, IIRC), since neither the ESP nor the application you made seem to close the connection. Could you try explicitly closing it, as well as instructing the application to close with the HTTP header `Connection: close`?

igrr, 2016-05-10:
Please try using ESP8266WebServer library instead. There are several examples which will get you started. This library avoids many of the errors you are making in managing incoming connections, and it also provides facilities to parse request arguments, headers, and so on.

closed 2016-05-10 by igrr