esp8266/Arduino #1973: Form POST with checkbox not working

viktak, 2016-04-25:
### Basic Infos

#### Hardware
Hardware:			ESP-12

### Description

Running a standard web server on the ESP. Have a basic form with a single checkbox on it: 
`<form class="form-horizontal" id="RelayForm0" role="form" method="post">
<input name="chk0" id="chk0" onclick="document.getElementById('RelayForm0').submit();" type="checkbox" checked=""></form><input name="chk0" id="chk0" onclick="document.getElementById('RelayForm0').submit();" type="checkbox" checked="">
</form>`

Suppose the checkbox is unchecked when the page loads. Clicking the checkbox submits the form, and the following code can correctly detect the changed state of the checkbox:
`    for (size_t i = 0; i < server.args(); i++) {
      Serial.print(server.argName(i));
      Serial.print(": ");
      Serial.println(server.arg(i));
    }
`
Output on serial:
`chk0: on`

However, when the page loads as the checkbox checked, clicking the checkbox submits the form, but the same code doesn't have the checkbox value.
Output on serial:
`plain:`

I would expect the code to detect the server arguments and the output should be like this:
`chk0: off`

### Settings in IDE

Using PlatformIO

Anybody has met this behaviour? What am I missing?

Thanks in advance!

krzychb, 2016-04-25:
Hi @viktak,

Looking on the posted HTML code I am wondering if one opening and two closing ````</form> ```` tags could cause the issue depending on which check box you click.

Krzysztof

[image]

viktak, 2016-04-25:
Hi @krzychb 
The actual HTML is like this:
`
<form class="form-horizontal" id="RelayForm0" role="form" method="post"><input name="chk0" id="chk0" onclick="document.getElementById('RelayForm0').submit();" type="checkbox" checked=""></form>`
or 
`
<form class="form-horizontal" id="RelayForm0" role="form" method="post"><input name="chk0" id="chk0" onclick="document.getElementById('RelayForm0').submit();" type="checkbox" ></form>`
when it's not checked.

I don't know how the bad one made it to the original post.

I am now thinking that the problem lies around the fact that checkboxes don't have a value attribute. Other controls I have tried (and worked perfectly) all have the value attribute.

viktak, 2016-04-25:
Further experimenting shows that if I have multiple checkboxes in the form, then it behaves as follows:
at postback:
each checkbox that is checked appears as `chk0: on`, `chk1: on`, etc. - server arguments are fine
each checkbox that is **not **checked does not appear. - server arguments are missing
if none of the checkboxes are checked then `plain:` appears.  - server arguments are missing

me-no-dev, 2016-04-25:
try adding ```enctype="application/x-www-form-urlencoded"``` to the form properties

me-no-dev, 2016-04-25:
also I'm not positive that checkboxes send their status if not checked. Maybe you want to look towards AJAX form submission if you want to send the values of the checkboxes

viktak, 2016-04-26:
@me-no-dev 

> try adding  enctype="application/x-www-form-urlencoded"  to the form properties

I'm afraid, this didn't give better results.

viktak, 2016-04-26:
@me-no-dev 

> also I'm not positive that checkboxes send their status if not checked

It is possible that this is by design. I come from a c# background and there (server-side) checkboxes report back properly :) so I expected the same behaviour here. 

Anyway, thanks for the AJAX idea, will look into it. I also had an idea to use a hidden control.

If someone could confirm this is indeed the expected behaviour, it would be great.

ps: I will be away for 3 days, so I will not check back here until Friday.

hallard, 2016-04-26:
Got the same, and spent long time to figure it out, when checkbox are not checked you dont't receive the form value/data I also expected to have values 0/1 or off/on on posted form ;-)
What I've done is by default on form handler, consider they're all unchecked and check only the one I receive checked
```arduino
    // Since Checkbox unchecked are not send (we receive only the one checked as cfg_box=on)
    // We clear all related options of them, if checked we will be receive it
    config.config &= ~(CFG_LCD|CFG_DEBUG|CFG_RGB_LED|CFG_AP|CFG_STATIC|CFG_WIFI);
```
Any better option will be appreciated ;-)

me-no-dev, 2016-04-26:
```js
var xmlHttp = new XMLHttpRequest();
xmlHttp.onreadystatechange = function(){
  if (xmlHttp.readyState == 4){
    if(xmlHttp.status != 200) alert("ERROR["+xmlHttp.status+"]: "+xmlHttp.responseText);
    else {
      //process the response
    }
  }
};
var formData = new FormData();
formData.append("check1", document.getElementById("chbox1").checked?"1":"0");
formData.append("check2", document.getElementById("chbox2").checked?"1":"0");
xmlHttp.open("POST", "/form");
xmlHttp.send(formData);
```

wendelb, 2016-04-27:
> It is possible that this is by design. I come from a c# background and there (server-side) checkboxes report back properly :) so I expected the same behaviour here. 

That's not fully correct here. (non JS) HTML Checkboxes don't send their off-state in a Form Request. This topic is heavily discussed on StackOverflow for example [here](https://stackoverflow.com/questions/4260918/why-do-html-checkboxes-only-transmit-an-on-value) or [here](https://stackoverflow.com/a/2770232) as well as [here](http://planetozh.com/blog/2008/09/posting-unchecked-checkboxes-in-html-forms/). The reason server side checkboxes do work that way is because your Server Application knows that there is a checkbox and sets the off-value if the on-state is missing.

You can easily try and test it yourself by using the developer tools integrated in most current browsers and look at the requests, that are sent after submitting your form. Values not specified there will not be reveived by your ESP.

In general: Sending an off-state would require you to define, what to send in case of that state. Otherwise, your checkbox might enter an undefined state (as you can set value="off" as your on-state).

viktak, 2016-05-03:
Thank you all for your input. I am still away, just wanted to keep this thread open. Will be back next week when I will deal with this.

Thanks again!

viktak, 2016-05-09:
Thanks for your ideas. I have now gotten around this using @hallard's method. I will also look into what @me-no-dev suggested.

Thanks again!

closed 2016-05-09 by viktak