esp8266/Arduino #1974: enable sprintf to support printing floats, let the user choose.

aguaviva, 2016-04-25:
Given that an ESP is as powerful as a Pentium Pro, does it make sense to remove the floating point support just to save some mem? 

At a minimum we should let the user choose whether to support %f feature or not.

TheAustrian, 2016-04-25:
I concatenate with String(floatVariable, numberOfDecimals) to print floats - I don't know if that helps in your application, though. I would not mind having %f, though.

aguaviva, 2016-04-25:
thanks and I'd also appreciate a proper fix, there a lots of confused people out there.

Even basic could print floating point numbers :) lets fix this please :)

igrr, 2016-04-25:
Closing as duplicate of #1179.

closed 2016-04-25 by igrr

aguaviva, 2016-04-26:
Is anybody working in this? If anybody submits a pull request with the c implementation used in gcc, would it be accepted?

igrr, 2016-04-26:
This can be implemented as part of a pending pull request: #1752.
Please follow up in the issue linked above: #1179.

aguaviva, 2016-04-26:
Thanks, this helps! BTW you guys rock!