esp8266/Arduino #1978: mdns not answare to linux mdns-scan

far5893, 2016-04-26:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			generic 
Core Version:      	2.1.0

### Description

mdns not answer to linux mdns-scan  
mdns.begin is executed after receiving ip from dhcp in sta mode

addService return void so i don't know if it exit because does not addService
(but i suppose it can return error with minor changes). 

mdns.update is located in loop.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

me-no-dev, 2016-04-28:
mdns is global and asynchronous service.
the following is all you need to start it and add a service
```cpp
void setup(){
  MDNS.begin(hostname);
  MDNS.addService("http","tcp",80);
}
```
If you are using ArduinoOTA, then you should know that MDNS is started by ArduinoOTA and you need to just add the service after you ArduinoOTA.begin()

far5893, 2016-04-29:
I'm not using arduino OTA, do you suggest to remove mdns.update() from loop ?

igrr, 2016-04-29:
Yes, it's sufficient to do `MDNS.begin` and whatever `MDNS.addService` needed in your application.
Make sure you're not creating an instance of MDNS though (as in `ESP8266mDNS mdns;`).

far5893, 2016-04-29:
Ok, now work , but  i have to wait dhcp for assigned ip, what happen if i change ip (dhcp lease) of if i lost for a while AP signal (autoreconect enable) or if  i use wifimulti ?
There is possibility to hook MDNS begin with wifievent to make ip change traking  automatically?

igrr, 2016-05-16:
Re automatic (re-)start of mDNS, there is an open ticket: #1828.

closed 2016-05-16 by igrr

far5893, 2016-05-16:
ok , i will wait for mdns improveent