esp8266/Arduino #1979: WiFiClientSecure.h library error(peekBytes(uint8_t*, size_t)' marked override, but does not override)

parasshah143, 2016-04-26:
### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description
Arduino: 1.6.6 Hourly Build 2015/07/21 05:42 (Windows 7), Board: "lolin(nodemcu V3), Serial, 80 MHz, 9600"

In file included from Publish.ino:1:0:
C:\Users\Paras.Shah\AppData\Local\Arduino15\packages\wemos\hardware\esp8266\1.0.4\libraries\ESP8266WiFi\src/WiFiClientSecure.h:44:10: error: 'size_t WiFiClientSecure::peekBytes(uint8_t*, size_t)' marked override, but does not override
   size_t peekBytes(uint8_t *buffer, size_t length) override;
          ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

### Debug Messages

Arduino: 1.6.6 Hourly Build 2015/07/21 05:42 (Windows 7), Board: "lolin(nodemcu V3), Serial, 80 MHz, 9600"

In file included from Publish.ino:1:0:
C:\Users\Paras.Shah\AppData\Local\Arduino15\packages\wemos\hardware\esp8266\1.0.4\libraries\ESP8266WiFi\src/WiFiClientSecure.h:44:10: error: 'size_t WiFiClientSecure::peekBytes(uint8_t*, size_t)' marked override, but does not override
   size_t peekBytes(uint8_t *buffer, size_t length) override;
          ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

messages here
```
Arduino: 1.6.6 Hourly Build 2015/07/21 05:42 (Windows 7), Board: "lolin(nodemcu V3), Serial, 80 MHz, 9600"

In file included from Publish.ino:1:0:
C:\Users\Paras.Shah\AppData\Local\Arduino15\packages\wemos\hardware\esp8266\1.0.4\libraries\ESP8266WiFi\src/WiFiClientSecure.h:44:10: error: 'size_t WiFiClientSecure::peekBytes(uint8_t*, size_t)' marked override, but does not override
   size_t peekBytes(uint8_t *buffer, size_t length) override;
          ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

igrr, 2016-04-27:
Judging by the `packages\wemos\hardware\esp8266\1.0.4` part, you seem to be using an outdated ESP8266 package from Wemos. Please remove that (`C:\Users\Paras.Shah\AppData\Local\Arduino15\packages\wemos`) and use the following boards manager package URL:
`http://arduino.esp8266.com/stable/package_esp8266com_index.json`.

parasshah143, 2016-04-28:
Thank for your help. Need more help from you.

Question 1:
Can you please tell me what Ip i should write for Ntp server Asia? 
From internet I could find the server name but not Ip.

Question 2:

While Compiling I am facing error:
	at java.util.regex.Pattern$Branch.match(Pattern.java:4604)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4658)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4785)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4717)
	at java.util.regex.Pattern$BranchConn.match(Pattern.java:4568)
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3777)
	at java.util.regex.Pattern$Branch.match(Pattern.java:4604)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4658)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4785)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4717)
	at java.util.regex.Pattern$BranchConn.match(Pattern.java:4568)
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3777)
	at java.util.regex.Pattern$Branch.match(Pattern.java:4604)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4658) 

Looking forward to your response.

Thanks,
Paras Shah

parasshah143, 2016-04-29:
.error: failed sending 1055 bytes
.error: failed sending 1054 bytes
.error: failed sending 1054 bytes
.error: failed sending 1052 bytes
.error: failed sending 1055 bytes
.error: failed sending 1053 bytes
.error: failed sending 1051 bytes
.error: failed sending 1076 bytes
.error: failed sending 1071 bytes
.error: failed sending 1065 bytes
.error: failed sending 1078 bytes
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
[image]

parasshah143, 2016-04-29:
[image]

closed 2016-04-29 by parasshah143