esp8266/Arduino #1980: 2.2.0 compatibility issue with ILI9341

HorstWalter, 2016-04-26:
I updated to core 2.2.0 last night - to see if it solves my other issue with randomly failing WiFiClient::connect()[https://github.com/esp8266/Arduino/issues/1964](url) - and since then my ILI9341 TFT display stopped working, using the Adafruit GFX library.

From my obervations, the GFX::fillScreen() method brings the ILI9341 controller into a bad state. The display turns blank and freezes. I recognized that fillScreen() calls SPI::writePattern(). It seems your latest change to SPI::writePattern() causes an incompatibility. Maybe it is a problem of the GFX library, but I doubt it.

Strange enough, it is only the first fillScreen() call that causes the freeze. After several hourts of experimenting, I found a workaround. Inserting the following code after the first fillScreen() call:

	tft.writecommand(0x11);		// Sleep Out
	delay(120);				// the controller needs 120 ms to wake up

brings the display back to life and successive calls to fillScreen() do not cause any problems. The freeze seems to be caused, because the ILI9341 controller entered its sleep state.

WereCatf, 2016-04-26:
Assuming you're talking about Links2004's version of the ILI9341-library I don't understand how it could be writePattern() that's the problem as it's also used by fastHLine and fastVLine, and as you yourself say, fillScreen() works after the first time it's called.

You could always replace SPI.h and SPI.cpp with the versions they were before the change and see if it fixes the problem -- if it does, then I suppose the function needs to be looked at.

WereCatf, 2016-04-26:
In addition, I have several ILI9341-displays and I can't make any of them break with the newer writePattern().

WereCatf, 2016-04-26:
I just tried with Links2004's library and nope, no issues -- works just as it's supposed to.

HorstWalter, 2016-04-26:
It is not the Links2004 version, it is the version of Adafruit, they now support the ESP8266. If it works with Links2004, then it seems to be a problem in the Adafruit code.

WereCatf, 2016-04-26:
Adafruit's original version doesn't even use writePattern(). You must be confused somewhere.

HorstWalter, 2016-04-26:
Right, it seems to be the Link2004 version. Was confused by the comment header.

Here is my code:
`Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_DC);

void SetupTft()
{
	pinMode(TFT_DC, OUTPUT);	// this is important!

	// boot display
	tft.begin();
	tft.setRotation(3);
	tft.setTextSize(1);
	tft.fillScreen(gBkgColor);
	SetTextColor(ILI9341_WHITE);
}`

If I comment out the call to fillScreen() the display does not freeze.

Two things should be mentioned:
1) I am not using CS. It is hardwired via pulldown to ground. So I defined ILI9341_USE_NO_CS which required to fix some logic in the code (nothing crtitical).
2) I am not using RST (default constructor sets it to -1). It is hardwired via pulllup to 3.3V.

With 2.1.0 it worked 100% stable. Never had an issue with frozen display. If you say you have no problems, I assume you are using RST?

WereCatf, 2016-04-27:
No, I am not using RST. You know, writePattern() just writes bits out the SPI-port, if your code was working right it, but writePattern() was doing something weird it'd just end up messing the pixels on the display. For it to be able to put the display to sleep DC-pin would have to be held low -- putting display to sleep or whatever are commands and the display only ever accepts commands when DC is low, never when it's high. writePattern() does not touch any other pins than MISO, MOSI and SCLK. This makes me think whatever changes you made in the library are the reason behind your issues.

As I said, you can always grab SPI.cpp from e.g. https://github.com/esp8266/Arduino/raw/703612927091453787cf933f37429fcc33ad785e/libraries/SPI/SPI.cpp and replace the current one with that, but I highly doubt it'll change anything.

HorstWalter, 2016-04-27:
ILI9341_USE_NO_CS was named USE_NO_CS in the header and ILI9341_USE_NO_CS in the .cpp file  - or vice versa. That was all I changed. If no one else has identical issues, we should close this case since I have a workaround.

closed 2016-04-27 by HorstWalter