esp8266/Arduino #1981: ctx:cont issue

rokison, 2016-04-26:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	(not sure here, would 0.4.5 sound reasonable?)

### Description

I am getting a ctx:cont error when I run my latest program. I gather that this is something called the "software watchdog", but have no idea what's causing it. It also seems to happen at random stages in my program depending on only minor changes (like which characters are in my specials array)

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K (64K SPIFFS)
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		(not shown)

### Sketch

```cpp

/*
   Web client sketch for IDE v1.0.1 and w5100/w5200
   Uses POST method.
   Posted November 2012 by SurferTim
*/
#include <ESP8266WiFi.h>
#include <SPI.h>

const char* ssid = "HamptonCloud";
const char* pass = "givemewifi";

//change this for your unique MSISDN and Password
char MSISDN[] = "447417894200";
char password[] = "6a996a8d341b73d6af1ee72b9746bd5a";

//change this for the phone number you want the messages sent to
char target[] = "447810202143";
char message[] = "test";

//the truphone Hampton SMS server
char serverName[] = "hampton.tp.mu";

//the HTTP port of the server
int serverPort = 80;

// change to the page on that server
char pageName[] = "/sendsms.php";

char urlRequest1[]="<?php\n$access = \"447417800001:jkiug2et987246iohqekjbvp8yqweegjkn\";\n$api_url = \"http://hampton.tp.mu/sendsms.php\";\n$data = array (\n'";
char urlRequest2[]="' => $from,\n'";;
char urlRequest3[]="'   => $to,\n'";;
char urlRequest4[]="' => $message\n);\n$url = $api_url;\n$ch = curl_init();\ncurl_setopt($ch, CURLOPT_USERPWD, $access);\ncurl_setopt($ch, CURLOPT_URL, $url);\ncurl_setopt($ch, CURLOPT_RETURNTRANSFER, true);\ncurl_setopt($ch, CURLOPT_CUSTOMREQUEST, 'POST');\ncurl_setopt($ch, CURLOPT_POSTFIELDS, json_encode($data));\n$result = json_decode(curl_exec($ch));\n$status = curl_getinfo($ch, CURLINFO_HTTP_CODE);\ncurl_close($ch);\n\n?>";;
char urlRequestFull[900];

WiFiClient client;

//count how many texts have been sent in this session
int textCount = 0;

// set this to the number of milliseconds delay
// this is 30 seconds
#define delayMillis 30000UL

unsigned long thisMillis = 0;
unsigned long lastMillis = 0;

void setup() {
  Serial.begin(115200);

  wifiConnect();

  delay(2000);
  Serial.println(F("Ready"));

}

void loop()
{

  thisMillis = millis();

  //check to see if enough time has passed to trigger a new message
  if (thisMillis - lastMillis > delayMillis)
  {
    lastMillis = thisMillis;

    generateRequest();

    if (!postPage(pageName, urlRequestFull)) Serial.print(F("Fail "));
    else Serial.print(F("Pass "));
    textCount++;
    Serial.println(textCount, DEC);
  }
  while(true);
}

byte postPage(char* page, char* thisData)
{
  int inChar;
  char outBuf[80];

  Serial.print(F("connecting to server..."));

  if (client.connect(serverName, serverPort) == 1)
  {
    Serial.println(F("connected"));

    // send the header
    sprintf(outBuf, "POST %s HTTP/1.1", page);
    Serial.println(outBuf);
    sprintf(outBuf, "Host: %s", serverName);
    Serial.println(outBuf);
    Serial.println(F("Connection: close\r\nContent-Type: application/x-www-form-urlencoded"));
    sprintf(outBuf, "Content-Length: %u\r\n", strlen(thisData));
    Serial.println(outBuf);

    // send the body (variables)
    char buf[1000];
    //sendLongMessage(thisData);
    Serial.print(urlEncode(buf, thisData));
  }
  else
  {
    Serial.println(F("failed"));
    return 0;
  }

  int connectLoop = 0;

  //now check for a response
  while (client.connected())
  {
    while (client.available())
    {
      inChar = client.read();
      Serial.write(inChar);
      connectLoop = 0;
    }

    delay(1);
    connectLoop++;
    if (connectLoop > 10000)
    {
      Serial.println();
      Serial.println(F("Timeout"));
      client.stop();
    }
  }

  Serial.println();
  Serial.println(F("disconnecting."));
  client.stop();
  return 1;
}

void wifiConnect() {
  Serial.print("Connecting to ");
  Serial.println(ssid);
  //WiFi.begin(ssid, pass, localIPAddress);
  WiFi.begin(ssid, pass);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");

  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

//char specials[] = "$&+,/:;=?@ <>#%{}|~[]`\n";
char specials[] = "&+,@ <>#%{}|~[]`\n\"";

static char hex_digit(char c)
{
  return "0123456789ABCDEF"[c & 0x0F];
}

char *urlEncode(char *dst, char *src)
{ char c, *d = dst;
  while (c = *src++)
  { if (strchr(specials, c))
    { *d++ = '%';
      *d++ = hex_digit(c >> 4);
      c = hex_digit(c);
    }
    *d++ = c;
  }
  *d = '\0';
  return dst;
}

void generateRequest() {
  sprintf(urlRequestFull,"%s%s%s%s%s%s%s",urlRequest1,MSISDN,urlRequest2,target,urlRequest3,message,urlRequest4);
}

void sendLongMessage(char* thisData){
  char buf[91];
  char buf2[200];
  for(unsigned int i=0;i<10;i++){
memcpy(buf, &thisData[i*10], 90 );
buf[90] = '\0';
client.print(urlEncode(buf2, buf));
  }
  
}

void reference()
{

  client.println("User-Agent: Arduino/1.0");
  client.println("Connection: close");

}

```

### Debug Messages

```
ctx: cont 
sp: 3ffeac10 end: 3ffeade0 offset: 01b0

>>>stack>>>
3ffeadc0:  00000000 00000000 3ffeae04 4020184a  
3ffeadd0:  00000000 00000000 3ffe9dc0 40100378  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

Any help gratefully received!
Many thanks

liquidfalcon, 2016-04-26:
Your infinite loop `while (true);` in `loop` would generally cause the WDT to fire quite shortly after your program starts.

rokison, 2016-04-27:
Thanks, that seems to have fixed that particular issue.
I now just need to figure out why my HTTP message is being rejected by the server!

closed 2016-05-10 by igrr