esp8266/Arduino #1982: What is the purpose of seemingly redundant lines in platform.txt file?

krzychb, 2016-04-26:
#### Hardware
Hardware:			any board
Core Version:      	Latest commit [234a874](https://github.com/esp8266/Arduino/commit/234a8743c9533d1b12d7962802869fe68e83179f)

#### Description

I am wondering what is the purpose of [this script line](https://github.com/esp8266/Arduino/blob/master/platform.txt#L121) in platform.txt file?
It looks to me like some old version of [another line above](https://github.com/esp8266/Arduino/blob/master/platform.txt#L108).

By the same token, what is the purpose of [all these lines](https://github.com/esp8266/Arduino/blob/master/platform.txt#L114-L120) that look redundant to [these lines above](https://github.com/esp8266/Arduino/blob/master/platform.txt#L101-L106)?
I have commented them out and do not see any issues uploading in Arduino IDE.1.6.8.

Could you, please, clarify what is the application that requires [this part](https://github.com/esp8266/Arduino/blob/master/platform.txt#L114-L121) of platform.txt file?

<br />

Krzysztof

krzychb, 2016-04-27:
The reason I entered this issue is my confusion seeing the hardcoded network port ❶.  I thought I had to manually enter it in [platform.txt](https://github.com/esp8266/Arduino/blob/master/platform.txt)  if I change it in the sketch.

Then, after looking another time, I realized that network port is resolved by Arduino IDE and passed as parameter called ‘’’network.port’’’ in different line ❷.

Therefore the line with hardcoded network port ❶ looks to me like a dead code or a code kept for some other application.

Therefore, to avoid possible confusion by others, I propose to remove everything marked as ❸ unless it is kept for some other application.

What do you think?

Krzysztof

[image]

igrr, 2016-04-28:
I think this is a leftover from the "Upload Using" menu which was recently removed. Should be safe now to remove these lines.

me-no-dev, 2016-04-28:
yup :) exactly :)

krzychb, 2016-04-28:
Done and dusted :smile:

closed 2016-04-28 by krzychb