esp8266/Arduino #1983: list.push_back() causes linker error

stnkl, 2016-04-27:
### Problem description

The code below gives me a weird linker error. The C++ code itself should be fine so maybe it is caused by some conflict with Arduino-ESP8266 related libraries that I'm not aware of? Are lists not supported on the ESP8266? I am compiling the sketch for the Generic ESP8266 Module.

### Sketch

```cpp

#include <list>

void setup ()
{
  std::list<int> mylist;
  mylist.push_back(0);
}

void loop() {

}
```

### Error Message

```
sketch\sketch_apr28a.ino.cpp.o:(.text.setup+0x0): undefined reference to `std::__detail::_List_node_base::_M_hook(std::__detail::_List_node_base*)'

sketch\sketch_apr28a.ino.cpp.o: In function `construct<std::_List_node<int>, int>':

[...]\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\ext/new_allocator.h:120: undefined reference to `std::__detail::_List_node_base::_M_hook(std::__detail::_List_node_base*)'

collect2.exe: error: ld returned 1 exit status

exit status 1
```

igrr, 2016-04-28:
We don't link to libstdc++, but this can be addressed once newlib support is finalized and merged in (#1752).

stnkl, 2016-04-28:
Alright, thanks. Until then I'll be using LinkedList then.

closed 2016-06-23 by igrr