esp8266/Arduino #1984: Exception (28). Crashes instantly on startup

shenry64, 2016-04-28:
### Basic Infos

#### Hardware
Hardware:	        ESP-01
Core Version:      	2.2.0

### Description

The unit crashes upon start up. I've tried the code with multiple units and also put a capacitor over Vcc and Ground. I've managed to get the example OSCuino example [OSCuino example](https://github.com/CNMAT/OSC/blob/master/examples/ESP8266sendMessage/ESP8266sendMessage.ino) and running so I'm not sure where the issue lies.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <OSCMessage.h>
boolean newData = false;
byte type;
byte data;
int timer, start = 0;
byte rc;

byte lacci =0xFF; //byte identifiers for serial communication
byte laxi = 0xFE;
byte layi = 0xFD;
byte lazi = 0xFC;
byte lgxi = 0xFB;
byte lgyi = 0xFA;
byte lgzi = 0xF9;

byte racci = 0xF8;
byte raxi = 0xF7;
byte rayi = 0xF6;
byte razi = 0xF5;
byte rgxi = 0xF4;
byte rgyi = 0xF3;
byte rgzi = 0xF2;

//left vars for averaging

int average_lacc, average_lax, average_lay, average_laz, average_lgx, average_lgy, average_lgz;
int total_lax,total_lacc,total_lay,total_laz,total_lgx,total_lgy,total_lgz;
const int nu = 10; //Amount of values it averages from (high means slower response times but smoother curve)
int lacc[nu],lax[nu],lay[nu],laz[nu],lgx[nu],lgy[nu],lgz[nu]; //pass me to a_stup var 1 (yeah ok)
int lacc_indx = 0;
int lax_indx = 0;
int lay_indx = 0;
int laz_indx = 0;
int lgx_indx = 0;
int lgy_indx = 0;
int lgz_indx = 0; ////pass me to a_setup var 2

//right vars for averaging
int average_racc, average_rax, average_ray, average_raz, average_rgx, average_rgy, average_rgz;
int total_rax,total_racc,total_ray,total_raz,total_rgx,total_rgy,total_rgz;
int racc[nu],rax[nu],ray[nu],raz[nu],rgx[nu],rgy[nu],rgz[nu];
int racc_indx = 0;
int rax_indx = 0;
int ray_indx = 0;
[... 371 lines cut ...]

### Exception Deocder
```
Decoding 21 results
0x4022d4e8: meas_tone_pwr_db at ?? line ?
0x4022a256: tx_pwctrl_init_cal at ?? line ?
0x4022a244: tx_pwctrl_init_cal at ?? line ?
0x4022a64f: tx_pwctrl_init at ?? line ?
0x4022be74: chip_v6_initialize_bb at ?? line ?
0x4022be89: chip_v6_initialize_bb at ?? line ?
0x40227cd3: phy_init at ?? line ?
0x40105efa: register_phy_ops at ?? line ?
0x4022cb0d: register_chipv6_phy at ?? line ?
0x4021c66e: wifi_station_set_default_hostname at ?? line ?
0x4021c659: wifi_station_set_default_hostname at ?? line ?
0x4020249d: __wrap_register_chipv6_phy at C:\Users\shenr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_phy.c line 261
0x40230cd0: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4010564e: spi_flash_read at ?? line ?
0x40219227: rijndaelKeySetupEnc at ?? line ?
0x4021989c: user_uart_wait_tx_fifo_empty at ?? line ?
0x40219877: user_uart_wait_tx_fifo_empty at ?? line ?
0x40105016: wdt_feed at ?? line ?
0x4020396c: OSCMessage::send(Print&) at C:\Users\shenr\Documents\Arduino\libraries\OSC-master/OSCMessage.cpp line 110

```

WereCatf, 2016-04-28:
Well, for one:
`if (ident = lacci){`
should be
`if (ident == lacci){`
etc.

shenry64, 2016-04-29:
Ooh what a mistake to make. That seems to have fixed it. Thanks!

closed 2016-04-29 by shenry64

hallard, 2016-04-30:
@shenry64 

You could even avoid repeat of sending message with a boolean, code will be much smaller to read and you avoid code duplication (even if compiler should optimize it)

Like something like this (not sure about OSC declaration and implementation but it could be something similar)

```arduino
void process_and_send_data(byte ident, byte var){
  int averaged_value;
  bool send = true;
   OSCMessage msg;     

  if (ident == lacci) {
     msg("/left/acc");     
     msg.add(average(lacc,lacc_indx,total_lacc,average_lacc));
  } else if (ident == laxi){
     msg("/left/ax");
     msg.add(average(lax,lax_indx,total_lax,average_lax));
  } else if (ident == layi){
     msg("/left/ay");
     msg.add(average(lay,lay_indx,total_lay,average_lay));
  } else if (ident == lazi){
     msg("/left/az");
     msg.add(average(laz,laz_indx,total_laz,average_laz));
  } else if (ident == lgxi){
     msg("/left/gx");
     msg.add(average(lacc,lgx_indx,total_lgx,average_lgx));
  } else if (ident == lgyi){
     msg("/left/gy");
     msg.add(average(lgy,lgy_indx,total_lgy,average_lgy));
  } else if (ident == lgzi){
     msg("/left/gz");
     msg.add(average(lgz,lgz_indx,total_lgz,average_lgz));
  } else  if (ident == racci){
     msg("/right/acc");
     msg.add(average(racc,racc_indx,total_racc,average_racc));
  } else if (ident == raxi){
     msg("/right/ax");
     msg.add(average(rax,rax_indx,total_rax,average_rax));
  } else if (ident == rayi){
     msg("/right/ay");
     msg.add(average(ray,ray_indx,total_ray,average_ray));
  } else if (ident == razi){
     msg("/right/az");
     msg.add(average(raz,raz_indx,total_raz,average_raz));
  } else if (ident == rgxi){
     msg("/right/gx");
     msg.add(average(rgx,rgx_indx,total_rgx,average_rgx));
  } else if (ident == rgyi){
     msg("/right/gy");
     msg.add(average(rgy,rgy_indx,total_rgy,average_rgy));
  } else if (ident == rgzi){
     msg("/right/gz");
     msg.add(average(rgz,rgz_indx,total_rgz,average_rgz));
  } else {
    send = false;
  }

  if (send {
    Udp.beginPacket(outIp, outPort);
    msg.send(Udp);
    Udp.endPacket();
    msg.empty();
  }
}
```