esp8266/Arduino #1985: gzip Compressed content not served corectly by ESP8266WebServer lib

mkeyno, 2016-04-28:
hi @igrr 
I use the ESP8266WebServer example in AP mode with couple of webpage in SPIFF 
In webpage have code like this 
`
<script type="text/javascript" src="js.tar.gz"></script>
`
my browser send the following header content 
`
Accept-Encoding: gzip, deflate

`
my code process as follow
`
bool handleFileRead(String path){

  if(path.endsWith("/")) path += "index.html";  
  String contentType = getContentType(path); 
  Serial.print(" \t[contentType] " + contentType); // application/x-gzip
  String pathWithGz = path + ".gz";  
  if(SPIFFS.exists(pathWithGz) || SPIFFS.exists(path)){
                           if(SPIFFS.exists(pathWithGz))  path += ".gz";                           
                            File file = SPIFFS.open(path, "r");
                            size_t sent = server.streamFile(file, contentType);///>>>>>>>>>
                            file.close();
                            Serial.println("\t[SENT]");
                            return true;
                            }
  Serial.println("\t[Not Found]");
  return false;
}
`
content type will selected to "Content-Type: application/x-gzip" and it should send it to browser to decompress it but it is not and send it with different contact type which result to not desire action
I think  the problem is , the library don't send the proper header file and it should have following line to notify browser and decompress it
`
Content-Encoding: gzip
`
any help really appreciated

mkeyno, 2016-05-05:
hi @WereCatf, @me-no-dev  it seems Iggor   so busy and understand that for this big man however I really appreciated if you could check this issue and have your comments 
BR

WereCatf, 2016-05-05:
You're not giving all that many details to go on with, do you actually have a gzipped file on SPIFFS that you're trying to send or do you just expect the ESP to compress it on-the-fly?

mkeyno, 2016-05-05:
no , the problem is the webserver library do not send the correct header  so the browser can understand to what to do with .gz file , it must send "Content-Encoding: gzip" in the header, otherwise browser simply download it and not decompress it , it is a bug and should be corrected

igrr, 2016-05-05:
I see that you are passing the file to `server.streamFile` function. Here's the source code for this function:
```c++
template<typename T> size_t streamFile(T &file, const String& contentType){
  setContentLength(file.size());
  if (String(file.name()).endsWith(".gz") &&
      contentType != "application/x-gzip" &&
      contentType != "application/octet-stream"){
    sendHeader("Content-Encoding", "gzip");
  }
  send(200, contentType, "");
  return _currentClient.write(file, HTTP_DOWNLOAD_UNIT_SIZE);
}
```

So if the requested file actually has a .gz extension, its content-type will be set to `application/x-gzip`, and it will be treated as a gzip archive — i.e. the server will not instruct the browser to decompress.
I think this works as expected. If you want the file to be treated as an HTML, image, or whatever, you should set the correct content-type. Then streamFile function will detect that this is *not* an archive, and will add `Content-Encoding` header for you.

WereCatf, 2016-05-05:
Yes, that's what I was just about to say. @mkeyno your handleFileRead() works as it should -- if you have both index.html and index.html.gz on your SPIFFS and a user requests index.html, the function will serve index.html.gz if it exists, but sets its content-type according to the non-gzipped index.html, and then streamFile() sends the header telling the browser to decode it. That is why I asked: do you actually have both the gzipped and non-gzipped file there or not, but you refused to answer the question.

If you don't want to keep both versions then you need to modify your code to determine the correct content-type by removing the ".gz" ending from the filename.

mkeyno, 2016-05-05:
thanks dear @igrr for reply , but I use your sample code and for example change my webpage.html or any source code in  webpage to .gz extension, then according to following code it should find the content type and pass to steamfile , but when I enable my browser header show plugin , it shows my browser not received the expected header content type  and it seams stream function choose another type for it
following is the famous getContentType function

String getContentType(String filename){
       if(server.hasArg("download")) return "application/octet-stream";
  else if(filename.endsWith(".htm")) return "text/html";
  else if(filename.endsWith(".html")) return "text/html";
  else if(filename.endsWith(".css")) return "text/css";
  else if(filename.endsWith(".js")) return "application/javascript";
  else if(filename.endsWith(".png")) return "image/png";
  else if(filename.endsWith(".gif")) return "image/gif";
  else if(filename.endsWith(".jpg")) return "image/jpeg";
  else if(filename.endsWith(".ico")) return "image/x-icon";
  else if(filename.endsWith(".xml")) return "text/xml";
  else if(filename.endsWith(".pdf")) return "application/x-pdf";
  else if(filename.endsWith(".zip")) return "application/x-zip";
  else if(filename.endsWith(".gz")) return "application/x-gzip";
  return "text/plain";
}

mkeyno, 2016-05-05:
thanks @WereCatf  but I remove other extension and have only .gz extension, my browser header show plugin ,shows the server send the incorrect content type , it should mentioned 'Content-Encoding' but it wont

WereCatf, 2016-05-05:
@mkeyno Let me guess: you're trying to request a URL like http://esps.ip.address.here/index.html.gz and you're wondering why it's sending you exactly what you requested?

WereCatf, 2016-05-05:
The problem here is that you're still not giving enough information.

If you removed the non-gzipped file and you only have the gzipped one, and you are requesting a URL like, say, "/index.html",  your handleFileRead() won't work because the line `String contentType = getContentType(path); ` tries to get the content-type from the non-gzipped file -- since the file doesn't exist there it doesn't get content-type for it!

If, on the other hand, you are requesting a URL like, say, "/index.html.gz" then obviously content-type will be set to x-gzip, because that's the file is -- you'd need to modify your function to figure out the content-type in another way.

Either way, the problem lies with your code, not the ESP8266-code, and you need to fix your handleFileRead().

mkeyno, 2016-05-05:
dear @WereCatf  not exactly as I expected , my server send .gz extension as normal stream type and when my browser check the header cant understand it is the Content-Encoding", "gzip"  so them just download it not decompress it , I don't why what is my wrong so the webserver don't send the correct content type to browser

mkeyno, 2016-05-05:
ok dear @WereCatf  if you say library work as it expected I'll check my code again and let you know

mkeyno, 2016-05-05:
I comperes all pages and component to .gz format and replace it with original , but it wont work I thinks the following could should be change 
from
`
bool handleFileRead(String path){ 
   Serial.print(" Request page : " + path); 
  if(path.endsWith("/")) path += "index.html";  
  String contentType = getContentType(path);   
  String pathWithGz = path + ".gz"; 
  if(SPIFFS.exists(pathWithGz) || SPIFFS.exists(path)){
                           if(SPIFFS.exists(pathWithGz))  path += ".gz";                           
                            File file = SPIFFS.open(path, "r");
                            size_t sent = server.streamFile(file, contentType);
                            file.close();
                            Serial.println("\t[SENT]");
                            return true;
                            }
  Serial.println("\t[Not Found]");
  return false;
}
`
to this
`
bool handleFileRead(String path)
{
  String contentType="";
  Serial.print(" Request page : " + path); 
  if(path.endsWith("/"))        path += "index.html"; 
  if(path.endsWith(".gz"))contentType = "application/x-gzip"; 
  String pathWithGz = path + ".gz"; 
  if(SPIFFS.exists(pathWithGz)) contentType = "application/x-gzip"; 
  else                          contentType=getContentType(path);
  
  if(SPIFFS.exists(pathWithGz) || SPIFFS.exists(path))
    {                                                   
	File file = SPIFFS.open(path, "r");
	size_t sent = server.streamFile(file, contentType); 
	file.close();
	Serial.println("\t[SENT]");
	return true;
	}
  Serial.println("\t[Not Found]");
  return false;
}
`

WereCatf, 2016-05-05:
I already told you what the problem is, you are setting contentType wrong! If you want to serve index.html.gz and have the browser decompress it, you need to set contentType to "text/html," **not** "application/x-gzip"

You keep trying to set contentType to "application/x-gzip" and the browser will obviously handle it as a gzip-file to download, not as compressed content for it to view!

mkeyno, 2016-05-05:
@WereCatf you mean for index.html.gz  contentType  need  set to "text/html," and for other compress component like MyJavaScript.js.gz set to "application/x-gzip" or all compress file must set to "text/html,"

WereCatf, 2016-05-05:
All the files' contentTypes have to match what they'd be if they weren't gzipped. Of course you're not going to serve javascript-files as html, you set it to "application/javascript" and that is why I said you need to set the correct contentType. You could use something like e.g. the following:
```
String getContentTypeGz(String filename){
if(server.hasArg("download")) return "application/octet-stream";
else if(filename.endsWith(".htm.gz")) return "text/html";
else if(filename.endsWith(".html.gz")) return "text/html";
else if(filename.endsWith(".css.gz")) return "text/css";
else if(filename.endsWith(".js.gz")) return "application/javascript";
else if(filename.endsWith(".png.gz")) return "image/png";
else if(filename.endsWith(".gif.gz")) return "image/gif";
else if(filename.endsWith(".jpg.gz")) return "image/jpeg";
else if(filename.endsWith(".ico.gz")) return "image/x-icon";
else if(filename.endsWith(".xml.gz")) return "text/xml";
else if(filename.endsWith(".pdf.gz")) return "application/x-pdf";
else if(filename.endsWith(".zip.gz")) return "application/x-zip";
else if(filename.endsWith(".gz")) return "application/x-gzip";
return "application/x-gzip";
} 
```

mkeyno, 2016-05-05:
thanks @WereCatf  so you only change last line  to `return "application/x-gzip";` do I have any problem with this function  when I use mix files (.html & .gz) in SPIFF

WereCatf, 2016-05-05:
"so you only change last line to"

No, read what I wrote.

arlaor, 2016-05-11:
Hi @igrr, @WereCatf, @me-no-dev
Is it mandatory to upload the files to the memory of ESP8266 V1 in .GZ format?
Why not up the files directly to ESP8266 in JS format?

When I create a .html page for my project with [https://01.org/rib/online/](url) this generates files:
jquery.mobile-1.1.0.js and jquery-1.6.4.js
but no
jquery.mobile-1.1.0.js.GZ and jquery-1.6.4.js.GZ

if I want to upload my files to the memory of ESP8266 V1 in .JS format as me directly it generates the creator of html files [https://01.org/rib/online/](url) , 

what should I change programming lines in my sketch for I take the JS files?

I'm trying to PLS USE the code @AdySan / WittyCloudTest, but my files are not GZ but JS, so I have a malfunction skit

 Are sketch lines would change not having GZ files?
<Bool handleFileRead (String path) {
  Serial.println ( "handleFileRead:" + path);
  if (path.endsWith ( "/")) path + = "index.htm";
  String contentType = getContentType (path);
  PathWithGz String = path + ".gz";
  if (SPIFFS.exists (pathWithGz) || SPIFFS.exists (path)) {
    if (SPIFFS.exists (pathWithGz))
      path + = ".gz";
    File file = SPIFFS.open (path, "r");
    size_t sent = server.streamFile (file, contentType);
    file.close ();
    return true;>

Does not, whether it is required to convert the files GZ format, which program should I use to convert?

AdySan, 2016-05-11:
As I understand, large JS files are typically Gzipped to make the page load faster. The browser is supposed to unzip them before using them. I had some trouble running the GZipped files on SPIFFS before so I just unzipped them and uploaded to SPIFFS again. It should work without any changes. loading JQuery is noticeably slower though.

WereCatf, 2016-05-11:
@arlaor Of course you can use uncompressed files if you wish, there is no requirement to gzip the files first. You do have a bug in your code, though:
`PathWithGz String = path + ".gz";`
should be
`String PathWithGz = path + ".gz";`

arlaor, 2016-05-11:
@WereCatf
> Of course you can use uncompressed files if you wish, there is no requirement to gzip the files first.

But if I use uncompressed files, which lines of code should I change?
I tried to use uncompressed and the page is not loaded properly and is too slow

@AdySan
> loading JQuery is noticeably slower though.

I have seen that your use jQuery file in your project. if you do not you use these, that program to create html pages should I use?

WereCatf, 2016-05-11:
@arlaor The code you copypasted works fine as-is for uncompressed files, you have a bug somewhere else if your sketch isn't working with uncompressed files.

AdySan, 2016-05-11:
@arlaor I'm using the FSBrowser [example](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer/examples/FSBrowser/data) as a base, it dousn't really need any js files. I wanted my page to look a little better so used some JQuery controls. Even without that file, the page should load fine.

arlaor, 2016-05-11:
@AdySan @WereCatf
this is my page project [http://drive.google.com/uc?export=view&id=0B3jhI_HoLLTLY0REMjVxcF9tNFVFRTM0c0dVcVF6SllRODVr](url)

but this is what appears loaded
sometimes not load images
and controls the basic load form
don't colors
nothing
basic page

[http://drive.google.com/uc?export=view&id=0B3jhI_HoLLTLTnl2NHNlQ3JuVmlpSkU3M1lpMjNpb1FUYS1Z](url)

@AdySan
I used your code, as this loads a page index.html
and charge my project files with the same structure to ESP8266, but does not work

when I enter my IP error of these lines is generated
FileNotFound: coming from
  server.onNotFound([](){
    if(!handleFileRead(server.uri()))
      server.send(404, "text/plain", "FileNotFound !handleFileRead");

when I enter http://Witty.local/edit error of these lines is generated

FileNotFound: coming from
 server.on("/list", HTTP_GET, handleFileList);
  //load editor
  server.on("/edit", HTTP_GET, [](){
    if(!handleFileRead("/edit.htm")) server.send(404, "text/plain", "FileNotFound handleFileList");

It is why I asked it on your page: Which of the 320 lines that have this sketch, is throwing the index.htm page in the browser? im confuse

arlaor, 2016-05-12:
@AdySan @WereCatf
Question: What is the library or libraries that decodes and load add-ons included in the index.html (* .json, * .js, * gz etc.) and launch the page on the network?

AdySan, 2016-05-12:
@arlaor I'd say that would be [this](https://github.com/AdySan/WittyCloudTest/blob/master/WittyCloudTest/WittyCloudTest.ino#L339) line. The ESP8266WebServer library handles loading of Index.html. Can you try running the FSBrowser example and let us know if that works for you?

arlaor, 2016-05-12:
@AdySan
FSBrowser example works well

arlaor, 2016-05-12:
@AdySan @WereCatf
Here's what I do is change the files loaded into the ESP8266 with my project, maintaining the same structure of files FSBrowser example.
In the sketch [FSBroser.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/FSBrowser/FSBrowser.ino#L73) I change the line 73 (index.htm to index.html):
if (path.endsWith ("/")) path + = "index.html";

I upload the new sketch to ESP8266 and when I open the page http: //esp8266fs.local/
this is takes about 4 minutes (too slow) to load and generated without any format.
[image](http://drive.google.com/uc?export=view&id=0B3jhI_HoLLTLTnl2NHNlQ3JuVmlpSkU3M1lpMjNpb1FUYS1Z)

Therefore I do not understand that but I must modify the sketch to fully open the index.html page with .js dependences

AdySan, 2016-05-12:
Can you just rename your `index.html` file to `index.htm`?

arlaor, 2016-05-12:
@AdySan
to rename index.htm, it disappears the color format on page opening it from my pc.
To upload the ESP8266 no different happens, remains the same.

These are my [files](https://drive.google.com/open?id=0B3jhI_HoLLTLQjg4Q0JkdXpoUWs)

AdySan, 2016-05-12:
@arlaor can you try a flat file structure, not sure if SPIFFS can handle folders.

arlaor, 2016-05-12:
but then the page should not upload images that are in the folder /data/images
I will rise again removing the data folder subfolders.

WereCatf, 2016-05-13:
@arlaor @AdySan I dug into this and the problem is: your filenames are too long. The maximum length of a filename (including its path) is 31 characters. For the files at the root of SPIFFS the path is "/" so for them the maximum length of the filename itself is 30, and for any subdirectories it's obviously even less.

The sketch itself works fine, but you need to shorten your filenames a lot and I recommend you don't use that many subdirs, either. The filename "jquery-mobile-structure-1-1-0.css" is already way over 30 characters in and of itself, for example, but then the path of "/src/css/" leaves you even fewer characters you can use for the filename itself.

WereCatf, 2016-05-13:
@igrr The maximum filename-length should probably be mentioned and explained somewhere, and there are a couple of useful, if not all-that-critical, updates to pellepl's SPIFFS which might be worth merging to the ESP's copy.

arlaor, 2016-05-13:
@WereCatf  @AdySan
ok I understand I will make the changes and to compress files

closed 2016-05-18 by mkeyno

treii28, 2016-08-19:
Why would the streamFile method be using a check for .gz on the end of an existing filename that already has an extension? About 99% of the examples I see out there in net land are suggesting to use a FAT16 file system with is only going to support 8x3 filenames so checking for .gz on the end of something   8x3 isn't going to find anything in the vast majority of cases.
Only if you're using a newer version of SdFat on a Fat32 SD card will you even run into long filenames in the first place. And is this code even SdFat LFN friendly?

treii28, 2016-08-19:
Another possibility - what I'm going to try here shortly. If you're limited by the filesystem and don't want to implement SdFile long file names, try using old dos style approaches. What I did was create a gz directory in the root of my sd card and then I'm putting sub directories for the filetypes. (e.g. js, xml, css, htm) but the files inside are going to have nothing but an 8x2 filename where the 2 is .gz

`
  if (path.endsWith("/")) path += "index.htm";

  if (path.endsWith(".src")) path = path.substring(0, path.lastIndexOf("."));
  else if (path.endsWith(".htm")) dataType = "text/html";
  else if (path.endsWith(".css")) dataType = "text/css";
  else if (path.endsWith(".js")) dataType = "application/javascript";
  else if (path.endsWith(".png")) dataType = "image/png";
  else if (path.endsWith(".gif")) dataType = "image/gif";
  else if (path.endsWith(".jpg")) dataType = "image/jpeg";
  else if (path.endsWith(".ico")) dataType = "image/x-icon";
  else if (path.endsWith(".xml")) dataType = "text/xml";
  else if (path.endsWith(".pdf")) dataType = "application/pdf";
  else if (path.endsWith(".zip")) dataType = "application/zip";
  else if (path.endsWith(".gz")) {
    if(path.startsWith("/gz/html")) dataType = "text/html";
    else if(path.startsWith("/gz/js")) dataType = "application/javascript";
    else if(path.startsWith("/gz/css")) dataType = "text/css";
    else if(path.startsWith("/gz/xml")) dataType = "text/xml";
    else dataType = "application/x-gzip";
  }
`

treii28, 2017-10-07:
I have been playing around with the Asynchronous server code for ESP8266. I wrote my own version of a SPIFFS based server that can handle gzip'd content for various types. I also wrote in some path handlers for the pseudo-directory nature of the SPIFFS file system that will check for alternate spellings on any url. (It first tries to figure out if the url points to a pseudo-dir, and if yes, it will look for an index. If no index exists for the path, it will give a typical file listing.  If those methods fail, it then checks first what you requested as-is, then if not found and no ".min." is in the filename, it will try inserting .min before the extension, if it doesn't end in gz it will also check .gz then finally check minified-gzipped.
I'm still optimizing it some but it is working now reasonably well.

If nothing else, it gives examples of how the content types and encodings work:
[ESPAsyncFlashWebServer](https://github.com/treii28/ESPAsyncFlashWebServer)