esp8266/Arduino #1990: Can't upload (RX pin)

wathanyuphromma, 2016-05-01:
I am building a project using esp8266.
When I connected 

Vcc (esp ) => Vcc (Arduino 3.3 v)
Ground (esp ) =>Ground (Arduino)
CH_PD pin => 3.3 v
TX (esp) => TX (Arduino)
RX (esp) => RX (Arduino)

everything is fine. I can upload and use AT command in serial monitor.
But I have downloaded a project file and see how they are connected like this

Vcc (esp ) => Vcc (Arduino 3.3 v)
Ground (esp ) =>Ground (Arduino)
CH_PD pin => 3.3 v
RX (esp) => TX (Arduino)
TX (esp) => RX (Arduino)

I follow the instruction and suddenly found many errors.
This is the error shown

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x41
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x54
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x2b
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x52
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x53
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x54
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x0d
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x0a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x0a
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x0a

After that I tried to pull the RX(Arduino) pin out , just let it empty and the result is I can upload
But seems it did not work for the output.

So, what should I do? can anyone give me a suggestion?

duncan-a, 2016-05-01:
The 3.3V from the Arduino is severely limited on current and that causes
all sorts of problems with the ESP8266 - it may be that...

wathanyuphromma, 2016-05-02:
So, what should I do next in order to solve this problem?

wathanyuphromma, 2016-05-02:
So, what should I do in order to solve this problem?

duncan-a, 2016-05-02:
To solve the current limitation problem, feed a good, stable, clean, 3.3V
supply to the ESP from an external power supply with sufficient current
capacity.

You don't say which version of Arduino you are using, and you also don't
mention level shifting. Most Arduino's (like the Uno, etc) are 5V logic and
the ESP is 3.3V logic - feeding 5V logic into the ESP inputs can kill it...

Google how to use ESP with Arduino...

closed 2016-12-10 by wathanyuphromma