esp8266/Arduino #1991: Version 2.2.0 Arduino BSP cause HTTPUpdateClient to fail

sk1tt1sh, 2016-05-02:
Hello,

I have verified by making 2.1.0 sketch and uploading to the ESP, and putting a slightly modified 2.1.0 sketch and putting in a folder with the name that the sketch is looking for. When I call the update it works and I see the new serial output.

However I update arduino boards package for esp to 2.2.0 and repeat the same process and the sketch attempts to update but then resets the board and runs the original sketch.

Oddly it prints this at 115200 instead of 74880 when it resets into the normal sketch
---------
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld
---------

I'm using ESP12-e.

Thank you.

sk1tt1sh, 2016-05-16:
Is there a change to how to use this library?

igrr, 2016-05-16:
You need to reset the board after uploading sketch over serial, otherwise it will get stuck while doing software reset later on.

sk1tt1sh, 2016-05-16:
When doing httpUpdate from 2.1.0 based app to 2.2.0 app with httpUpdate even resetting power for the board does not allow it to complete update. However 2.1.0 can use httpupdate to 2.1.0 version over and over without power reset at all. 

Also, thank you so much for all your work on this project I sincerely hope my tone does not appear ungrateful as I appreciate everything in this project!

sk1tt1sh, 2016-07-12:
https://github.com/esp8266/Arduino/pull/2251 

Issue is caused by a call to reboot being made when updateSPIFFS is called.

closed 2016-07-12 by sk1tt1sh