esp8266/Arduino #1992: WiFi.BSSID() returns slightly wrong address

kdsoo, 2016-05-02:
### Basic Infos

#### Hardware
Hardware:			huzzah esp8266
Core Version:      	2.2.0-rc1

### Description

I need to fetch mac address of AP which esp8266 get connected.

I tried both methods to get the address.
WiFi.BSSID() and WiFi.BSSIDstr()

It turns out to be returning slightly adjusted address then the original one.

the mac address expected to be observed is like follows:
192.168.0.1              ether   90:9f:33:59:02:52   C                     wlan0

but the address indicated by the WiFi.BSSID() and WiFi.BSSIDstr() is like follows:
Node connected to AP: 90:9F:33:5A:02:52

I used following code to fetch the mac address

			uint8* mac = WiFi.BSSID();
			for (int i = 0; i < 6; ++i) {
				String digit = String(*mac, HEX);
				if (digit.length() < 2)
					APMAC += String("0") + digit;
				else
					APMAC += digit;
				mac++;
				Serial.print("["); Serial.print(i); Serial.print("]"); Serial.println(*mac, HEX);
			}
			Serial.print("Node connected to AP: ");
			Serial.print(WiFi.SSID()); Serial.print(", ");

or just Serial.println(WiFi.BSSIDstr()) which returns same wrong one.

Is this a known bug and any work around exists? or I'm just doing in wrong way?

kdsoo, 2016-05-03:
Sorry. It turns out to be my WiFi access point's bug perhaps.
I tried scanning with my laptop's wifi interface and it observes same mac address with the esp node observed.
Only the address showing on WiFi ap's admin page is different.
Weird indeed...
anyway closing this issue. sorry for disturbance.

closed 2016-05-03 by kdsoo

datenen, 2016-05-04:
I think you will find that the MAC addresses are intentionally different between the AP mode and the STA mode. I believe in essence they are different effective "radios" and have to have different MAC addresses. We observe this behavior in our application, which enables both AP and STA access.

Hope this helps!

kdsoo, 2016-05-09:
@datenen 
yes I think you are right.
In the same context, the mac address displayed in WiFi access point's admin page is useless. Who on earth wants to know the mac address which has no chance to be observed?
btw, my wifi ap is from korean company ""iptimes"". I'm running several of access points from that company and they all shows incorrect mac address in their admin pages.